hbldh/bleak #345: How connect two ble device and recevier two ble device data??

lai-Ryan, 2020-10-21:
* bleak version:0.8.0
* Python version:3.7.2
* Operating System:windows10
* BlueZ version (`bluetoothctl -v`) in case of Linux:  no

### Description
Planning to receive data from two BLE devices at the same time and for a long time,
Currently, two BLE devices can be connected at the same time, but using asyncio cannot be used at the same time
Receiving notifications from two devices (because only one at a time), the result will be one notification, receiving a lot of times and the other one is rarely.

### What I Did
def notification_handler1(sender, data):
    print("Device1")
def notification_handler2(sender, data):
    print("Device2")

async def connect_1(address1,CHARACTERISTIC_UUID1, loop):
    async with BleakClient(address1, loop=loop) as client:
        B1 = await client.connect()
        if B1 == True:
            print("Connect[3C:A5:08:0A:B2:3C]successful!!")
        await client.start_notify(CHARACTERISTIC_UUID1, notification_handler1)
        #asyncio.sleep(10.0) 

async def connect_2(address2,CHARACTERISTIC_UUID2, loop):
    async with BleakClient(address2, loop=loop) as client:
        B2 = await client.connect()
        if B2 == True:
            print("Connect[3C:A5:08:0A:B2:3C]successful!!")
        await client.start_notify(CHARACTERISTIC_UUID2, notification_handler2)
        await asyncio.sleep(20.0) 

def threads1():
    print("1111thread1")
    address1 = "3C:A5:08:0A:B2:3C"
    CHARACTERISTIC_UUID1 = "0000ffe1-0000-1000-8000-00805f9b34fb"
    loop = asyncio.new_event_loop()
    asyncio.set_event_loop(loop)
    loop.run_until_complete(connect_1(address1,CHARACTERISTIC_UUID1,loop))
    
def threads2():
    print("2222thread2")
    address2 = "3C:A5:19:0A:15:D2"
    CHARACTERISTIC_UUID2 = "0000be02-0000-1000-8000-00805f9b34fb"
    loop = asyncio.new_event_loop()
    asyncio.set_event_loop(loop)
    loop.run_until_complete(connect_2(address2,CHARACTERISTIC_UUID2,loop))
def main():
    threads_1 = threading.Thread(target=threads1)
    threads_2 = threading.Thread(target=threads2)
    #threads_2 = threading.Thread(target=thr)
    threads = [threads_1,threads_2]
    [ t.start() for t in threads ]
if __name__ == "__main__":
    main()
==============================================================
[result-display]
Device1
Device1
Device1
Device1
Device1
Device1
Device1
Device1
Device1
Device2
Device1
Device1
Device1
Device1

hbldh, 2020-10-21:
Check out the example with two devices for how to approach it. It is better without the `threading` module.

https://github.com/hbldh/bleak/blob/develop/examples/two_devices.py

lai-Ryan, 2020-10-21:
Hi hbldh~~
I would like to ask about this example[ two_device.py],
Are two BLE Device connect and two devices receiving data?

hbldh, 2020-10-22:
> Q1:But instead of two devices receiving data at the same time, only one device is receiving data at a time!!
how to two devices together recevier data??

In the example, both devices register for notifications and receive them. Try running it after modifying for your usecase.

> Q2:two device notify_uuid is different how to Modify the program??

There are many ways. Examples:

 - Two different `connect_to_device` methods with different hard-coded notification uuids.
- Send the desired notification uuid as input to `connect_to_device`

Experiment. Use the `two_devices.py` as a rough sketch and work your way forward.

closed 2021-03-03 by dlech