hbldh/bleak #350: Connecting to device with 6s advertising - error timeout

pawelm87, 2020-10-24:
* bleak version: 0.9.1
* Python version: 3.8.6
* Operating System: Win10
* BlueZ version (`bluetoothctl -v`) in case of Linux: -

### Description

I have device which send advertising frame ever 6s. When I used bleak in version 0.8.0 I can connected to it. When I updated bleak to the last version my simple script to connect this same device stop working. Some time I got the error - timeout. I found the place in code bleak where raise Exceptions when time 10s eclapse.

bleak/bleak/backends/dotnet/client.py
```
        # Start a GATT Session to connect
        event = asyncio.Event()
        self._connect_events.append(event)
        try:
            self._session = await wrap_IAsyncOperation(
                IAsyncOperation[GattSession](
                    GattSession.FromDeviceIdAsync(self._requester.BluetoothDeviceId)
                ),
                return_type=GattSession,
            )
            # This keeps the device connected until we dispose the session or
            # until we set MaintainConnection = False.
            self._session.MaintainConnection = True
            await asyncio.wait_for(event.wait(), timeout=10)
        except BaseException:
            handle_disconnect()
            raise
        finally:
            self._connect_events.remove(event)

        await self.get_services()

        return True
```
line 230 raise Exceptions after 10s.

### What I Did
```
Number of discover node attempts: 2
         {'addressRandom': 'c0:e9:61:61:3c:a2', 'addressPublic': '80e12607f099'}
Number of connection attempts: 0
Traceback (most recent call last):
  File "c:/Users/SmartBox/Desktop/bleak/bleak-test-update_FW/main.py", line 224, in <module>
    updateConf = loop.run_until_complete(run(address=randomMACData["addressRandom"],
  File "C:\Users\x\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "c:/Users/x/Desktop/bleak/bleak-test-update_FW/main.py", line 56, in run
    async with BleakClient(address, timeout = TIMEOUT_CONNECTION) as client:
  File "C:\Users\x\AppData\Local\Programs\Python\Python38\lib\site-packages\bleak\backends\client.py", line 60, in __aenter__
    await self.connect()
  File "C:\Users\x\AppData\Local\Programs\Python\Python38\lib\site-packages\bleak\backends\dotnet\client.py", line 236, in connect
    await asyncio.wait_for(event.wait(), timeout=10)
  File "C:\Users\x\AppData\Local\Programs\Python\Python38\lib\asyncio\tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
```

My simple scirpt:
```
import asyncio
from bleak import BleakClient

async def run(address: str, sdata: str):
    async with BleakClient(address, timeout = 20) as client:

    await client.write_gatt_char(char_specifier=uuid, data=sdata)

/some code/
loop = asyncio.get_event_loop()
loop.run_until_complete(run(address=validAddress, data=frameToSend))
```

hbldh, 2020-10-26:
This was odd. This should work better in 0.9.1 than in 0.8.0 imo...

Could you change the row
```python
await asyncio.wait_for(event.wait(), timeout=10)
```
to
```python
await asyncio.wait_for(event.wait(), timeout=self._timeout)
```
and then experiment with regulating the timeout in
```python
async with BleakClient(address, timeout = <increasing timeout value>) as client:
```

We do not use the provided timeout there, but maybe we should.

If that still does not work, try running the [`connect_by_bledevice.py`](https://github.com/hbldh/bleak/blob/develop/examples/connect_by_bledevice.py) example and make sure that you actually get a device.

pawelm87, 2021-03-08:
Hi, 

When I manually change timeout from (const value) 10 to self._timeout everything work fine.

In the new repository on github I see that timeout was connected to input value 'timeout' - not const value 10.0

line 240 - bleak/bleak/backends/dotnet/client.py
```python
            )
            # This keeps the device connected until we dispose the session or
            # until we set MaintainConnection = False.
            self._session.MaintainConnection = True
            await asyncio.wait_for(event.wait(), timeout=timeout)
        except BaseException:
            handle_disconnect()
            raise
        finally:
            self._connect_events.remove(event)```

but when I try update the library I still have timeout as const value 
`pip.exe install bleak --upgrade`

```python
            )
            # This keeps the device connected until we dispose the session or
            # until we set MaintainConnection = False.
            self._session.MaintainConnection = True
            await asyncio.wait_for(event.wait(), timeout=10)
        except BaseException:
            handle_disconnect()
            raise
        finally:
            self._connect_events.remove(event)
```

It means that the library was update but not release yet ?

When new version will be released ?

hbldh, 2021-03-16:
The default branch in Github is `develop`, on which we stage all changes that will eventually be released. The 10 is a bug and will be changed to `timeout` in next release.

A new release will be made when we are content with the work done on `develop` and when we find time enough to do the actual release procedure.

closed 2021-03-26 by pawelm87