hbldh/bleak #352: MacOS: objc.error Invalid Object

Tejas2805, 2020-10-26:
* bleak version:
* Python version: 3.8
* Operating System: MacOS
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Trying to connect to CC2650 SensorTag and read data from it. I have followed the code given but for some reason I am not being able to connect to it. I keep getting the "Invalid Object" error. For some I keep getting a lot of "objects" i.e. peripheral devices (in my opinion). Is there any way to fix it and get only one object? 

### What I Did

I ran discover.py to see the MAC address of the SensorTag and then copied that address into my SensorTag.py file. A part of the code of my SensorTag.py file has been given below to take a look at and guide me. The given portion is relevant to the error being caused.

```

async def run(address):
    print(address)
    async with BleakClient(address) as client:
        x = await client.is_connected()
        print("Connected: {0}".format(x))

        led_and_buzzer = LEDAndBuzzer()

        light_sensor = OpticalSensor()
        await light_sensor.start_listener(client)

        humidity_sensor = HumiditySensor()
        await humidity_sensor.start_listener(client)

        barometer_sensor = BarometerSensor()
        await barometer_sensor.start_listener(client)

        acc_sensor = AccelerometerSensorMovementSensorMPU9250()
        gyro_sensor = GyroscopeSensorMovementSensorMPU9250()
        magneto_sensor = MagnetometerSensorMovementSensorMPU9250()

        movement_sensor = MovementSensorMPU9250()
        movement_sensor.register(acc_sensor)
        movement_sensor.register(gyro_sensor)
        movement_sensor.register(magneto_sensor)
        await movement_sensor.start_listener(client)

        cntr = 0

        while True:
            # we don't want to exit the "with" block initiating the client object as the connection is disconnected
            # unless the object is stored
            await asyncio.sleep(1.0)

            if cntr == 0:
                # shine the red light
                await led_and_buzzer.notify(client, 0x01)

            if cntr == 5:
                # shine the green light
                await led_and_buzzer.notify(client, 0x02)

            cntr += 1

            if cntr == 10:
                cntr = 0

if __name__ == "__main__":
    """
    To find the address, once your sensor tag is blinking the green led after pressing the button, run the discover.py
    file which was provided as an example from bleak to identify the sensor tag device
    """

    import os

    os.environ["PYTHONASYNCIODEBUG"] = str(1)
    address = (
        "54:6c:0e:b5:56:00"
        if platform.system() != "Darwin"
        else "A15BB875-6116-4922-BF97-2464B32FB327"
    )
    print(address)
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run(address))
    loop.run_forever()

```

Screenshot of the error occurring: 

[image]

hbldh, 2020-11-25:
What macOS version do you have?
What version of Bleak do you use?

I am by no means sure this will work, since I do not have a Mac to test it on, but try to replace line 78-80 in `` with this:

```python
from Foundation import (
    NSUUID,
    NSArray,
)
peripherals = self._manager.central_manager.retrievePeripheralsWithIdentifiers_(
    NSArray([NSUUID(x.UUIDString()) for x in self._identifiers.keys()]),
)
```
It shouldn't be necessary imo, but it just might do the trick.

dlech, 2020-12-07:
A simpler fix for the same issue was reported to work in #331 but the issue was closed before a PR was made.

PR #380 (2020-12-07): Fixed ``BleakScanner.discover()`` on older versions of macOS.
Fixes #331
Fixes #352
commit 4591364c (2020-12-07): Fixed ``BleakScanner.discover()`` on older versions of macOS.

Fixes #331
Fixes #352
closed 2020-12-07 by dlech