hbldh/bleak #358: discover() is throwing a traceback error.

andrealmeida98, 2020-11-13:
* bleak version: 0.9.1
* Python version: 3.7.3
* Operating System: Raspberry pi 3B+ OS lite

### Description

Trying to find available BLE devices.

### What I Did

I executed your code : 

```
import asyncio
from bleak import discover

async def run():
    devices = await discover()
    for d in devices:
        print(d)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```

and got this error running `python3 ble.py`

```
Traceback (most recent call last):
  File "ble.py", line 10, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "ble.py", line 5, in run
    devices = await discover()
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/discovery.py", line 173, in discover
    ).asFuture(loop)
txdbus.error.RemoteError: org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 2 matched rules; type="method_call", sender=":1.24" (uid=1000 pid=1210 comm="python3 ble.py ") interface="org.freedesktop.DBus.ObjectManager" member="GetManagedObjects" error name="(unset)" requested_reply="0" destination="org.bluez" (uid=0 pid=442 comm="/usr/lib/bluetooth/bluetoothd ")
```

I really don't know the meaning of it, but I would appreciate some help (: 

Thank you in advance.

dhalbert, 2020-11-13:
There is an issue about which firmware to use on the RPi 3B+ and 4. See https://github.com/hbldh/bleak/issues/332 about rolling back the firmware. I got a different issue than you did, but you can try rolling it back and see what happens.

dhalbert, 2020-11-13:
Also, the access denied error may be due to your user not being in the `bluetooth` group. Add the user and reboot (and then you may get a different error which might be solved by the firmware fix).

andrealmeida98, 2020-11-13:
> Also, the access denied error may be due to your user not being in the `bluetooth` group. Add the user and reboot (and then you may get a different error which might be solved by the firmware fix).

It worked ! After adding the user to the bluetooth group, the code ran without problems. Thank you (:

closed 2020-11-13 by andrealmeida98