hbldh/bleak #360: Adapter parameter ignored on BleakClient

bmoore27, 2020-11-16:
* bleak version: 0.9.1
* Python version: 3.7.3
* Operating System: Debian 10.4
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description
Setting the adapter parameter seems to be ignored by BleakClient.

### What I Did
In my python script I am probing some GATT characteristics and I am trying to use the second BT adapter (hci1).

Running `hciconfig` I can see both adapters are enabled:
```
 $ hciconfighci 
hci1:   Type: Primary  Bus: USB
        BD Address: 34:C9:F0:89:73:4D  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:50925 acl:46 sco:0 events:1829 errors:0
        TX bytes:6246 acl:36 sco:0 commands:549 errors:0

hci0:   Type: Primary  Bus: UART
        BD Address: DC:A6:32:6B:88:2E  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:23070479 acl:13168 sco:0 events:579837 errors:0
        TX bytes:66202 acl:1868 sco:0 commands:2470 errors:0
```
In my python script, I initate the BleakClient like so:
```
async with BleakClient(address_or_ble_device=address, adapter="hci1") as client:
```

After the script runs I rerun the `hciconfig` command and I see that the TX/RX has only increased for `hic0`, see below:
```
 hciconfig
hci1:   Type: Primary  Bus: USB
        BD Address: 34:C9:F0:89:73:4D  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:50925 acl:46 sco:0 events:1829 errors:0
        TX bytes:6246 acl:36 sco:0 commands:549 errors:0

hci0:   Type: Primary  Bus: UART
        BD Address: DC:A6:32:6B:88:2E  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:23072805 acl:13213 sco:0 events:579873 errors:0
        TX bytes:66561 acl:1886 sco:0 commands:2481 errors:0
```

rhyst, 2020-11-20:
Also encountered this. Seems to be because the `connect` method uses the static `find_device_by_address` method which does not take the adapter as an argument.

You might be able to work around this by doing the scan yourself:

```
scanner = BleakScanner()
devices = await scanner.discover(device=adapter_name)
found = None
for device in devices:
    if (device.address == mac_address):
        found = device
        if not desk:
            print('not found')
            os._exit(1)
client = BleakClient(found, device=adapter_name)
```

You can initialise BleakScanner with the `device` kwarg but this is also seems to be ignored when you run `scanner.discover`.

hbldh, 2020-11-20:
This does indeed seem to be a bug. It should be a small thing to fix, will try to look at it during the weekend.

PR #383 (2020-12-11): Adapter kwarg fixes
Fixes #360 and fixes #381
PR #391 (2020-12-11): Release/v0.10.0
## Added

* Added ``AdvertisementData`` class used with detection callbacks across all
  supported platforms. Merged #334.
* Added ``BleakError`` raised during import on unsupported platforms.
* Added ``rssi`` parameter to ``BLEDevice`` constructor.
* Added ``detection_callback`` kwarg to ``BleakScanner`` constructor.

## Changed

* Updated minimum PyObjC version to 7.0.1.
* Consolidated implementation of ``BleakScanner.register_detection_callback()``.
  All platforms now take callback with ``BLEDevice`` and ``AdvertisementData``
  arguments.
* Consolidated ``BleakScanner.find_ [... cut]
closed 2020-12-11 by dlech