hbldh/bleak #365: Bleak Import Issue

yujie-tao, 2020-11-23:
* bleak version: 0.6.1
* Python version: 3.7.3
* Operating System: Mac Big Sur
* BlueZ version (`bluetoothctl -v`) in case of Linux:

Hi

I have CoreBluetooth installed on my laptop. No problem to run the from `CoreBluetooth import CBPeripheral` in python script. However, when I try to execute the `from bleak import BleakClient`, I get the "Bleak requires the CoreBluetooth Framework" exception. I recently upgraded my system to Big Sur; I had no problem to import BleakClient with the older system. Do anyone know why is that?

hbldh, 2020-11-24:
First of all, try to upgrade Bleak to 0.9.1 and try again.

Other than that, it looks like you need to turn your Bluetooth adapter on: https://support.apple.com/en-hk/guide/mac-help/blth1008/mac

yujie-tao, 2020-11-24:
Thanks for responding! When I switched to Bleak 0.9.1, it gave me error as `[1]    74763 abort      python3 test.py`. 

I do have the bluetooth on and also tried to switch it off and back on.

hbldh, 2020-11-24:
I am not sure this is Bleak throwing an exception...

How have you installed your Python? Via Homebrew? You might need to do a reinstall of Python in that case. This issue might help: https://github.com/Homebrew/homebrew-core/issues/44996.

yujie-tao, 2020-11-24:
Hi

I installed Python through the package installer on the python website. I tried to delete and reinstall the python but the error still remain.

I also tested on my friend's laptop (which is also running on Big Sur), the same error happened as well.

hbldh, 2020-11-24:
Seems to be a problem with `pyobjc` on Big Sur: https://github.com/ronaldoussoren/pyobjc/issues/309#issuecomment-648350096

Try installing the required `pyobjc` libraries from the `pyobjc-7-branch` branch.

yujie-tao, 2020-11-24:
I updated the PyObjC with the one under `pyobjc-7-branch`, sadly the issue still remains...

hbldh, 2020-11-24:
I am not able to help any further I am afraid, since I have not macOS to test on at the moment and for quite some time into the future. Hopefully someone else can shed som more light on the matter.

dhalbert, 2020-11-25:
I am running bleak successfully on Big Sur. My pip3 versions are:

bleak 0.9.1
pyobjc*  6.2

dlech, 2020-12-07:
A crash in PyObjC v7 was fixed in PyObjC v7.0.1. See #372.

alcpereira, 2021-05-02:
Have you tried #438 ?

yujie-tao, 2021-05-02:
wow #438 helped to solve it! Thank you @alcpereira, will close this issue.

closed 2021-05-02 by yujie-tao