hbldh/bleak #366: Code=13 "The value's length is invalid."

mgls, 2020-11-24:
* bleak version: 0.9.1
* Python version: 3.9
* Operating System: macOS 10.15.7 (Catalina)
* BlueZ version (`bluetoothctl -v`) in case of Linux: N/A

I am calling the `write_gatt_char` method below: 

`answer = await client.write_gatt_char(CHARACTERISTIC, bytes.fromhex(bt_command), response=True)`

In this case, `bt_command` is 346 characters long.

I'm expecting this to be a successful write call as I'm using the exact bt_command value sent directly from my bluetooth client and what others have used to successfully run this command using other bluetooth libraries.

However, I get the following traceback:

`Exception in callback PeripheralDelegate.did_write_value_for_characteristic(<CBPeripheral...e = connected>, <CBCharacteri...otifying = NO>, Error Domain=...h is invalid.})
handle: <Handle PeripheralDelegate.did_write_value_for_characteristic(<CBPeripheral...e = connected>, <CBCharacteri...otifying = NO>, Error Domain=...h is invalid.})>
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bleak/backends/corebluetooth/PeripheralDelegate.py", line 356, in did_write_value_for_characteristic
    raise BleakError(
bleak.exc.BleakError: Failed to write characteristic FFC1: Error Domain=CBATTErrorDomain Code=13 "The value's length is invalid." UserInfo={NSLocalizedDescription=The value's length is invalid.}`

In the same section of the code, I am able to successfully use the `bt_command` - however, the maximum character length that works for me is 190 characters long. Any bt_command at or below 190 characters executes successfully anything at or above 191 characters fails with this command.

As a workaround I've tried to split the unsuccessful (>190 character) bt_command into two commands and call them back-to-back but that has it's own effort and consequences.

Does the second parameter of the `write_gatt_char` have a character limit or do I need to do something differently? Or is this a macOS issues? Or something else?

Thanks!

hbldh, 2020-11-25:
This is all I know on this topic: https://developer.apple.com/forums/thread/43420

There is a limit imposed by device and MTU. Bleak cannot find it out. It is something that needs to be handled by the developer, specifically for each type of peripheral. Sorry.

mgls, 2020-11-25:
Thanks for confirming the limit. I've programmatically split the `bt_command` at 190 characters. 

Do you know if when I split the command into two strings, can I simply split the string into two parts (option 1 below) or do I need to ensure both parts have the proper prefix and suffix (option 2 below) ? 

So assuming P = Prefix, D = Data Payload, and S = Suffix

My original `bt_command` may have `PPPPDDDDDDDDDDDDSS` - of course more P and D characters - but for illustration purposes can my two new strings be:

**Option 1 - Split _WITHOUT ADDING_ Prefix/Suffix:**
string1 = **PPPP**DDDDDD
string2 = DDDDDD**SS**

**Option 2 - Split _ADDING_ Prefix/Suffix:**
string1 = **PPPP**DDDDDD**SS**
string2 = **PPPP**DDDDDD**SS**

I've tried it both ways without the expected results - so wanted to narrow down future attempts for debugging. 

Thanks!

hbldh, 2020-11-26:
I have no way of knowing that, I'm sorry. It all depends on how your peripheral accepts the data.

mgls, 2020-11-29:
> I have no way of knowing that, I'm sorry. It all depends on how your peripheral accepts the data.

Thanks @hbldh - I was able to programmatically split the `bt_command` before sending the `write` command. Essentially Option 1 in my last comment and that works in case this can help others in the future. 

I'm closing this out. Appreciate you building this library!

closed 2020-11-29 by mgls