hbldh/bleak #370: getting wrong UUID

dev-ale, 2020-11-29:
* bleak version: 0.91
* Python version: 3.73
* Operating System: Raspberry Pi OS
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.5

### Description

```python
await self.client.connect()
services = await self.client.get_services()
characteristics = services.get_service('00000000-347a-45b8-807c-455410d7e203').characteristics
Char_UUID = characteristics[0].uuid
```
we have a bluetooth device which provides a service with **only one** characteristic.
With the python code above we want to connect to the device and read all services. (BleakGATTServiceCollection)
From this collection we want to get a service by the UUID.
From this service we want to read the characteristics.
But we get back a wrong UUID.

We use an Adafruit Feather nRF52840

This problem has occurred since we reflashed the microcontroller.
The UUID we get is still the old one from the Arduino Code that was installed before on the device.

But when we read the UUID with the smartphone it is correct. That means it must be the Python code.

Arduino Code on **Adafruit Feather nRF52840 Sense**
```c
#include <bluefruit.h>
#include <Adafruit_BMP280.h>
#include <Adafruit_SHT31.h>
#include <avr/dtostrf.h>
#include <Adafruit_NeoPixel.h>
#include <Arduino_APDS9960.h>
uint8_t const brickliServiceUUID[]  = {0x03, 0xe2, 0xd7, 0x10, 0x54, 0x45, 0x7c, 0x80, 0xb8, 0x45, 0x7a, 0x34, 0x00, 0x00, 0x00, 0x00};
uint8_t const ledCharUUID[]     = {0x03, 0xe2, 0xd7, 0x10, 0x54, 0x45, 0x7c, 0x80, 0xb8, 0x45, 0x7a, 0x34, 0x02, 0x00, 0x00, 0x00};

Adafruit_NeoPixel pixel = Adafruit_NeoPixel(1, PIN_NEOPIXEL, NEO_GRB + NEO_KHZ800);

BLEService brickliService        = BLEService(brickliServiceUUID);
BLECharacteristic ledChar         = BLECharacteristic(ledCharUUID);

int r = 1;
int g = 1;
int b = 1;
uint8_t old[3] = {(uint8_t) r,(uint8_t) g,(uint8_t) b};
uint8_t packetbuffer[3];
void setup() {
  //create Neopixel
  pixel.begin();
  set_RGB(r,g,b);
  //create BLE 
  Bluefruit.begin();
  //set name
  Bluefruit.setName("0000-0002");
 //start service
  brickliService.begin();
  //setup Characterisic
  ledChar.setPermission(SECMODE_OPEN, SECMODE_OPEN);
  ledChar.setProperties(CHR_PROPS_READ | CHR_PROPS_WRITE);
  ledChar.setFixedLen(3);
  //add Characterisic to service
  ledChar.begin();
  startAdv();
}

void loop() {
  if ( Bluefruit.connected() ) {
    delay(1000);
    ledChar.read(packetbuffer, sizeof(packetbuffer));
    
    r = packetbuffer[0];
    g = packetbuffer[1];
    b = packetbuffer[2];
  
    if(old != packetbuffer){
        set_RGB(r,g,b);
        uint8_t r_ = (uint8_t) r;
        uint8_t g_ = (uint8_t) g;
        uint8_t b_ = (uint8_t) b;
        
        uint8_t ledstatusarray[3] = { r_, g_, b_ };
        //set Value of Characterisitcs
        ledChar.write(ledstatusarray, sizeof(ledstatusarray));//to unicode
    }
  }

}
void startAdv(void){
  // Advertising packet
  Bluefruit.Advertising.addFlags(BLE_GAP_ADV_FLAGS_LE_ONLY_GENERAL_DISC_MODE);
  Bluefruit.Advertising.addTxPower();

  //add serive to advertisment
  Bluefruit.Advertising.addService(brickliService);
 
  // Include Name
  Bluefruit.Advertising.addName();
  
  Bluefruit.Advertising.restartOnDisconnect(true);
  Bluefruit.Advertising.setInterval(32, 244);    // in unit of 0.625 ms
  Bluefruit.Advertising.setFastTimeout(30);      // number of seconds in fast mode
  Bluefruit.Advertising.start(0);                // 0 = Don't stop advertising after n seconds  
}

void set_RGB(int red, int green, int blue) {
  pixel.setPixelColor(0, pixel.Color(red,green,blue));
  pixel.show();
}

```

### What we Did

* Reflash the Microcontroller (same issue)
* Try it on a completly diffrent setup with raspi and microcontroller (same issue)
* Raspberry Pi reboot
* Pycache deleted
* Dependencies upgrade
* Read the BLE Device with a Smartphone and the correct UUID is shown.

hbldh, 2020-12-03:
Could you please run the `service_explorer.py` example and post the results here?

dev-ale, 2020-12-03:
`(env) ➜  examples git:(master) ✗ python3 service_explorer.py
Connected: True
[Service] 00000000-347a-45b8-807c-455410d7e203: Unknown
        [Characteristic] 00000011-347a-45b8-807c-455410d7e203: (Handle: 15) (read) | Name: HIDP, Value: b'\xbb\x8e\xb6' 
[Service] 00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile
        [Characteristic] 00002a05-0000-1000-8000-00805f9b34fb: (Handle: 11) (indicate) | Name: Service Changed, Value: None 
                [Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 13) | Value: b'\x02\x00'`

dlech, 2020-12-03:
When you change the structure of services/characteristics on a device, you have to remove the device from BlueZ so that it will read everything again. Otherwise BlueZ give the cached values from the first time the device was connected. You can use the `bluetoothctl` command line tool to do this.