hbldh/bleak #371: The scan result shows 'Unknown' name of the device

edward-jw-yu, 2020-11-29:
* bleak version: 0.9.1
* Python version: 3.7
* Operating System: win10 2004 19041

### Description
After I run the following code
`import asyncio
from bleak import BleakScanner

async def run():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())`
It shows 'A4:C1:38:01:21:FC: Unknown'
But I find the device in the system shows name 'SigMesh'

hbldh, 2020-12-03:
The Windows UWP Bluetooth API does not always provide a proper name for the device. I have explored a long time and the current state is the best that Bleak is capable of. Sorry.

dlech, 2020-12-07:
Does this device actually send the name in the advertising data? For example by setting "complete local name" data?

dlech, 2020-12-07:
Also see #175

pjbosco, 2021-02-21:
I proposed a possible solution on #175.

commit cad81adb (2021-03-01): Merge pull request #458 from pjbosco/fix-dotnet-unknown-name

Fix for the 'Unknown' name in scan results on Windows 10. 

Fixes #371
closed 2021-03-01 by hbldh