hbldh/bleak #372: NotImplementedError: _dyld_shared_cache_contains_path not available

takeru, 2020-11-29:
* bleak version: 0.9.1
* Python version: 3.8.5
* Operating System: macOS 10.15.6

### Description

NotImplementedError: _dyld_shared_cache_contains_path not available

### What I Did

search.py
```
import asyncio
from bleak import discover

async def run():
    devices = await discover()
    for d in devices:
        print(d)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```

run
```
python search.py
```

output
```
% python search.py
Traceback (most recent call last):
  File "/Users/takeru/.pyenv/versions/3.8.5-SwitchBot/lib/python3.8/site-packages/bleak/__init__.py", line 55, in <module>
    from CoreBluetooth import CBPeripheral
  File "/Users/takeru/.pyenv/versions/3.8.5-SwitchBot/lib/python3.8/site-packages/CoreBluetooth/__init__.py", line 10, in <module>
    import Foundation
  File "/Users/takeru/.pyenv/versions/3.8.5-SwitchBot/lib/python3.8/site-packages/Foundation/__init__.py", line 9, in <module>
    import CoreFoundation
  File "/Users/takeru/.pyenv/versions/3.8.5-SwitchBot/lib/python3.8/site-packages/CoreFoundation/__init__.py", line 16, in <module>
    objc.pathForFramework("/System/Library/Frameworks/CoreFoundation.framework"),
  File "/Users/takeru/.pyenv/versions/3.8.5-SwitchBot/lib/python3.8/site-packages/objc/_dyld.py", line 148, in pathForFramework
    fpath, name, version = infoForFramework(dyld_find(path))
  File "/Users/takeru/.pyenv/versions/3.8.5-SwitchBot/lib/python3.8/site-packages/objc/_dyld.py", line 142, in dyld_find
    return dyld_framework(filename, framework_name, version)
  File "/Users/takeru/.pyenv/versions/3.8.5-SwitchBot/lib/python3.8/site-packages/objc/_dyld.py", line 101, in dyld_framework
    if _dyld_shared_cache_contains_path(f):
NotImplementedError: _dyld_shared_cache_contains_path not available

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "search.py", line 2, in <module>
    from bleak import discover
  File "/Users/takeru/.pyenv/versions/3.8.5-SwitchBot/lib/python3.8/site-packages/bleak/__init__.py", line 57, in <module>
    raise BleakError("Bleak requires the CoreBluetooth Framework") from ex
bleak.exc.BleakError: Bleak requires the CoreBluetooth Framework
```

Change _dyld.py and run.

lib/python3.8/site-packages/objc/_dyld.p
```
#try:
#    from objc._objc import _dyld_shared_cache_contains_path
#except ImportError:
#    _dyld_shared_cache_contains_path = None
_dyld_shared_cache_contains_path = None
```

I can see device list. Works well.

Is it issue of pyobjc-core? or pyobjc-**?

```
% pip list
Package                        Version
------------------------------ -------
bleak                          0.9.1
pip                            20.1.1
pyobjc-core                    7.0
pyobjc-framework-Cocoa         7.0
pyobjc-framework-CoreBluetooth 7.0
pyobjc-framework-libdispatch   7.0
setuptools                     47.1.0
```

```
% pip list
Package                        Version
------------------------------ -------
bleak                          0.9.1
pip                            20.1.1
pyobjc-core                    6.2.2
pyobjc-framework-Cocoa         6.2.2
pyobjc-framework-CoreBluetooth 6.2.2
pyobjc-framework-libdispatch   6.2.2
setuptools                     47.1.0
```
OK,  works with this versions.

@ronaldoussoren

sn-uaudio, 2020-11-30:
I have the same issue with macOS 10.15.7 and python v3.8.6. Installing the pyobjc-* dependencies with v6.2.2 also fixes this for me.

ronaldoussoren, 2020-11-30:
This is a bug in PyObjC.  There will be a bug fix release soon, hopefully later tonight (CET).

hbldh, 2020-12-03:
Could someone using Bleak and getting it to work with the mentioned bugfix please mention it here so that the issue can be closed?

PR #379 (2020-12-07): CoreBluetooth: update PyObjC to 7.0.1
This avoids a crash on macOS <= 10.15 in PyObjC 7.0.0.

Fixes #372
commit d969bf76 (2020-12-07): CoreBluetooth: update PyObjC to 7.0.1

This avoids a crash on macOS <= 10.15 in PyObjC 7.0.0.

Fixes #372
closed 2020-12-07 by dlech