hbldh/bleak #373: pairing with encryption, no MITM protection / Just Works

ChrisHoS, 2020-11-30:
* bleak version: 0.9.1
* Python version: 3.7
* Operating System: Windows 10 Enterprise (Build 18363.1198) [64Bit]
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Wanted to pair to a slave with encryption level 2 / Just Works / LE Security Mode 1, Level 2

### What I Did
1. set up the server with Nordic nRF Connect and nRF52840 Dongle
[image]

2. connected, tried to pair to slave

Comment: 
- If encryption for the characteristic is disabled, everything works fine.
- On the final peripheral, it sends a security request what shall trigger a pairing request by the master / central device. 
  So, probably, [this method](https://docs.microsoft.com/en-us/uwp/api/windows.devices.enumeration.deviceinformationpairing.tryregisterforallinboundpairingrequests?view=winrt-19041) is needed to be exposed to Bleak.
  

I apologize for not providing a non-vendor specific example.

```
import asyncio
import uuid
import logging
from  bleak import BleakClient, discover

DATA_RX_CHARACTERISTIC_UUID = "EF680903-9B354-9339-B105-2FFA9740042"
address = "E8:40:C6:EF:5F:C7"

async def scan():
    devices = await discover()
    for d in devices:
        print(f"{d.name}: {d.address}: {d.rssi}dB")

async def run_ble(address):
    async with BleakClient(address) as client:
        await client.is_connected()
        await client.pair(protection_level=2)
        # hello world = 0x68 0x65 0x6c 0x6c 0x6f 0x20 0x77 0x6f 0x72 0x6c 0x64
        await client.write_gatt_char(uuid.UUID(DATA_RX_CHARACTERISTIC_UUID), str.encode("hello world"))

def run():
    loop = asyncio.get_event_loop()
    loop.run_until_complete(scan())
    loop.run_until_complete(run_ble(address))

if __name__ == "__main__":
    logging.basicConfig(level=logging.DEBUG)
    run()

DEBUG:bleak.backends.dotnet.scanner:Received E8:40:C6:EF:5F:C7: nRF Connect.
DEBUG:bleak.backends.dotnet.scanner:Received E8:40:C6:EF:5F:C7: Unknown.
DEBUG:bleak.backends.dotnet.client:Connecting to BLE device @ E8:40:C6:EF:5F:C7
DEBUG:bleak.backends.dotnet.client:_ConnectionStatusChanged_Handler: 1
DEBUG:bleak.backends.dotnet.client:Get Services...
INFO:bleak.backends.dotnet.client:Services resolved for BleakClientDotNet (E8:40:C6:EF:5F:C7)
DEBUG:bleak.backends.dotnet.client:Disconnecting from BLE device...
DEBUG:bleak.backends.dotnet.client:_ConnectionStatusChanged_Handler: 0
Traceback (most recent call last):
  File "C:/git/fire/asd/test_nrf52_dongle_server_with_bleak_client/src/bleak_client.py", line 32, in <module>
    run()
  File "C:/git/fire/asd/test_nrf52_dongle_server_with_bleak_client/src/bleak_client.py", line 27, in run
    loop.run_until_complete(run_ble(address))
  File "C:\Users\honeggec\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 587, in run_until_complete
    return future.result()
  File "C:/git/fire/asd/test_nrf52_dongle_server_with_bleak_client/src/bleak_client.py", line 19, in run_ble
    await client.pair(protection_level=2)
  File "C:\git\fire\asd\test_nrf52_dongle_server_with_bleak_client\.env\lib\site-packages\bleak\backends\dotnet\client.py", line 331, in pair
    "Cannot set minimally required protection level yet..."
NotImplementedError: Cannot set minimally required protection level yet...

```

hbldh, 2020-12-03:
Maybe that would do the trick. I was content with the `protection_level=1` as a start on Windows. It is not good enough, but it points the way for future expasion at least. I also had no device to test it with...

You are free to try to get it to work, since I will not have time and energy to do it for quite some time I am afraid.

patrick-dojofive, 2023-09-08:
Is the winnt backend restricted to protection_level 1 still?

EugeneNdungu, 2024-06-24:
i'd like to follow up on @patrick-dojofive question. I am attempting to pair with protection_level 2 but the logs I get indicate that the protection level used to pair is protection_level 1(i might also just be doing something wrong). I am working with:

* bleak version: 0.22.2
* Python version: 3.10.9
* Operating System: Windows 11

Here are the logs that I was able to collect( i apologize if the logs provided are not sufficient, it is what I was able to get for now but I am working on getting more logs)

`2024-06-24 14:41:14 - DEBUG - bleak.backends.winrt.client :: Connecting to BLE device @ CD:5D:68:45:A0:5F
2024-06-24 14:41:14 - DEBUG - bleak.backends.winrt.client :: getting services (service_cache_mode=None, cache_mode=None)...
2024-06-24 14:41:15 - DEBUG - bleak.backends.winrt.client :: max_pdu_size_changed_handler: 251
2024-06-24 14:41:15 - DEBUG - bleak.backends.winrt.client :: session_status_changed_event_handler: id: BluetoothLE#BluetoothLE48:89:e7:29:1c:f5-cd:5d:68:45:a0:5f, error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.ACTIVE: 1>
2024-06-24 14:41:15 - DEBUG - bleak.backends.winrt.client :: CD:5D:68:45:A0:5F: services changed
2024-06-24 14:41:16 - DEBUG - bleak.backends.winrt.client :: CD:5D:68:45:A0:5F: services changed
2024-06-24 14:41:17 - INFO - bluetooth_controller :: Connected: True
2024-06-24 14:41:17 - DEBUG - bluetooth_controller :: 123456
2024-06-24 14:41:19 - INFO - bleak.backends.winrt.client :: device id: 'BluetoothLE#BluetoothLE48:89:e7:29:1c:f5-cd:5d:68:45:a0:5f'
2024-06-24 14:41:19 - INFO - bleak.backends.winrt.client :: protection level: <DevicePairingProtectionLevel.ENCRYPTION: 2>
2024-06-24 14:41:20 - DEBUG - bleak.backends.winrt.client :: session_status_changed_event_handler: id: BluetoothLE#BluetoothLE48:89:e7:29:1c:f5-cd:5d:68:45:a0:5f, error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.CLOSED: 0>
2024-06-24 14:41:20 - DEBUG - bleak.backends.winrt.client :: max_pdu_size_changed_handler: 23
2024-06-24 14:41:20 - DEBUG - bleak.backends.winrt.client :: closing requester
2024-06-24 14:41:20 - DEBUG - bleak.backends.winrt.client :: closing session
2024-06-24 14:41:20 - INFO - bleak.backends.winrt.client :: Pairing result: 0, Protection level used: 1
2024-06-24 14:41:20 - INFO - bleak.backends.winrt.client :: Paired to device with protection level <DevicePairingProtectionLevel.NONE: 1>.`

EugeneNdungu, 2024-06-26:
After collecting some more logs, I identified that the issue was most likely due to a mismatch between the client's pairing request and the server's pairing response. Below are some key observations:

**Client's(PC) Pairing Request:**
- IO Capability: No Input, No Output (0x03)
- AuthReq: 0x29 (CT2 Flag, Secure Connection Flag, Bonding Flags: Bonding)
  - CT2 Flag: True
  - Secure Connection Flag: True
  - MITM Flag: False
  - Bonding Flags: Bonding (0x1)
- Key Distribution:
  - Initiator Key Distribution: Link Key, Signature Key (CSRK), Id Key (IRK)
  - Responder Key Distribution: Link Key, Signature Key (CSRK), Id Key (IRK), Encryption Key (LTK)

**Server's Pairing Response:**
- IO Capability: No Input, No Output (0x03)
- AuthReq: 0x00 (Bonding Flags: No Bonding)
  - CT2 Flag: False
  - Secure Connection Flag: False
  - MITM Flag: False
  - Bonding Flags: No Bonding (0x0)
- Key Distribution:
  - Initiator Key Distribution: None
  - Responder Key Distribution: None

We are looking into how the server firmware can be configured to match the client's pairing request requirements. 

As it is right now though, is there a way that one can use bleak to match the client's pairing requirements with those of the server as provided above?