hbldh/bleak #374: BleakClientBlueZDBus.start_notify() misses initial notifications with fast Bluetooth devices

scsims, 2020-12-02:
* bleak version: 0.9.1
* Python version: 3.7.3
* Operating System: Raspberry Pi OS (buster)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

When subscribing to notifications on a Bluetooth device that starts sending notifications quickly, sometimes the first 1 or more notifications are missed off and only the subsequent ones are passed to the callback registered in start_notify(). The device in question is transferring serial data and so attempts to begin sending immediately its descriptor is written and proceeds to send as quickly as it can.

### What I Did

The logs pasted below show the dbus-monitor output of the notification subscription showing all of the packets being sent to bleak. The second log shows bleak's debug logs for this same exchanges and it can be seen that the first 4 packets are received by BleakClientBlueZDBus._properties_changed_callback but they're not matched to the callback in the dict. Only after those 4 have been received, has the callback made it into the dict and then subsequent packets are matched.

The problem was resolved experimentally by modifying the Bluetooth device to have a delay between having its descriptor written to enable notifications and then it sending the first notification. Whilst this resolves this issue for this particular device, the problem obviously exists for other devices. The issue is caused because within BleakClientBlueZDBus.start_notify(), the dbus message is sent before start_notify's 'callback' argument is added to the BleakClientBlueZDBus._notification_callbacks dict. 

The solution was to move the call to self._bus.callRemote below the code that adds 'callback' to _notification_callbacks and _subscriptions so that they're populated before the first notification is received. I'll add a pull request containing these changes.

**dbus-monitor  --system "type='signal',sender='org.bluez'"**
```
signal time=1606906959.491162 sender=:1.15 -> destination=(null destination) serial=13570 path=/org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.GattCharacteristic1"
   array [
      dict entry(
         string "Notifying"
         variant             boolean true
      )
   ]
   array [
   ]
signal time=1606906959.492195 sender=:1.15 -> destination=(null destination) serial=13572 path=/org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.GattCharacteristic1"
   array [
      dict entry(
         string "Value"
         variant             array of bytes [
               08 70 00 b1 01 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00
            ]
      )
   ]
   array [
   ]
signal time=1606906959.502902 sender=:1.15 -> destination=(null destination) serial=13573 path=/org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.GattCharacteristic1"
   array [
      dict entry(
         string "Value"
         variant             array of bytes [
               00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
            ]
      )
   ]
   array [
   ]
signal time=1606906959.503119 sender=:1.15 -> destination=(null destination) serial=13574 path=/org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.GattCharacteristic1"
   array [
      dict entry(
         string "Value"
         variant             array of bytes [
               01 00 00 00 01 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00
            ]
      )
   ]
   array [
   ]
signal time=1606906959.503231 sender=:1.15 -> destination=(null destination) serial=13575 path=/org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.GattCharacteristic1"
   array [
      dict entry(
         string "Value"
[... 120 lines cut ...]
   ]
```

**bleak debug logging for the above notification exchange**
```
DEBUG:bleak.backends.bluezdbus.client:DBUS: path: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039, domain: org.bluez.GattCharacteristic1, body: {'Notifying': True}
DEBUG:bleak.backends.bluezdbus.client:DBUS: path: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039, domain: org.bluez.GattCharacteristic1, body: {'Value': [8, 112, 0, 177, 1, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0]}
DEBUG:bleak.backends.bluezdbus.client:DBUS: path: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039, domain: org.bluez.GattCharacteristic1, body: {'Value': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]}
DEBUG:bleak.backends.bluezdbus.client:DBUS: path: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039, domain: org.bluez.GattCharacteristic1, bod{'Value': [1, 0, 0, 0, 1, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]}
DEBUG:bleak.backends.bluezdbus.client:DBUS: path: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039, domain: org.bluez.GattCharacteristic1, body: {'Value': [130, 0, 0, 0, 141, 0, 0, 0, 12, 0, 0, 0, 37, 6, 4, 2, 135, 73, 0, 0]}
DEBUG:bleak.backends.bluezdbus.client:DBUS: path: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039, domain: org.bluez.GattCharacteristic1, body: {'Value': [0, 0, 0, 0, 2, 0, 0, 0, 1, 0, 0, 0, 1, 0, 0, 0, 0, 0, 176, 65]}
DEBUG:bleak.backends.bluezdbus.client:GATT Char Properties Changed: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039 | [{'Value': [0, 0, 0, 0, 2, 0, 0, 0, 1, 0, 0, 0, 1, 0, 0, 0, 0, 0, 176, 65]}, []]
DEBUG:bleak.backends.bluezdbus.client:DBUS: path: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039, domain: org.bluez.GattCharacteristic1, body: {'Value': [0, 0, 176, 65, 0, 0, 0, 0, 5, 0, 0, 0, 0, 25, 0, 0, 0, 0, 0, 0]}
DEBUG:bleak.backends.bluezdbus.client:GATT Char Properties Changed: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039 | [{'Value': [0, 0, 176, 65, 0, 0, 0, 0, 5, 0, 0, 0, 0, 25, 0, 0, 0, 0, 0, 0]}, []]
DEBUG:bleak.backends.bluezdbus.client:DBUS: path: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039, domain: org.bluez.GattCharacteristic1, body: {'Value': [1, 0, 0, 0, 41, 1, 17, 2, 3, 18, 32, 55, 1, 17, 2, 3, 18, 32, 0, 2]}
DEBUG:bleak.backends.bluezdbus.client:GATT Char Properties Changed: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039 | [{'Value': [1, 0, 0, 0, 41, 1, 17, 2, 3, 18, 32, 55, 1, 17, 2, 3, 18, 32, 0, 2]}, []]
DEBUG:bleak.backends.bluezdbus.client:DBUS: path: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039, domain: org.bluez.GattCharacteristic1, body: {'Value': [17, 2, 3, 18, 32, 73, 240, 246, 213, 179, 112, 0]}
DEBUG:bleak.backends.bluezdbus.client:GATT Char Properties Changed: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039 | [{'Value': [17, 2, 3, 18, 32, 73, 240, 246, 213, 179, 112, 0]}, []]
DEBUG:bleak.backends.bluezdbus.client:DBUS: path: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039, domain: org.bluez.GattCharacteristic1, body: {'Value': [0, 0, 220, 0]}
DEBUG:bleak.backends.bluezdbus.client:GATT Char Properties Changed: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039 | [{'Value': [0, 0, 220, 0]}, []]
DEBUG:bleak.backends.bluezdbus.client:DBUS: path: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039, domain: org.bluez.GattCharacteristic1, body: {'Value': [0, 255, 253, 155, 243, 116]}
DEBUG:bleak.backends.bluezdbus.client:GATT Char Properties Changed: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039 | [{'Value': [0, 255, 253, 155, 243, 116]}, []]
DEBUG:bleak.backends.bluezdbus.client:DBUS: path: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039, domain: org.bluez.GattCharacteristic1, body: {'Value': [1, 255, 157, 230, 59, 191, 52, 234, 174, 143, 28, 225, 218, 121, 253, 77, 99, 167]}
DEBUG:bleak.backends.bluezdbus.client:GATT Char Properties Changed: /org/bluez/hci0/dev_70_B3_D5_F6_F0_49/service001d/char0039 | [{'Value': [1, 255, 157, 230, 59, 191, 52, 234, 174, 143, 28, 225, 218, 121, 253, 77, 99, 167]}, []]
```

hbldh, 2020-12-03:
Interesting find, thank you! Will review this PR immediately!

scsims, 2020-12-03:
Tried it out on develop after the merge and it works great. Thanks!

PR #391 (2020-12-11): Release/v0.10.0
## Added

* Added ``AdvertisementData`` class used with detection callbacks across all
  supported platforms. Merged #334.
* Added ``BleakError`` raised during import on unsupported platforms.
* Added ``rssi`` parameter to ``BLEDevice`` constructor.
* Added ``detection_callback`` kwarg to ``BleakScanner`` constructor.

## Changed

* Updated minimum PyObjC version to 7.0.1.
* Consolidated implementation of ``BleakScanner.register_detection_callback()``.
  All platforms now take callback with ``BLEDevice`` and ``AdvertisementData``
  arguments.
* Consolidated ``BleakScanner.find_ [... cut]
closed 2020-12-03 by scsims