hbldh/bleak #377: connection bleak timeout error from central device

ethanHuntoo7, 2020-12-03:
* bleak version: 0.9.1
* Python version: 3.7
* Operating System: raspbian buster-lite
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50 

### Description

I was simply trying to connect a peripheral device with a central device using bleak nordic service

### What I Did
first I ran a peripheral device with an advertisement that can be seen in my android scanning ble app

 then in central deivce I just ran client code in the below link https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/master/examples/ble_uart_echo_client.py

it start scanning and then scanning stop after trying to make a connection with the bellow error

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/bleak/backends/bluezdbus/client.py", line 135, in connect
    ).asFuture(loop)
txdbus.error.RemoteError: org.freedesktop.DBus.Error.UnknownObject: Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "testClient.py", line 33, in <module>
    ble.connect(advertisement)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ble/__init__.py", line 305, in connect
    connection = self._adapter.connect(advertisement.address, timeout=timeout)
  File "/usr/local/lib/python3.7/dist-packages/_bleio/adapter_.py", line 373, in connect
    return self.await_bleak(self._connect_async(address, timeout=timeout))
  File "/usr/local/lib/python3.7/dist-packages/_bleio/adapter_.py", line 134, in await_bleak
    return future.result(timeout)
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 432, in result
    return self.__get_result()
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/usr/local/lib/python3.7/dist-packages/_bleio/adapter_.py", line 384, in _connect_async
    await client.connect(timeout=timeout)
  File "/usr/local/lib/python3.7/dist-packages/bleak/backends/bluezdbus/client.py", line 142, in connect
    self.address
bleak.exc.BleakError: Device with address b8:27:eb:96:30:96 could not be found. Try increasing `timeout` value or moving the device closer.
```

hbldh, 2020-12-03:
Have you tried the recommendation in the exception? To increase the `timeout`?

dlech, 2020-12-09:
I think I have found the problem. For some types of devices (something to do with random address maybe?), the device is removed from BlueZ as soon as scanning is stopped. So if you try to connect to that device after scanning is stopped, it will fail.

As a workaround, try connecting before stopping scanning.

dlech, 2020-12-09:
Also see #363

dlech, 2020-12-13:
It also looks like there is a new experimental BlueZ adapter API that could be for just such an issue: https://github.com/bluez/bluez/blob/98f5da02ce148a61efdfb2557152448f04bab51f/doc/adapter-api.txt#L170-L206

ethanHuntoo7, 2020-12-14:
> I think I have found the problem. For some types of devices (something to do with random address maybe?), the device is removed from BlueZ as soon as scanning is stopped. So if you try to connect to [...]

Right now all the did is restart the bluetooth service in peripheral device using **sudo systemctl restart bluetooth.service** and restart the service and its working. I don't need to reboot my device now but I'm still looking what exactly is the issue that causing this error.

ethanHuntoo7, 2020-12-16:
bleak version: 0.10.0
Python version: 3.7
Operating System: raspbian buster-lite

now I'm using bleak 0.10.0

and it is giving me a different error

Traceback (most recent call last):
  File "testClient.py", line 29, in <module>
    ble.connect(advertisement)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ble/__init__.py", line 305, in connect
    connection = self._adapter.connect(advertisement.address, timeout=timeout)
  File "/usr/local/lib/python3.7/dist-packages/_bleio/adapter_.py", line 373, in connect
    return self.await_bleak(self._connect_async(address, timeout=timeout))
  File "/usr/local/lib/python3.7/dist-packages/_bleio/adapter_.py", line 134, in await_bleak
    return future.result(timeout)
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 432, in result
    return self.__get_result()
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/usr/local/lib/python3.7/dist-packages/_bleio/adapter_.py", line 384, in _connect_async
    await client.connect(timeout=timeout)
  File "/usr/local/lib/python3.7/dist-packages/bleak/backends/bluezdbus/client.py", line 103, in connect
    "Device with address {0} was not found.".format(self.address)
**bleak.exc.BleakError: Device with address b8:27:eb:96:30:96 was not found**

dlech, 2020-12-16:
If you run `bluetoothctl` on the RPi in a separate terminal window what do you see when you run your Python script?  In my case, saw that the device is deleted (denoted by `[DEL]` as soon as scanning stops before the connection could be made.

dlech, 2020-12-26:
Also, please try the `dbus-next-2` branch. I just added something that could help with this.

    pip install --force-reinstall https://github.com/hbldh/bleak/archive/dbus-next-2.zip#egg=bleak

dlech, 2022-09-03:
closing since this error happened in the txdbus library that is no longer used

closed 2022-09-03 by dlech