hbldh/bleak #381: rename "device" to "adapter"

dlech, 2020-12-07:
* bleak version: develop
* Python version: N/A
* Operating System: N/A
* BlueZ version (`bluetoothctl -v`) in case of Linux: N/A

### Description

There are places in the code where we use a `device` keyword argument to specify the the Bluetooth adapter to use to connect to a Bluetooth device.

I find this rather confusing since the term "device" is used in most places to mean a remote Bluetooth device. The term "adapter" seems to be the standard nomenclature for the local Bluetooth radio.

It would be nice to rename this argument to be less confusing. We can still check for the old name too so that we don't break existing code.

dhalbert, 2020-12-07:
Here is some previous related discussion: https://github.com/hbldh/bleak/pull/285#discussion_r486288571

hbldh, 2020-12-07:
Yes, I agree. I do not know why I used `device` in the first place. I ought to have used `adapter`, as was used in e.g. `gatttool` which I used prior to this, but it seems I confused the two somehow while implementing.

The #360 is a current issue related to this. [This row](https://github.com/hbldh/bleak/blob/develop/bleak/backends/bluezdbus/scanner.py#L238) should be changed to
```python
scanner = cls(timeout=timeout, **kwargs)
```
to be able to connect with anything else than `hci0`. Had it been called adapter from the start, I don't think this would have slipped past...

commit c17ac3bd (2020-12-07): Renamed "device" kwarg to "adapter" in BleakClient and BleakScanner.

Fixes #381
PR #383 (2020-12-11): Adapter kwarg fixes
Fixes #360 and fixes #381
PR #391 (2020-12-11): Release/v0.10.0
## Added

* Added ``AdvertisementData`` class used with detection callbacks across all
  supported platforms. Merged #334.
* Added ``BleakError`` raised during import on unsupported platforms.
* Added ``rssi`` parameter to ``BLEDevice`` constructor.
* Added ``detection_callback`` kwarg to ``BleakScanner`` constructor.

## Changed

* Updated minimum PyObjC version to 7.0.1.
* Consolidated implementation of ``BleakScanner.register_detection_callback()``.
  All platforms now take callback with ``BLEDevice`` and ``AdvertisementData``
  arguments.
* Consolidated ``BleakScanner.find_ [... cut]
closed 2020-12-11 by dlech