hbldh/bleak #393: BlueZ battery workaround

dlech, 2020-12-13:
* bleak version: 0.10.0
* Python version: N/A
* Operating System: Linux
* BlueZ version (`bluetoothctl -v`) in case of Linux: >= 5.48

### Description

Currently, we have a workaround for the standard battery service in the BlueZ backend based on BlueZ version. 

https://github.com/hbldh/bleak/blob/e968a7b666b4ba76477e25c24887fa57afda0e19/bleak/backends/bluezdbus/client.py#L467-L482

However, I have just learned that there is a `-P battery` option to `bluetoothd` that can disable this feature. So the version check will actually break things for someone who has set this option.

So the logic should probably be changed so that rather than looking at the BlueZ version, just say: if we are looking for the battery service there is no matching D-Bus GATT service, then check if the D-Bus `org.bluez.Battery1` interface exists.

---

For the curious, you can disable the BlueZ battery plugin by running `sudo systemctl edit bluetooth.service` and adding the following:

```ini
[Service]
ExecStart=
ExecStart=-/usr/lib/bluetooth/bluetoothd -P battery
```

dlech, 2021-01-09:
Based on https://github.com/hbldh/bleak/issues/411#issuecomment-755783668, I'm starting to think that maybe we should just remove the battery service workaround instead and recommend that users disable the BlueZ battery plugin if they want to use devices with the battery service with Bleak. It means the OS will loose access to the battery service, but it will prevent BlueZ from interfering with Bleak.

hbldh, 2021-01-11:
I agree. The current Battery API does make the BlueZ backend behave different from the other OS backends. If it can be disabled, then it will produce a more coherent solution for all backends.

But it does require users to make non-trivial changes in the OS. Even though it is a simple fix, it does add a strange step to the installation procedure for Linux.

karlp, 2022-04-28:
I don't actually have a strong opinion on the battery service _itself_ but yes, I would say that bleak should make the battery service look "normal" cross platform, regardless of bluez version or the presence/lack thereof of the battery plugin.  My reasoning being that that leaves the _system_ interfaces "as is"  which to me is the major appeal of a dbus based BLE toolkit rather than a HCI based toolkit.

dlech, 2022-09-03:
Up until now, we missed https://github.com/bluez/bluez/commit/713f6f09f017abfccc2da9dbf2b32adf9ecb8e58 which exposes the battery service in BlueZ again. The version check for this is fixed in #976. So eventually, we can just remove the battery workaround completely when we drop support for BlueZ < 5.55.

natrajmuthusamy, 2023-01-05:
Thanks dlech...

dlech, 2023-03-17:
Closing since the workaround is no longer needed.

closed 2023-03-17 by dlech