hbldh/bleak #395: Bluez backend: Can't connect to device returned from detection callback

lchish, 2020-12-13:
* bleak version: 0.10.0
* Python version: 3.7.3
* Operating System: raspbian buster
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description
"path" isn't set in the device in the detection callback and crashes if you try and use the code in the following way:

```
import asyncio
from bleak import BleakScanner, BleakClient
from bleak.backends.device import BLEDevice
from bleak.backends.scanner import AdvertisementData
import logging

logging.basicConfig()

search_name = "Mo03F4000B"

class Test:
    def __init__(self):
        self.bt_client = None
        loop = asyncio.get_event_loop()
        loop.run_until_complete(self.run())
        
    def simple_callback(self, device: BLEDevice, advertisement_data: AdvertisementData):
        if search_name in device.name:
            print(f"Found {device}")
            self.bt_client = device
        
    async def run(self):
        scanner = BleakScanner()
        scanner.register_detection_callback(self.simple_callback)

        while self.bt_client is None:
            print(self.bt_client)
            await scanner.start()
            await asyncio.sleep(1.0)
            await scanner.stop()
        print(f"Connecting to {self.bt_client}")
        async with BleakClient(self.bt_client) as client:
            svcs = await client.get_services()
            print(f"{svcs}")

t = Test()
```
Output:
```
Found D5:D5:27:2F:0A:B2: Mo03F4000B000000000
Connecting to D5:D5:27:2F:0A:B2: Mo03F4000B000000000
Traceback (most recent call last):
  File "detection_callback.py", line 46, in <module>
    t = Test()
  File "detection_callback.py", line 25, in __init__
    loop.run_until_complete(self.run())
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "detection_callback.py", line 42, in run
    async with BleakClient(self.bt_client) as client:
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 58, in __init__
    self._device_path = address_or_ble_device.details["path"]
KeyError: 'path'
```

devbis, 2020-12-29:
Hey, 

I have faced a similar problem. After scanning for some (short) time on Linux the bleak raises an IndexError:

```
Unhandled Error
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/txdbus/protocol.py", line 140, in dataReceived
    self.rawDBusMessageReceived(raw_msg)
  File "/usr/lib/python3.9/site-packages/txdbus/protocol.py", line 266, in rawDBusMessageReceived
    self.signalReceived(m)
  File "/usr/lib/python3.9/site-packages/txdbus/client.py", line 642, in signalReceived
    self.router.routeMessage(msig)
  File "/usr/lib/python3.9/site-packages/txdbus/router.py", line 128, in routeMessage
    r.match(m)
--- <exception caught here> ---
  File "/usr/lib/python3.9/site-packages/txdbus/router.py", line 67, in match
    self.callback(m)
  File "/usr/lib/python3.9/site-packages/bleak/backends/bluezdbus/scanner.py", line 265, in parse_msg
    props["Address"], props["Alias"], props, props.get("RSSI", 0)
builtins.KeyError: 'Address'
```

Is it possible to check for correct props before fetching its content?

dlech, 2020-12-29:
This has been fixed in the develop branch already. Try this:

    pip install --force-reinstall https://github.com/hbldh/bleak/archive/develop.zip#egg=bleak

devbis, 2020-12-29:
It seems it is not fixed in the develop branch.
The only fix I could find is the commit from https://github.com/lchish/bleak/ and it is not merged yet.
Could you please point out the commit that fixes it?

dlech, 2020-12-29:
Hmm... I think I confused it with something else. @lchish would you like to make a pull request with your fix (and a changelog entry)?

devbis, 2020-12-29:
Looks like the commit doesn't fix the problem (I applied it on my instance) and it fails on the first `props["Address"]` access.
I'll try to investigate more to get more information on the `props` object.

dlech, 2020-12-29:
You could also try the `dbus-next-2` branch here. I may have already fixed it there too, but it is not well tested yet.

    pip install --force-reinstall https://github.com/hbldh/bleak/archive/dbus-next-2.zip#egg=bleak

devbis, 2020-12-29:
BTW, the props object for txdbus passes `{'ServicesResolved': True}` dict and it doesn't have Address and Alias keys.

Thanks for the hint, I'll definitely try the dbus-next branch

devbis, 2020-12-29:
I've tried dbus-next-2 branch and it raises the similar issue (a lot of exceptions) on the line 271:
https://github.com/hbldh/bleak/blob/dbus-next-2/bleak/backends/bluezdbus/scanner.py#L271

```
ERROR:root:A message handler raised an exception: 'Address'.
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/dbus_next/message_bus.py", line 656, in _process_message
    result = handler(msg)
  File "/usr/lib/python3.9/site-packages/bleak/backends/bluezdbus/scanner.py", line 298, in _parse_msg
    self._invoke_callback(msg_path, message)
  File "/usr/lib/python3.9/site-packages/bleak/backends/bluezdbus/scanner.py", line 271, in _invoke_callback
    props["Address"],
KeyError: 'Address'
```

dlech, 2020-12-29:
Can you enable debug logging with the `dbus-next-2` branch so that we can see what D-Bus signals are being received?

devbis, 2020-12-29:
I'm a little confused to understand DBUS logs.
So I copied the full dump of my program with debug enabled for bleak.

https://pastebin.com/zmkKhWgq

Let me know can I do anything else with this issue.

dlech, 2020-12-29:
Did the crash occur again after the last line of that log?

dlech, 2020-12-29:
Nevermind, I see it in the middle now.

dlech, 2020-12-29:
I just pushed another change to the `dbus-next-2` branch that might fix this.

devbis, 2020-12-29:
I tried bleak with your updates, but the issue still exists, unfortunately.
Here are new logs from my app:

https://pastebin.com/aYd14LpA

dlech, 2020-12-30:
I think I see what is going on now. I think the scanner is stopping and restarting while a device is connected. This can trigger a race condition where a "PropertiesChanged" signal is received before "GetManagedObjects" returns in the scanner start method. I've added a fix for this in the `dbus-next-2` branch.

devbis, 2020-12-30:
Looks like that problem was fixed, for 5 minutes I haven't seen any of those.
Thank you very much for fixing it!

But another problem occurred in the connect method after some time.

```
  File "/usr/lib/python3.9/site-packages/bleak/backends/bluezdbus/client.py", line 273, in connect
    assert reply.message_type == MessageType.METHOD_RETURN
AssertionError
```
https://pastebin.com/cXKPkHPE

Should I create another issue for it?

And one more KeyError during scanning:

```
  File "/usr/lib/python3.9/site-packages/dbus_next/message_bus.py", line 656, in _process_message
    result = handler(msg)
  File "/usr/lib/python3.9/site-packages/bleak/backends/bluezdbus/scanner.py", line 290, in _parse_msg
    del self._devices[obj_path]
KeyError: '/org/bluez/hci0/dev_18_D8_91_DE_BE_C4'
```

https://pastebin.com/Eg6m7ydz

dlech, 2020-12-30:
Better error checking instead of `assert reply.message_type == MessageType.METHOD_RETURN` is on my TODO list before merging that branch. But yes, let's open a new issue for it anyway. It is helpful to know what errors are possible so we know how to handle them.

dlech, 2020-12-30:
And please open a separate issue for the `KeyError`.

devbis, 2020-12-30:
Done. Thank you for your work and fixes.

dlech, 2021-05-11:
Seems like this issue has been fixed.

closed 2021-05-11 by dlech