hbldh/bleak #396: Constant Timeout in MacOS

cassioiks, 2020-12-14:
* bleak version: bleak==0.10.0
* Python version: Python 3.7.7
* Operating System: MacOS 10.15.7

### Description

I'm trying to run the [example to print all services](https://github.com/hbldh/bleak/blob/develop/examples/get_services.py) from a BLE device, and I get a timeout every time. I'm trying to connect specifically to a Philips Hue Bulb, but it happens to all BLE devices in range.
It doesn't look like something specific to bleak, as I also tried with the app LightBlue and I get timeouts as well.

NB: I do get a `device` object back from `BleakScanner.find_device_by_address`.

### What I Did

```
"""
Connect by BLEDevice
"""

import asyncio
import platform

from bleak import BleakClient, BleakScanner

async def print_services(mac_addr: str):
    device = await BleakScanner.find_device_by_address(mac_addr)
    async with BleakClient(device) as client:
        svcs = await client.get_services()
        print("Services:", svcs)

loop = asyncio.get_event_loop()
loop.run_until_complete(print_services("161420B4-1DD2-4C0F-9972-32A29357D1D5"))
```

```
Traceback (most recent call last):
  File "/Users/user/git/testbleak/connect.py", line 24, in <module>
    loop.run_until_complete(print_services("161420B4-1DD2-4C0F-9972-32A29357D1D5"))
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 587, in run_until_complete
    return future.result()
  File "/Users/user/git/testbleak/connect.py", line 13, in print_services
    async with BleakClient(device) as client:
  File "/usr/local/lib/python3.7/site-packages/bleak/backends/client.py", line 59, in __aenter__
    await self.connect()
  File "/usr/local/lib/python3.7/site-packages/bleak/backends/corebluetooth/client.py", line 90, in connect
    await manager.connect_(self._device_info, timeout=timeout)
  File "/usr/local/lib/python3.7/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 152, in connect_
    self._connection_state_changed.wait(), timeout=timeout
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/tasks.py", line 449, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```

olorton, 2021-06-22:
-    bleak version: bleak==0.12.0
-    Python version: Python 3.7.9
-    Operating System: macOS 10.14.6

I have been suffering with this exact issue for some time now. In particular, with the line: `async with BleakClient(device) as client:` which times out.

However, this is possibly more interesting in my case because I have no problem with my 2019 MacBook Pro (Mojave), but my Hackintosh (also Mojave) with a Bluetooth dongle does not work 99% of the time. I _have_ got it to work at least once, so there appears to be an intermittent error of some kind. I was initially using v0.10 of bleak, but have also tested this on the latest v0.12 just now. Using the 4.0 dongle to connect to other devices such as a speaker and iPhone works fine.

Since I have a working macOS computer and non-working/occasionally-working macOS computer, I would be happy to run any tests which may help get to the bottom of this.

See this output showing 11 successive errors followed by 1 success (line 191): https://gist.github.com/olorton/91e0d779ffe72b23dbf123e7a4a7050a

dlech, 2021-06-22:
Have you tried increasing the timeout? What is the output if you enable logging?

olorton, 2021-06-23:
Increasing the timeout from the default of 10s to 30s did not yield an improvement. Here is the debug log:

    2021-06-23 09:51:06,448 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManagerDidUpdateState_
    2021-06-23 09:51:06,448 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Bluetooth powered on
    2021-06-23 09:51:06,449 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: 'isScanning' changed
    2021-06-23 09:51:06,495 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
    2021-06-23 09:51:06,495 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device C6594CE4-2070-4387-BA9D-B1397DB53FC4: None @ RSSI: -88 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fb853c17fb0>
    2021-06-23 09:51:06,498 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
    2021-06-23 09:51:06,498 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 208FFF79-B315-4CC9-B842-95DC854B4D6C: STLB240 @ RSSI: -87 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fb853c17fb0>
    2021-06-23 09:51:06,498 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: 'isScanning' changed
    2021-06-23 09:51:06,499 bleak.backends.corebluetooth.client DEBUG: CentralManagerDelegate  at <CentralManagerDelegate: 0x7fb853c6e9b0>
    2021-06-23 09:51:06,499 bleak.backends.corebluetooth.client DEBUG: Connecting to BLE device @ 208FFF79-B315-4CC9-B842-95DC854B4D6C
    2021-06-23 09:51:36,501 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Connection timed out after 30 seconds.
    2021-06-23 09:51:36,520 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDisconnectPeripheral_error_
    2021-06-23 09:51:36,520 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Peripheral Device disconnected!
    Traceback (most recent call last):
      File "./get_temperature_data.py", line 11, in <module>
        loop.run_until_complete(oxhome.run(address))
      File "/Users/oliver/.pyenv/versions/3.7.9/lib/python3.7/asyncio/base_events.py", line 587, in run_until_complete
        return future.result()
      File "/Users/oliver/Code/oxhome/oxhome.py", line 104, in run
        async with BleakClient(address, timeout=30) as client:
      File "/Users/oliver/.local/share/virtualenvs/oxhome-UL5lUEBP/lib/python3.7/site-packages/bleak/backends/client.py", line 61, in __aenter__
        await self.connect()
      File "/Users/oliver/.local/share/virtualenvs/oxhome-UL5lUEBP/lib/python3.7/site-packages/bleak/backends/corebluetooth/client.py", line 118, in connect
        await manager.connect(self._peripheral, disconnect_callback, timeout=timeout)
      File "/Users/oliver/.local/share/virtualenvs/oxhome-UL5lUEBP/lib/python3.7/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 157, in connect
        await asyncio.wait_for(future, timeout=timeout)
      File "/Users/oliver/.pyenv/versions/3.7.9/lib/python3.7/asyncio/tasks.py", line 449, in wait_for
        raise futures.TimeoutError()
    concurrent.futures._base.TimeoutError

NlsMyFox, 2021-06-25:
Hello @dlech,

I am facing the same issue than @olorton with a 2020 MacBook PRO with MacOS 11.4, Bleak 0.12.0 and Python 3.9.
I have also tried to increase the timeout to 30s without any effect.
The issue is exactly the same, the lib tries to connect to my devices and exit after tiemout with "Timeout error" exception.

Thanks by advance for your help.
KR

dlech, 2021-06-25:
Do these devices work with other BLE apps on the same Mac? Do the devices use random addresses that could be changing?

NlsMyFox, 2021-06-25:
My devices have fixed address and works with other BLE Apps. Moreover, the current code is working well on Linux machine. Seems to be link to the CoreBluetooth backend...

dlech, 2021-06-25:
Just to clarify, the other BLE apps are on the same Mac? So we can conclude that the problem is with Bleak and not CoreBluetooth in general or that specific computer?

NlsMyFox, 2021-06-25:
Yes it is, devices can be discovered and connected by BlueLight.

dlech, 2021-06-25:
The next step would be to [capture Bluetooth packets](https://bleak.readthedocs.io/en/latest/troubleshooting.html#macos) to see what Bleak is doing different from BlueLight.

hbldh, 2021-06-28:
When I was trying to connect to a Philips Hue lightbulb (see [example]()) I had to do Resets in the Philips Hue iOS app before connecting to it again, and it was slower to connect to (in Windows) than most other devices. It would lock up in between connects if I connected to the bulb with any other device than my computer and Bleak. 

It was a finicky device that I never really understood fully, but that is an experience I have had with a lot of BLE devices...

buyuk-dev, 2021-08-12:
I have experienced similar issue, while attempting to connect Muse EEG to my macbook pro 2015 running Big Sur, python3 and latest develop branch bleak package. I also tried recreating the issue with just CoreBluetooth package with PyObjC and it seems like in my case the issue lies deeper than bleak.

Here's a snippet of code that recreated the issue for me.
https://github.com/buyuk-dev/playground/blob/master/corebluetooth/test.py

Additionally, I have collected traces with packet logger, which I don't yet have proper understanding of as I'm new to all of this, but there are two interesting entries in there. Several times there is an entry "HCI command disallowed", and "Not in UHE mode". I am looking into it, but its difficult to find more info. If anyone has ideas on debugging approaches I'd appreciate it.

yunnong770, 2021-09-03:
What's the status on this issue? I am running into the exact same problem with windows. Couldn't run the example code to get services of the device.