hbldh/bleak #398: Unable to pair prior connect

5frank, 2020-12-19:
* bleak version: 0.10.0 and master branch e968a7b666b4ba76477e25c24887fa57afda0e19
* Python version: 3.7.3
* Operating System: Linux-4.19.0-13-amd64-x86_64-with-debian-10.7
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50
* BLE device address scheme: Private Random Resolvable address

### Description
Unable to pair (or trust) prior connect and can not connect before pair. 

### What I Did

As the D-bus resources `ClientBlueZDBus:{_reactor, _bus}` is only initialized in the `connect` method I added  check and initialization to both `connect` and `pair`:
```python
        if self._reactor is None:
            self._reactor = get_reactor(loop)

        if self._bus is None:
            self._bus = await txdbus_connect(self._reactor, busAddress="system").asFuture(
                loop
            )
```
However this did not resolve my issue. The only way I can successful pair is with following sequence:
 - remote call "StartDiscovery"
- set property "Trusted" = True
 - remote call "Pair"
- wait on property "Connected" to be True. This can take several seconds even though property "Paired" is True. Also, the state change only seem to occur if Discovery enabled/started.
- remote call "StopDiscovery"

Trying to call `BleakClientBlueZDBus:connect` before or after the above gives the following error:
 `txdbus.error.RemoteError: org.freedesktop.DBus.Error.UnknownObject: Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist`. 

Questions:
- Either something behaves different with my setup or pairing have not been working for some time? 
  If paring works with another setup it would be interesting to know?

- What is the expected/intended behavior? Should it be possible to call `connect` after a successful call to `pair`?

- I am tempted to suggest that pairing should be handled when  connecting. (i.e. as in MacOS). Not sure
  it is good idea or how the implementation details would look like. Thoughts?

Somewhat related but closed issues:
- https://github.com/hbldh/bleak/issues/309
- https://github.com/hbldh/bleak/issues/363

dlech, 2020-12-26:
Please try the `dbus-next-2` branch. I just added something that could help with `txdbus.error.RemoteError: org.freedesktop.DBus.Error.UnknownObject: Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist`.

    pip install --force-reinstall https://github.com/hbldh/bleak/archive/dbus-next-2.zip#egg=bleak

The workaround requires BlueZ >= 5.49 and experimental features enabled. (`bluetoothd` has to be run with the `--experimental` option).

    sudo systemctl edit bluetooth.service

Edit the file to look like this:

```ini
[Service]
ExecStart=
ExecStart=-/usr/lib/bluetooth/bluetoothd --experimental
```

hbldh, 2020-12-30:
This is interesting! Due to not finding a way to pair before connecting in (as I recall, I might be wrong...) BlueZ, I decided to make the Bleak API such that connection is required before attempting pairing. In retrospect, I think an added utility method for pairing before might be needed. I am not sure where though: either as a separate import (in e.g. `bleak.utils`) or as a classmethod on the BleakClient. The latter is most accessible, but I am not sure having it there is self-explanatory enough to just add.

lamachine, 2021-02-24:
Have not tried this solution yet, but on Linux (latest raspian) I am getting the 

```
"Device with address {0} was not found.".format(self.address)
bleak.exc.BleakError: Device with address CD:19:4A:C4:C2:BC was not found.
```

First I scan for all the devices, and find 4 including the one of interest

I tried using bluetoothctl, and I can see it appear with [NEW], then I get [CHG] RSSI is nil even though I moved the device right next to the Pi, then [DEL]

My program then comes back with the error on every device I try.  Should this be a separate thread/issue or is it associated with this one?

5frank, 2021-02-26:
@lamachine hard to tell if it is a related issue. what is the address schema? Any clues from `btmon` output? I belive the address might change after pairing for some devices.

dlech, 2022-09-03:
Closing as duplicate of #309.

closed 2022-09-03 by dlech