hbldh/bleak #399: Device with address {0} was not found.".format(self.address)

sandhu0409, 2020-12-21:
* bleak version:0.10.0
* Python version:3.7
* Operating System:Raspbian burster lite
* BlueZ version: 5.50

### Description

Device with address {0} was not found.".format(self.address)

### What I Did
I tried to connect a peripheral device using central device with code from:
https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/master/examples/ble_uart_echo_client.py

but its showing Bleak error

Device with address {0} was not found.".format(self.address)
bleak.exc.BleakError: Device with address b8:27:eb:9b:d5:fa was not found.

```
Traceback (most recent call last):
  File "central.py", line 33, in <module>
    ble.connect(advertisement)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ble/__init__.py", line 305, in connect
    connection = self._adapter.connect(advertisement.address, timeout=timeout)
  File "/usr/local/lib/python3.7/dist-packages/_bleio/adapter_.py", line 373, in connect
    return self.await_bleak(self._connect_async(address, timeout=timeout))
  File "/usr/local/lib/python3.7/dist-packages/_bleio/adapter_.py", line 134, in await_bleak
    return future.result(timeout)
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 432, in result
    return self.__get_result()
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/usr/local/lib/python3.7/dist-packages/_bleio/adapter_.py", line 384, in _connect_async
    await client.connect(timeout=timeout)
  File "/usr/local/lib/python3.7/dist-packages/bleak/backends/bluezdbus/client.py", line 103, in connect
    "Device with address {0} was not found.".format(self.address)
bleak.exc.BleakError: Device with address b8:27:eb:9b:d5:fa was not found.
```

dlech, 2020-12-21:
Also see #377

sandhu0409, 2020-12-22:
> Also see #377

No solution there.
In bleak 0.9.1: 
I got timeout error
And in bleak 0.10.1:
I got no address found

Same as 377 issue

dlech, 2020-12-22:
Did you try the suggestion of monitoring `bluetoothctl` as suggested in https://github.com/hbldh/bleak/issues/377#issuecomment-746538356 ?

dlech, 2020-12-26:
Also, please try the `dbus-next-2` branch. I just added something that could help with this.

    pip install --force-reinstall https://github.com/hbldh/bleak/archive/dbus-next-2.zip#egg=bleak

inonoob, 2021-01-15:
hey, 

I also got this issue. 

    bleak version:0.10.0
    Python version:3.7
    Operating System:Raspbian burster lite
    BlueZ version: 5.50

What have I done: 
- See #398 (comment) for more info ==> installed dbus-next-2 version ==> no fix 
- I try to check `bluetoothctl` . I can see the scanning but after I see the [del] command. 

does anyone have news on this ? 

Best Regards

dlech, 2021-01-15:
did you also enable experimental features in BlueZ and restart BlueZ?

inonoob, 2021-01-15:
Yep I did. 
I 'm going to do a fresh install of the Raspberry pi and install the dbus-next-2 version from the beginning. Let's see if that helps. 
Will keep you informed.

inonoob, 2021-01-15:
So it is working. 
- I did install a fresh Raspbian.
- Installed `sudo apt-get install python3-pip git -y`
- added the $USER to the bluetooth group  => `sudo usermod -a -G bluetooth $USER`
- added the `ExecStart=-/usr/lib/bluetooth/bluetoothd -- experimental` to the file : `nano /etc/systemd/system/bluetooth.target.wants/bluetooth.service` 
- did `sudo systemctl daemon-reload` 
- did `sudo systemctl restart bluetooth.service`
- Tried the discover_service.py` to test it and it's working. 

I have no idea why it is now working ?

closed 2021-10-07 by dlech