hbldh/bleak #404: Installation Error on MacOS Catalina 10.15.7 Python 3.6.4

emadul-credosense, 2020-12-29:
* bleak version: 0.10.0
* Python version: 3.6.4
* Operating System: MacOS Catalina version:10.15.7
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description
I have tried to install in mac using pycharm ide and got this error

ERROR: Command errored out with exit status 1:
     command: /Users/hackintosh2/PycharmProjects/soil-respiration/venv_mac/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/ff/x0l2h8797bv_nnm18s1tpp180000gn/T/pip-install-0nyabc6y/pyobjc-framework-cocoa_629ad390a03b43528eb5d5c4a7afbf14/setup.py'"'"'; __file__='"'"'/private/var/folders/ff/x0l2h8797bv_nnm18s1tpp180000gn/T/pip-install-0nyabc6y/pyobjc-framework-cocoa_629ad390a03b43528eb5d5c4a7afbf14/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/ff/x0l2h8797bv_nnm18s1tpp180000gn/T/pip-pip-egg-info-n0qiwqxx
         cwd: /private/var/folders/ff/x0l2h8797bv_nnm18s1tpp180000gn/T/pip-install-0nyabc6y/pyobjc-framework-cocoa_629ad390a03b43528eb5d5c4a7afbf14/
    Complete output (7 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/ff/x0l2h8797bv_nnm18s1tpp180000gn/T/pip-install-0nyabc6y/pyobjc-framework-cocoa_629ad390a03b43528eb5d5c4a7afbf14/setup.py", line 26, in <module>
        extra_link_args=["-framework", "CoreFoundation"],
      File "/private/var/folders/ff/x0l2h8797bv_nnm18s1tpp180000gn/T/pip-install-0nyabc6y/pyobjc-framework-cocoa_629ad390a03b43528eb5d5c4a7afbf14/pyobjc_setup.py", line 434, in Extension
        % (tuple(map(int, os.path.basename(data)[6:-4].split("."))))
    ValueError: invalid literal for int() with base 10: ''
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

dlech, 2020-12-29:
It looks like the error is in PyObjC rather than Bleak, so I would suggest checking to see if this is a known issue with PyObjC.

https://github.com/ronaldoussoren/pyobjc/issues

emadul-credosense, 2020-12-30:
installation problem solved after upgrading my python version to 3.9.

emadul-credosense, 2020-12-30:
I cannot find my ble device mac address, so that I can connect it
### Code:
import asyncio
from bleak import BleakScanner

async def run():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())

### Output:

F71D2415-CEEF-40B2-B5DA-B85660090673: Credosense SRC-5
08AEB3B9-C463-44AC-A9E9-69BC0A886BDD: Microsoft (b'\x01\t \x02\x8c\x8b\xc9u4\xb4c\xcb-\x8e\xb7\x9e\x94\x9d@NBzN \x8cgq')
AB6A065F-052E-452E-990D-06BBDD2D4E2E: [TV] Samsung 5 Series (43)
F0A0C60D-AEB0-4A85-92D3-2C1BF36872BE: Microsoft (b'\x01\t \x02x\x83\x97\xff\x13\xd9k\x9c\x7fz\xdeF%\x9ah)\x9b#\xfct|D6')
9B4A8188-8BC0-4ED2-8261-8E9CA95FDEF7: Samsung Electronics Co. Ltd. (b'B\x04\x01\x80\x8e\xbc\x14\x85\xc8R\x1a&K\x03\xcfac\x01\x00\x00\x00\x00\x00\x00')

dlech, 2020-12-30:
Apple doesn't provide access to Bluetooth addresses for security/privacy reasons. You will need a different way to identify your device. For example:

```python
if (d.name == 'Credosense SRC-5'):
    with BleakClient(d) as client:
        ...
```

emadul-credosense, 2020-12-31:
> Apple doesn't provide access to Bluetooth addresses for security/privacy reasons. You will need a different way to identify your device. For example:
> 
> ```python
> if (d.name == 'Credosense SRC-5'):
>     with BleakClient(d) as client:
>         ...
> ```

I have tried to run it but getting attribute error. How can i connect my bluetooth device from mac?

### Code: 

import asyncio
from bleak import BleakScanner, BleakClient

async def run():
    devices = await BleakScanner.discover()
    for d in devices:
        if d.name == 'Credosense SRC-5':
            with BleakClient(d) as client:
                print(client)

### Output:

Traceback (most recent call last):
  File "/Users/hackintosh2/PycharmProjects/test/main.py", line 13, in <module>
    loop.run_until_complete(run())
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/Users/hackintosh2/PycharmProjects/test/main.py", line 8, in run
    with BleakClient(d) as client:
AttributeError: __enter__

dlech, 2020-12-31:
I forgot, it needs to be `async with BleakClient(d) as client:`

emadul-credosense, 2020-12-31:
> I forgot, it needs to be `async with BleakClient(d) as client:`.
Still i could not connect it 
### Code:
import asyncio
from bleak import BleakScanner, BleakClient

async def run():
    devices = await BleakScanner.discover()
    for d in devices:
        if d.name == 'Credosense SRC-5':
            async with BleakClient(d) as client:
                print("done")

loop = asyncio.get_event_loop()
loop.run_until_complete(run())

### Output:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/locks.py", line 226, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/tasks.py", line 487, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/hackintosh2/PycharmProjects/test/main.py", line 14, in <module>
    loop.run_until_complete(run())
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/Users/hackintosh2/PycharmProjects/test/main.py", line 8, in run
    async with BleakClient(d) as client:
  File "/Users/hackintosh2/PycharmProjects/test/venv_mac/lib/python3.9/site-packages/bleak/backends/client.py", line 59, in __aenter__
    await self.connect()
  File "/Users/hackintosh2/PycharmProjects/test/venv_mac/lib/python3.9/site-packages/bleak/backends/corebluetooth/client.py", line 90, in connect
    await manager.connect_(self._device_info, timeout=timeout)
  File "/Users/hackintosh2/PycharmProjects/test/venv_mac/lib/python3.9/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 151, in connect_
    await asyncio.wait_for(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/tasks.py", line 489, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

hbldh, 2021-01-11:
The fact that that line 8 (`async with BleakClient(d) as client:`) in your `run` method is in the stack trace means that you got past the if-clause in line 7. Thus, the device was detected and it is a case of not being able to connect to a discovered device that seems to be the problem.

Could you do a 
```python
async with BleakClient(d, timeout=30) as client:
    print("done")
```
instead and see if a longer timeout is needed to connect to your device?

emadul-credosense, 2021-01-13:
> The fact that that line 8 (`async with BleakClient(d) as client:`) in your `run` method is in the stack trace means that you got past the if-clause in line 7. Thus, the device was detected and it is [...]

I'm getting same type of error.

### Code:

import asyncio
from bleak import BleakScanner, BleakClient

async def run():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)
        if d.name == 'CredoSense SRC-5':
            async with BleakClient(d, timeout=30) as client:
                print("done")

loop = asyncio.get_event_loop()
loop.run_until_complete(run())

### Output:

C9CC1495-18CD-4D3F-9D38-5B6A3830E88A: Apple, Inc. (b'\x10\x05\x19\x1ck\xea\x94')
DEA9ABF7-0A4E-4B83-AB97-03B315E60A0D: CredoSense SRC-5
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/locks.py", line 226, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/tasks.py", line 487, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/hackintosh2/PycharmProjects/bleak_test/main.py", line 15, in <module>
    loop.run_until_complete(run())
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/Users/hackintosh2/PycharmProjects/bleak_test/main.py", line 9, in run
    async with BleakClient(d, timeout=30) as client:
  File "/Users/hackintosh2/PycharmProjects/bleak_test/venv_mac/lib/python3.9/site-packages/bleak/backends/client.py", line 59, in __aenter__
    await self.connect()
  File "/Users/hackintosh2/PycharmProjects/bleak_test/venv_mac/lib/python3.9/site-packages/bleak/backends/corebluetooth/client.py", line 90, in connect
    await manager.connect_(self._device_info, timeout=timeout)
  File "/Users/hackintosh2/PycharmProjects/bleak_test/venv_mac/lib/python3.9/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 151, in connect_
    await asyncio.wait_for(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/tasks.py", line 489, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

Process finished with exit code 1

hbldh, 2021-01-13:
I don't know why this happens, but it is suspect an issue either with your device or with your computer. Questions that might help you explore why it doesn't work:

- [ ] Can you connect to the CredoSense SRC-5 device with e.g. your phone?
- [ ] Can your computer connect to any device (e.g. a mouse or a headset) using the native Bluetooth manager?
- [ ] Can you connect to the CredoSense SRC-5 device with your computer and the native Bluetooth manager?
- [ ] Can you connect to the CredoSense SRC-5 device with your computer and some other BLE software?
- [ ] Can you connect to any other device using your computer with Bleak?

emadul-credosense, 2021-01-14:
> I don't know why this happens, but it is suspect an issue either with your device or with your computer. Questions that might help you explore why it doesn't work:
>
> * [ ]  Can you connect to the [...]

******************************
I can  connect to the CredoSense SRC-5 device with my phone
My computer lost its connectivity after every time i connect my phone to it
CredoSense SRC-5 can communicate with both windows(using bleak library) and android.