hbldh/bleak #405: ProtocolError (Error: 0x0F: Connection Rejected due to Unacceptable BD_ADDR)

BioMycoBit, 2020-12-30:
* bleak version: 0.10.0
* Python version: 3.9
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: PyBluez 0.30

### client.write_gatt_char(on_uuid, [0x1], True)

Attempting to control Philips Hue lights via Bluetooth Low Energy.  I am able to successfully connect to the client and read the services and characteristics but **receive BD_ADDR errors for many of the gatt characteristics**.

### Example Error
[Characteristic] 97fe6561-0003-4f62-86e9-b71ee2da3d22: (Handle: 20) (read,write-without-response,write) | Name: Unknown, Value: _**b'Could not get GATT characteristics for 97fe6561-0003-4f62-86e9-b71ee2da3d22: ProtocolError (Error: 0x0F: Connection Rejected due to Unacceptable BD_ADDR)'**_

### Exploration

Very few references online in regards to Unacceptable BD_ADDR.  

BLE Connection failure : Connection is refused due to unacceptable BD_ADDR
https://stackoverflow.com/questions/64985001/ble-connection-failure-connection-is-refused-due-to-unacceptable-bd-addr

This example suggested that the end user was utilizing Bluetooth Classic and not BLE.  Didn't think that'd be possible with Bleak...

### Code
```p
import asyncio
import logging
from bleak import BleakClient

async def run(address, debug=False):
    log = logging.getLogger(__name__)
    if debug:
        import sys

        log.setLevel(logging.DEBUG)
        h = logging.StreamHandler(sys.stdout)
        h.setLevel(logging.DEBUG)
        log.addHandler(h)

    async with BleakClient(address, adapter='hci0') as client:
        x = await client.is_connected()
        log.info(f"{address} Connected: {x}")

        name_uuid = '00002a00-' + '0000' +'-1000-8000-00805f9b34fb'
        service_uuid = '932c32bd-' + '0000' + '-47a2-835a-a8d455b859dd'
        on_uuid = '932c32bd-' + '0002' + '-47a2-835a-a8d455b859dd'

        name = await client.read_gatt_char(name_uuid)
        service = client.services.get_service(service_uuid)
        characteristic = service.get_characteristic(on_uuid)

        log.info(f"Name: {name}")
        log.info(f"Service: {service}")
        log.info(f"Characteristic: {characteristic}")

        # on
        await client.write_gatt_char(on_uuid, [0x1])
        await client.write_gatt_char(on_uuid, [0x1], True)
        

if __name__ == "__main__":
    address = "D7:2E:A6:53:2C:96"   # Philips Hue Lightbulb

    loop = asyncio.get_event_loop()
    loop.set_debug(True)
    loop.run_until_complete(run(address, True))
```

### Output

```
D7:2E:A6:53:2C:96 Connected: True
Name: bytearray(b'Hue Lamp')
Service: 932c32bd-0000-47a2-835a-a8d455b859dd: Unknown
Characteristic: [38] 932c32bd-0002-47a2-835a-a8d455b859dd: Unknown
Traceback (most recent call last):
  File "C:/repos/python-mindwave-mobile/example/bleak/bleak_work.py", line 45, in <module>
    loop.run_until_complete(run(address, True))
  File "C:\Users\Brandon\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:/repos/python-mindwave-mobile/example/bleak/bleak_work.py", line 37, in run
    await client.write_gatt_char(on_uuid, [0x1], True)
  File "C:\repos\python-mindwave-mobile\venv\lib\site-packages\bleak\backends\dotnet\client.py", line 679, in write_gatt_char
    raise BleakError(
bleak.exc.BleakError: Could not write value [1] to characteristic 932c32bd-0002-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F: Connection Rejected due to Unacceptable BD_ADDR)
```

hbldh, 2020-12-30:
@BioMycoBit Could you please run the `service_explorer.py` example on your device and post the output here?

BioMycoBit, 2020-12-31:
@hbldh

service_explorer.py output

```
Connected: True
[Service] 00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile
	[Characteristic] 00002a05-0000-1000-8000-00805f9b34fb: (Handle: 2) (indicate) | Name: Service Changed, Value: None 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 4) | Value: b'\x02\x00' 
[Service] 00001800-0000-1000-8000-00805f9b34fb: Generic Access Profile
	[Characteristic] 00002a00-0000-1000-8000-00805f9b34fb: (Handle: 6) (read) | Name: Device Name, Value: b'Hue Lamp' 
	[Characteristic] 00002a01-0000-1000-8000-00805f9b34fb: (Handle: 8) (read) | Name: Appearance, Value: b'\x00\x00' 
[Service] 0000180a-0000-1000-8000-00805f9b34fb: Device Information
	[Characteristic] 00002a29-0000-1000-8000-00805f9b34fb: (Handle: 11) (read) | Name: Manufacturer Name String, Value: b'Philips' 
	[Characteristic] 00002a24-0000-1000-8000-00805f9b34fb: (Handle: 13) (read) | Name: Model Number String, Value: b'LWA003' 
	[Characteristic] 00002a28-0000-1000-8000-00805f9b34fb: (Handle: 15) (read) | Name: Software Revision String, Value: b'1.55.7_r28193' 
[Service] 0000fe0f-0000-1000-8000-00805f9b34fb: Vendor specific
	[Characteristic] 97fe6561-0001-4f62-86e9-b71ee2da3d22: (Handle: 18) (read) | Name: Unknown, Value: b'J\xcfU\t\x01\x88\x17\x00' 
	[Characteristic] 97fe6561-0003-4f62-86e9-b71ee2da3d22: (Handle: 20) (read,write-without-response,write) | Name: Unknown, Value: b'Could not get GATT characteristics for 97fe6561-0003-4f62-86e9-b71ee2da3d22: ProtocolError (Error: 0x0F: Connection Rejected due to Unacceptable BD_ADDR)' 
	[Characteristic] 97fe6561-0004-4f62-86e9-b71ee2da3d22: (Handle: 22) (write-without-response,write) | Name: Unknown, Value: None 
	[Characteristic] 97fe6561-0008-4f62-86e9-b71ee2da3d22: (Handle: 24) (write-without-response,write,indicate) | Name: Unknown, Value: None 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 26) | Value: b'\x00\x00' 
	[Characteristic] 97fe6561-2001-4f62-86e9-b71ee2da3d22: (Handle: 27) (read,write-without-response,write) | Name: Unknown, Value: b'\x00' 
	[Characteristic] 97fe6561-2002-4f62-86e9-b71ee2da3d22: (Handle: 29) (write-without-response,write) | Name: Unknown, Value: None 
	[Characteristic] 97fe6561-2004-4f62-86e9-b71ee2da3d22: (Handle: 31) (write-without-response,write) | Name: Unknown, Value: None 
	[Characteristic] 97fe6561-a001-4f62-86e9-b71ee2da3d22: (Handle: 33) (write-without-response,write) | Name: Unknown, Value: None 
[Service] 932c32bd-0000-47a2-835a-a8d455b859dd: Unknown
	[Characteristic] 932c32bd-0001-47a2-835a-a8d455b859dd: (Handle: 36) (read) | Name: Unknown, Value: b'Could not get GATT characteristics for 932c32bd-0001-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F: Connection Rejected due to Unacceptable BD_ADDR)' 
	[Characteristic] 932c32bd-0002-47a2-835a-a8d455b859dd: (Handle: 38) (read,write-without-response,write,notify) | Name: Unknown, Value: b'Could not get GATT characteristics for 932c32bd-0002-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F: Connection Rejected due to Unacceptable BD_ADDR)' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 40) | Value: b'\x00\x00' 
	[Characteristic] 932c32bd-0003-47a2-835a-a8d455b859dd: (Handle: 41) (read,write-without-response,write,notify) | Name: Unknown, Value: b'Could not get GATT characteristics for 932c32bd-0003-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F: Connection Rejected due to Unacceptable BD_ADDR)' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 43) | Value: b'\x00\x00' 
	[Characteristic] 932c32bd-0006-47a2-835a-a8d455b859dd: (Handle: 50) (write-without-response,write) | Name: Unknown, Value: None 
	[Characteristic] 932c32bd-0007-47a2-835a-a8d455b859dd: (Handle: 52) (read,write-without-response,write,notify) | Name: Unknown, Value: b'Could not get GATT characteristics for 932c32bd-0007-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F: Connection Rejected due to Unacceptable BD_ADDR)' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 54) | Value: b'\x00\x00' 
	[Characteristic] 932c32bd-1005-47a2-835a-a8d455b859dd: (Handle: 55) (read,write-without-response,write) | Name: Unknown, Value: b'Could not get GATT characteristics for 932c32bd-1005-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F: Connection Rejected due to Unacceptable BD_ADDR)' 
[Service] b8843add-0000-4aa1-8794-c3f462030bda: Unknown
	[Characteristic] b8843add-0001-4aa1-8794-c3f462030bda: (Handle: 58) (read) | Name: Unknown, Value: b'Could not get GATT characteristics for b8843add-0001-4aa1-8794-c3f462030bda: ProtocolError (Error: 0x0F: Connection Rejected due to Unacceptable BD_ADDR)' 
	[Characteristic] b8843add-0002-4aa1-8794-c3f462030bda: (Handle: 60) (write-without-response,write,indicate) | Name: Unknown, Value: None 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 62) | Value: b'\x00\x00' 
	[Characteristic] b8843add-0003-4aa1-8794-c3f462030bda: (Handle: 63) (write-without-response,write) | Name: Unknown, Value: None 
	[Characteristic] b8843add-0004-4aa1-8794-c3f462030bda: (Handle: 65) (read) | Name: Unknown, Value: b'Could not get GATT characteristics for b8843add-0004-4aa1-8794-c3f462030bda: ProtocolError (Error: 0x0F: Connection Rejected due to Unacceptable BD_ADDR)' 
```

hbldh, 2021-01-11:
This is indeed strange. Some services and characteristics seem to work in a BLE/GATT sense, while the error your received seems to be stemming from a regular Bluetooth connection. I have never seen that before...

Could you try to pair with the device directly after connecting?

```
async with BleakClient(address) as client:
        x = await client.is_connected()
        log.info(f"{address} Connected: {x}")
        paired = await client.pair()
        log.info(f"{address} Paired: {paired}")
```

Does that work and, if so, does it change the output of the `service_explorer.py` output?

BioMycoBit, 2021-01-11:
> ```
> async with BleakClient(address) as client:
>         x = await client.is_connected()
>         log.info(f"{address} Connected: {x}")
>         paired = await client.pair()
>         log.info(f"{address} [...]

@hbldh 

This does indeed work

```
D7:2E:A6:53:2C:96 Connected: True
D7:2E:A6:53:2C:96 Paired: True
```

One change between output

Original: 		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 4) | **Value: b'\x02\x00'**
```
[Service] 00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile
	[Characteristic] 00002a05-0000-1000-8000-00805f9b34fb: (Handle: 2) (indicate) | Name: Service Changed, Value: None
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 4) | Value: b'\x02\x00'
```

Updated: 		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 4) | **Value: b'\x00\x00'**
```
[Service] 00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile
	[Characteristic] 00002a05-0000-1000-8000-00805f9b34fb: (Handle: 2) (indicate) | Name: Service Changed, Value: None
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 4) | Value: b'\x00\x00'
```

hbldh, 2021-01-12:
If it works, that is great. It seems pairing was what was needed then. I will close this issue since the problem seems to have been solved.

BioMycoBit, 2021-01-12:
@hbldh 

in clarifying, it says it is paired but the same output error is being generated

the only change in the output was:
Original: [Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 4) | **Value: b'\x02\x00'**
Updated: [Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 4) | **Value: b'\x00\x00'**

```
bleak.exc.BleakError: Could not write value [1] to characteristic 932c32bd-0002-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F: Connection Rejected due to Unacceptable BD_ADDR)
```

hbldh, 2021-01-12:
Ah. Then, once again, I have no idea why this occurs. Will contemplate on this for a while.

BioMycoBit, 2021-01-12:
@hbldh 

Appreciate that Henrik.

hbldh, 2021-01-13:
Ok, so I got hold of a Hue lamp from a neighbour and did some testing. It is available in the `feature/pairing-improvement-dotnet` branch as of yet and is in PR #421.

It seems that the device needs to be connected to in the official Philips Hue Bluetooth app and reset from there to be able to use with Bleak-type of BLE software. After that one needs to do a encrypted pairing to enable reading and writing of characteristics.

After that the `ProtocolError (Error: 0x0F: Connection Rejected due to Unacceptable BD_ADDR)` disappears when running `service_explorer.py` and one can control the bulb with Bleak.

I included an example file (`philips_hue.py`) on my findings trying to control it. If you want the challange of finding out yourself how to do it, then don't look at it. The only thing is that you need the code in the `feature/pairing-improvement-dotnet` branch or wait until we release version 0.10.1.

BioMycoBit, 2021-01-14:
@hbldh 

I was able to install the ```feature/pairing-improvement-dotnet``` branch and utilize ```philips_hue.py``` successfully.

Thanks so much Henrik.

closed 2021-01-14 by hbldh

mvrkusm, 2024-05-14:
@BioMycoBit Hi,

Is this Philips Hue controlling script available somewhere? I ran into similar problems.