hbldh/bleak #406: dbus-next backend: Multiple AssertionErrors on checking for MessageType.METHOD_RETURN

devbis, 2020-12-30:
* bleak version: snapshot, dbus-next-2 branch
* Python version: 3.9
* Operating System: OpenWrt
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.54

### Description
After several minutes of running my software that is based on bleak, I receive AssertionError exceptions:

```
  File "/usr/lib/python3.9/site-packages/bleak/backends/bluezdbus/client.py", line 273, in connect
    assert reply.message_type == MessageType.METHOD_RETURN
AssertionError
```
or

```
  File "/usr/lib/python3.9/site-packages/bleak/backends/scanner.py", line 67, in __aexit__
    await self.stop()
  File "/usr/lib/python3.9/site-packages/bleak/backends/bluezdbus/scanner.py", line 167, in stop
    assert reply.message_type == MessageType.METHOD_RETURN
AssertionError
```

(And several other places. If I comment out this assert, another one raises an exception after some time)

https://pastebin.com/cXKPkHPE

The actual message type is `MessageType.ERROR`.

/cc: @dlech

dlech, 2020-12-30:
I have added a change that should tell us what the actual error is now. @devbis Can you run again and get a log with this latest change?

devbis, 2020-12-31:
I ran my application for a while with the updated bleak and it seems that this issue is fixed. 
Thank you for your fixes!

closed 2020-12-31 by devbis

dlech, 2020-12-31:
Interesting, I guess it was already catching `BleakError`s in these cases?

dlech, 2020-12-31:
And thanks for all of the testing!

devbis, 2021-01-03:
Yes, I'm catching a bunch of error in my case. I suppose the bluetooth chip itself isn't stable in my case.
It often returns org.bluez.Error.Failed, org.bluez.Error.InProgress and other bluez errors.

dlech, 2021-01-03:
> org.bluez.Error.InProgress

This means you are trying to do something (like connect) before the previous attempt has finished. So it would be interesting to know the code paths that lead to this error.