hbldh/bleak #407: dbus-next backend: KeyError while scanning

devbis, 2020-12-30:
* bleak version: snapshot, dbus-next-2 branch
* Python version: 3.9
* Operating System: OpenWrt
* BlueZ version: 5.54

### Description
After several minutes of running my software that is based on bleak, I receive KeyError exception:

```
    File "/usr/lib/python3.9/site-packages/dbus_next/message_bus.py", line 656, in _process_message
    result = handler(msg)
  File "/usr/lib/python3.9/site-packages/bleak/backends/bluezdbus/scanner.py", line 290, in _parse_msg
    del self._devices[obj_path]
KeyError: '/org/bluez/hci0/dev_18_D8_91_DE_BE_C4'
```

The full log is here https://pastebin.com/Eg6m7ydz

/cc: @dlech

dlech, 2020-12-30:
This one should be fixed now.

devbis, 2020-12-31:
Thank you, yes it is fixed.

commit 5bf72668 (2020-12-30): Fixed KeyError on InterfacesRemoved signal in BlueZ scanner.

This handles the case where an InterfacesRemoved signal is received
but the device hasn't been "seen" yet.

Fixes: #407
closed 2020-12-31 by devbis