hbldh/bleak #408: dbus-next backend: write_gatt_char requres bytes instead of bytearray as data type

devbis, 2020-12-31:
* bleak version: snapshot, dbus-next-2 branch
* Python version: 3.9
* Operating System: OpenWrt
* BlueZ version: 5.54

### Description

Annotation for write_gatt_char defines that it should be bytearray type for data parameter.
https://github.com/hbldh/bleak/blob/dbus-next-2/bleak/backends/client.py#L178-L183

But using it with dbus-next with bytearray as data raises an exception because of the check for bytes
```
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/ble2mqtt/ble2mqtt.py", line 556, in _run_device_tasks
    t.result()
  File "/usr/lib/python3.9/site-packages/ble2mqtt/ble2mqtt.py", line 402, in manage_device
    await device.get_device_data()
  File "/usr/lib/python3.9/site-packages/ble2mqtt/devices/kettle_redmond.py", line 79, in get_device_data
    await super().protocol_start()
  File "/usr/lib/python3.9/site-packages/ble2mqtt/protocols/redmond.py", line 143, in protocol_start
    await self.client.write_gatt_char(
  File "/usr/lib/python3.9/site-packages/bleak/backends/bluezdbus/client.py", line 716, in write_gatt_char
    reply = await self._bus.call(
  File "/usr/lib/python3.9/site-packages/dbus_next/aio/message_bus.py", line 303, in call
    self._call(msg, reply_handler)
  File "/usr/lib/python3.9/site-packages/dbus_next/message_bus.py", line 582, in _call
    self.send(msg)
  File "/usr/lib/python3.9/site-packages/dbus_next/aio/message_bus.py", line 326, in send
    self._writer.schedule_write(msg, future)
  File "/usr/lib/python3.9/site-packages/dbus_next/aio/message_bus.py", line 82, in schedule_write
    self.buffer_message(msg, future)
  File "/usr/lib/python3.9/site-packages/dbus_next/aio/message_bus.py", line 78, in buffer_message
    (msg._marshall(negotiate_unix_fd=self.negotiate_unix_fd), copy(msg.unix_fds), future))
  File "/usr/lib/python3.9/site-packages/dbus_next/message.py", line 215, in _marshall
    body_block = Marshaller(self.signature, self.body)
  File "/usr/lib/python3.9/site-packages/dbus_next/_private/marshaller.py", line 8, in __init__
    self.signature_tree.verify(body)
  File "/usr/lib/python3.9/site-packages/dbus_next/signature.py", line 359, in verify
    type_.verify(body[i])
  File "/usr/lib/python3.9/site-packages/dbus_next/signature.py", line 287, in verify
    self._verify_array(body)
  File "/usr/lib/python3.9/site-packages/dbus_next/signature.py", line 220, in _verify_array
    raise SignatureBodyMismatchError(
dbus_next.errors.SignatureBodyMismatchError: DBus ARRAY type "a" with BYTE child must be Python type "bytes", got <class 'bytearray'>
```

I used this code:
```python
await client.write_gatt_char(
    uuid.UUID("6e400002-b5a3-f393-e0a9-e50e24dcca9e"),
    bytearray(0x01.to_bytes(2, byteorder="little")),
    True,
)
```

Removing conversion to bytearray fixes the problem but (I suppose) makes the code not portable.

dlech, 2020-12-31:
Should be fixed now.

devbis, 2021-01-03:
Yes, it works. Thanks for the quick fix!

commit 9e53c954 (2020-12-31): Fixed dbus-next "ay" must be bytes.

This fixes failed characteristic and descriptor writes when the data
argument is not already a bytes object. The low-level D-Bus API
requires strict typing.

Fixes: #408
closed 2021-01-03 by devbis