hbldh/bleak #411: Trouble connecting to iPhone from Linux

cspensky, 2021-01-05:
* bleak version: dbus-next-2 (branch)
* Python version: Python 3.8.0
* Operating System: Ubuntu 18.04.5 LTS
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.54

### Description

Trying to initiate a simple connection with my phones.

### What I Did

I ran a simple connect script, and cannot seem to get Linux to connect to my iPhone.  Has anyone else had issues?

My test script:
```python
import asyncio
import platform

from bleak import BleakClient, BleakScanner

async def print_services(mac_addr: str):
    device = await BleakScanner.find_device_by_address(mac_addr)
    async with BleakClient(device) as client:
        svcs = await client.get_services()
        print("Services:", svcs)

# iPhone XR
mac_addr_iphone = (
    "f4:af:e7:a9:7b:7b"
    if platform.system() != "Darwin"
    else "5B18BB9D-EE12-40F2-AC66-C39BA6804410"
)

# Pixel 5
mac_addr_android = (
    "4b:31:c7:a8:69:8d"
    if platform.system() != "Darwin"
    else "906D3D50-776F-47E2-9B00-A566F75E8CCF"
)

print("Android")
loop = asyncio.get_event_loop()
loop.run_until_complete(print_services(mac_addr_android))

print("iPhone")
loop = asyncio.get_event_loop()
loop.run_until_complete(print_services(mac_addr_iphone))
```

OSX:
```bash
$ python linux_iphone.py
Android
Services: <bleak.backends.service.BleakGATTServiceCollection object at 0x10d3bf130>
iPhone
Services: <bleak.backends.service.BleakGATTServiceCollection object at 0x10d3bfb80>
```

Linux:
```bash
$ python linux_iphone.py
Android
Services: <bleak.backends.service.BleakGATTServiceCollection object at 0x7f6ae5eef340>
iPhone
Traceback (most recent call last):
  File "linux_iphone.py", line 38, in <module>
    loop.run_until_complete(print_services(mac_addr_iphone))
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "linux_iphone.py", line 13, in print_services
    async with BleakClient(device) as client:
  File "bleak/bleak/backends/client.py", line 59, in __aenter__
    await self.connect()
  File "bleak/bleak/backends/bluezdbus/client.py", line 290, in connect
    assert_reply(reply)
  File "bleak/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name)
bleak.exc.BleakDBusError: org.bluez.Error.Failed
```

dlech, 2021-01-05:
Can you run with debug logging enabled and post the logs for the failure?

cspensky, 2021-01-05:
Here is the relevant portion:
```bash
iPhone
DEBUG:asyncio:Using selector: EpollSelector
DEBUG:bleak.backends.bluezdbus.scanner:cached devices: {'/org/bluez/hci0/dev_54_B0_CE_EF_38_97': {'Address': '54:B0:CE:EF:38:97', 'AddressType': 'public', 'Name': 'SENSE  ', 'Alias': 'SENSE  ', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x06\xad\x01\xc6\x93\x89\n\xc4\xb1\x06\x00\x01\x00\x01\x02'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_F1_A9_4E_90_BC_65': {'Address': 'F1:A9:4E:90:BC:65', 'AddressType': 'random', 'Alias': 'F1-A9-4E-90-BC-65', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000feed-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ServiceData': {'0000feed-0000-1000-8000-00805f9b34fb': b'\x02\x00\xab\x12\x1b\xe4>i\x08R'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_E7_1A_68_63_94_EA': {'Address': 'E7:1A:68:63:94:EA', 'AddressType': 'random', 'Name': 'Tile', 'Alias': 'Tile', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000feed-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ServiceData': {'0000feed-0000-1000-8000-00805f9b34fb': b'\x02\x00\x18\xc9SA\x137\xc0\xc1'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_77_CB_C7_5A_13_7D': {'Address': '77:CB:C7:5A:13:7D', 'AddressType': 'random', 'Alias': '77-CB-C7-5A-13-7D', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['50400001-b5a3-f393-e0a9-e50e24dcca9e'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}, '/org/bluez/hci0/dev_6B_D1_C2_9D_65_CB': {'Address': '6B:D1:C2:9D:65:CB', 'AddressType': 'random', 'Alias': '6B-D1-C2-9D-65-CB', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x10\x05\x01\x98\xd1\xf4\x05'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_CC_98_8B_B7_2A_FA': {'Address': 'CC:98:8B:B7:2A:FA', 'AddressType': 'public', 'Name': 'LE_WH-1000XM3', 'Alias': 'LE_WH-1000XM3', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000fe03-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ServiceData': {'0000fe26-0000-1000-8000-00805f9b34fb': b'\n\xc9\\'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_4E_C8_9E_32_B5_EB': {'Address': '4E:C8:9E:32:B5:EB', 'AddressType': 'random', 'Alias': '4E-C8-9E-32-B5-EB', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000fe50-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ServiceData': {'0000fe50-0000-1000-8000-00805f9b34fb': b'\xd6\x00'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_3C_C9_BF_CC_1A_65': {'Address': '3C:C9:BF:CC:1A:65', 'AddressType': 'random', 'Alias': '3C-C9-BF-CC-1A-65', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000fd6f-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ServiceData': {'0000fd6f-0000-1000-8000-00805f9b34fb': b'\xf7\xf8\xe9\x87k\xbe\xebZ-o%\xa2\x7f\xc5tP\xa9#\xdf\x16'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_38_18_4C_1A_22_73': {'Address': '38:18:4C:1A:22:73', 'AddressType': 'public', 'Name': 'LE_WH-1000XM3', 'Alias': 'LE_WH-1000XM3', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000fe03-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}, '/org/bluez/hci0/dev_79_2D_35_89_C2_05': {'Address': '79:2D:35:89:C2:05', 'AddressType': 'random', 'Alias': '79-2D-35-89-C2-05', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_44_EA_28_D3_4F_3A': {'Address': '44:EA:28:D3:4F:3A', 'AddressType': 'random', 'Alias': '44-EA-28-D3-4F-3A', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['50400001-b5a3-f393-e0a9-e50e24dcca9e'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}, '/org/bluez/hci0/dev_76_88_5C_09_C4_CF': {'Address': '76:88:5C:09:C4:CF', 'AddressType': 'random', 'Alias': '76-88-5C-09-C4-CF', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_F8_FF_C2_20_EC_25': {'Address': 'F8:FF:C2:20:EC:25', 'AddressType': 'public', 'Alias': 'F8-FF-C2-20-EC-25', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x10\x05N\x1c/6e'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_20_99_0D_1B_5C_BB': {'Address': '20:99:0D:1B:5C:BB', 'AddressType': 'random', 'Alias': '20-99-0D-1B-5C-BB', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000fd6f-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez [... cut ...]
  File "/home/cspensky/allthenticate/core/python-libary/repositories/bleak/bleak/backends/client.py", line 59, in __aenter__
    await self.connect()
  File "/home/cspensky/allthenticate/core/python-libary/repositories/bleak/bleak/backends/bluezdbus/client.py", line 290, in connect
    assert_reply(reply)
  File "/home/cspensky/allthenticate/core/python-libary/repositories/bleak/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name)
bleak.exc.BleakDBusError: org.bluez.Error.Failed
```

dlech, 2021-01-05:
Interesting, we get the property change indicating that it did connect but the connect method fails. Looking through the [BlueZ source code](https://github.com/bluez/bluez/blob/c96d5ec7aafc8fd988f6f19b896115853b951d91/src/device.c#L5430) it looks like this can mean that another connection is in progress or there is an issue with bonding (I don't know anything about how bonding works).

dlech, 2021-01-05:
Although the latter may actually print an error message for `bluetoothd`. Try checking `sudo journalctl -R` or `sudo systemctl status bluetooth.service`.

cspensky, 2021-01-06:
Interesting.  BlueZ is definitely unhappy.

```bash
bluetoothd[17724]: connect error: Invalid exchange (52)
```

dlech, 2021-01-06:
I guess the next step is `wireshark` to see what is really going on.

dlech, 2021-01-06:
Also that "connect error" seems to be removed in BlueZ master, so maybe something that is fixed upstream already?

dlech, 2021-01-06:
https://stackoverflow.com/questions/43106595/what-does-python-socket-error-no-52-invalid-exchange-refer-to

cspensky, 2021-01-06:
I see... It seems like it has to do with pairing and requiring a PIN.

`btmon` output:
```bash
> HCI Event: Remote Name Req Complete (0x07) plen 255                                                                  #114 [hci0] 6.641460
        Status: Success (0x00)
        Address: F4:AF:E7:A9:7B:7B (OUI F4-AF-E7)
        Name: Chad’s iPhone
@ MGMT Event: Device Connected (0x000b) plen 30                                                                    {0x0003} [hci0] 6.641526
        BR/EDR Address: F4:AF:E7:A9:7B:7B (OUI F4-AF-E7)
        Flags: 0x00000000
        Data length: 17
        Name (complete): Chad’s iPhone
@ MGMT Event: Device Connected (0x000b) plen 30                                                                    {0x0001} [hci0] 6.641526
        BR/EDR Address: F4:AF:E7:A9:7B:7B (OUI F4-AF-E7)
        Flags: 0x00000000
        Data length: 17
        Name (complete): Chad’s iPhone
< HCI Command: Authentication Requested (0x01|0x0011) plen 2                                                           #115 [hci0] 6.641566
        Handle: 12
> HCI Event: Command Status (0x0f) plen 4                                                                              #116 [hci0] 6.643424
      Authentication Requested (0x01|0x0011) ncmd 1
        Status: Success (0x00)
> HCI Event: Link Key Request (0x17) plen 6                                                                            #117 [hci0] 6.644418
        Address: F4:AF:E7:A9:7B:7B (OUI F4-AF-E7)
< HCI Command: Link Key Request Reply (0x01|0x000b) plen 22                                                            #118 [hci0] 6.644457
        Address: F4:AF:E7:A9:7B:7B (OUI F4-AF-E7)
        Link key: 2228da60bf8e0964bb24a71d98f7afab
> HCI Event: Command Complete (0x0e) plen 10                                                                           #119 [hci0] 6.646440
      Link Key Request Reply (0x01|0x000b) ncmd 1
        Status: Success (0x00)
        Address: F4:AF:E7:A9:7B:7B (OUI F4-AF-E7)
> HCI Event: Auth Complete (0x06) plen 3                                                                               #120 [hci0] 6.651415
        Status: PIN or Key Missing (0x06)
        Handle: 12
@ MGMT Event: Authentication Failed (0x0011) plen 8                                                                {0x0003} [hci0] 6.651451
        BR/EDR Address: F4:AF:E7:A9:7B:7B (OUI F4-AF-E7)
        Status: Authentication Failed (0x05)
@ MGMT Event: Authentication Failed (0x0011) plen 8                                                                {0x0001} [hci0] 6.651451
        BR/EDR Address: F4:AF:E7:A9:7B:7B (OUI F4-AF-E7)
        Status: Authentication Failed (0x05)
= bluetoothd: connect error: Invalid exchange (52)                                                                                 6.653238
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                         #121 [hci0] 6.675749
        Handle: 12
        Reason: Remote User Terminated Connection (0x13)
```

dlech, 2021-01-06:
Did you try removing the device using `bluetoothctl`? 

The BlueZ connect method is goofy and tries to guess if you want a LE or BR/EDR (classic) connection. I'm not sure why there isn't a dedicated method for connecting LE only.

cspensky, 2021-01-06:
It's definitely something with pairing. If I set `pairable off` it will not connect and ask me to connect with a PIN.  `agent off` makes it even more exciting by popping the message up only briefly.  I'll report back after I dig in a bit.  Thanks for the sounding board!

cspensky, 2021-01-06:
After much digging, it appears to be BlueZ trying to read the battery status of my iPhone when I connect.  

I attached [PacketLogger to my iPhone](https://www.bluetooth.com/blog/a-new-way-to-debug-iosbluetooth-applications/) and it looks like Apple requires authentication to read that characteristic.
[image]

Odd behavior indeed and the fact that it is doing it by default, without a flag is pretty annoying.  Is this a Bleak thing?  Any insights into how to turn this off would be great.  Otherwise, I plan to continue to dig, and potentially submit a PR to the BlueZ folks.

dlech, 2021-01-07:
Try the suggestion at https://github.com/hbldh/bleak/issues/393#issue-765523895 to disable the battery plugin in BlueZ.

cspensky, 2021-01-07:
That worked! Thanks a million. 🙏

closed 2021-01-07 by cspensky