hbldh/bleak #412: Enumerating Connected Devices

xloem, 2021-01-06:
* bleak version:
```
$ pip3 show bleak
Name: bleak
Version: 0.10.0
Summary: Bluetooth Low Energy platform Agnostic Klient
Home-page: https://github.com/hbldh/bleak
Author: Henrik Blidh
Author-email: henrik.blidh@nedomkull.com
License: MIT
Location: /home/user/.local/lib/python3.8/site-packages
Requires: txdbus
Required-by: 
```
* Python version:
```
$ python3 --version
Python 3.8.7+
```
* Operating System:
```
$ cat /etc/redhat-release
Red Hat Enterprise Linux Workstation release 7.7 (Maipo)
```
* BlueZ version (`bluetoothctl -v`) in case of Linux:
```
$ bluetoothctl -v
bluetoothctl: 5.54
```
### Description

Discovery does not return connected devices.  This makes it unreasonable to find a device again, if an application terminates with it still connected.  I don't see a platform-independent way to enumerate connected devices in bleak.

### What I Did

I think I may have previously addressed this issue in another project on linux by using dbus's object manager to enumerate what devices the system knows about.  Source code lines are https://github.com/ukBaz/python-bluezero/pull/285/files#diff-d57297f7a4f2dff077526b516a15723592c185210ad1a0344909bf6f56ad23c4R34 and https://github.com/ukBaz/python-bluezero/pull/285/files#diff-0b983208ec15acea798d4ab2ea013b560b273d1248914cbb152c37f39f821f99L26

Here is a python script to test this on Bleak:
`test.py`:
```
import bleak

async def main():
    scanner = bleak.BleakScanner()
    ADDRESS = '00:55:DA:BB:1C:52'

    print(f'discovering ...')
    addresses = [device.address for device in await scanner.discover(timeout=1)]
    if ADDRESS not in addresses:
        raise Exception(f'failed to find {ADDRESS} at all')

    print(f'connecting ...')
    device = bleak.BleakClient(ADDRESS)
    await device.connect(timeout=1)

    print(f'rediscovering ...')
    addresses = [device.address for device in await scanner.discover(timeout=1)]

    print(f'disconnecting ...')
    await device.disconnect()

    if ADDRESS not in addresses:
        raise Exception(f'Test FAILED: enumerated {ADDRESS} while disconnected but not while connected')
    else:
        print(f'Test passed: {ADDRESS} enumerated while connected')

asyncio.run(main())
```
```
$ python3 test.py
discovering ...
connecting ...
rediscovering ...
disconnecting ...
Traceback (most recent call last):
  File "test.py", line 23, in <module>
    asyncio.run(main())
  File "/usr/local/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "test.py", line 19, in main
    raise Exception(f'Test FAILED: enumerated {ADDRESS} while disconnected but not while connected')
Exception: Test FAILED: enumerated 00:55:DA:BB:1C:52 while disconnected but not while connected
```

dlech, 2021-01-06:
Also see #367.

hbldh, 2021-01-12:
I never intended Bleak to replace or compete the native Bluetooth Managers in respective operating systems. Bleak in its current incarnation lets you connect and allows you to handle and listen to that very connection, nothing else. I'm sorry, but it falls outside the scope of the package.

closed 2021-01-12 by hbldh

xloem, 2021-01-12:
? how does this compete with the native manager?
this is a precursor to accessing a connected a device

hbldh, 2021-01-12:
Bleak was not designed to dig up connected devices in the OS and handle those connections. Maybe it can be done, but it is outside the scope of the package. Disconnect from them using the native Bluetooth managers instead and then use Bleak.

xloem, 2021-01-12:
this doesn't seem hard or big to me, but i won't be needing it any time soon.  let me know if you'd accept a PR or if forking the project is the way to go if we want apps to be able to reconnect after being killed.

hbldh, 2021-01-12:
The problem with Bleak and adding functionality is that I want it to be available across all operating systems. If one user likes Linux and wants to add functionality there, someone (most probably me) will have to implement that for the other OS backends as well. And if the functionality cannot be provided in all backends due to differences in APIs, then I am very reluctant to add it to Bleak. It is meant to be a cross-platform package with as few differences as possible, and I don't want to add more than there already are.

That said, if you try this out and make it work across all backends, then I will, most definitely, accept that PR.

xloem, 2021-01-12:
thanks =)

i was assuming that enumerating connected devices wasn't an issue on other platforms, i'm realising that could be very wrong.  the almost-entirely-uniform-interface policy makes a lot of sense and will likely help a lot of people.

dlech, 2021-01-13:
Instead of trying to connect to already connected devices, what if we made Bleak better at disconnecting instead? I've made some attempt at this in https://github.com/hbldh/bleak/pull/389/commits/3af7aeba802b83fcffbb45ee5485d1b1206b0127 but it seems to cause error messages sometimes because of a leftover task.

xloem, 2021-01-13:
@dlech that's important and I see you're on #315, but also doesn't resolve terminations when python doesn't have a chance to run cleanup, at least there should be some way to detect the issue and inform the user

dlech, 2021-01-13:
If the event loop is run with `asyncio.run()`, then (short of a segfault) it will always clean up.

xloem, 2021-01-14:
right, things like segfaults happen in out of memory situations and common consumer situations from web malware.  it's not usually a huge concern, the issue is that here the bt device becomes inaccessible until reboot for a normal user.

dlech, 2021-01-14:
> the issue is that here the bt device becomes inaccessible until reboot for a normal user.

How so? Couldn't a user use `bluetootctl` or the system settings/control panel to disconnect a Bluetooth device without rebooting in the rare case of a segfault in a Bleak app?

xloem, 2021-01-14:
right, but most users will have no idea how to do that, and the extra annoyance turn them off applications that use the library

leycec, 2022-04-09:
**Consider reopening this,** @dlech and/or @hbldh. On the one hand, robustly handling open zombie BLE connections is core functionality that absolutely should exist. On the other hand, this is an open-source free-as-in-beer-and-speech passion project with inadequate funding – like most GitHub projects widely used across industry! <sup>[...including my own](https://github.com/beartype/beartype). :sweat: </sup>

These two things are both true. Nonetheless, the former ultimately trumps the latter. This feature is a soft prerequisite for deployment of production-ready real-world mobile apps. Technically, this feature isn't strictly required... but, pragmatically, it *really* is.

Funding or no funding, this eventually needs doing.

leycec, 2022-04-09:
Oh, great! Thanks so much for the quick heads-up, @dlech. You on the GitHub ball, as always. :+1: