hbldh/bleak #415: [feature] Synchronous Interface

xloem, 2021-01-07:
To make it easier for developers to provide for different bluetooth backends (where most are synchronous), it would be nice to provide a synchronous interface to bleak.  One approach is using asgiref's async_to_sync function to wrap the existing functions.

hbldh, 2021-01-07:
No, this is not in bleak's scope. There are synchronous BLE client libraries already. Bleak is by design using asyncio. 

You are free to implement it yourself and release as a package on PyPI, but I don't want it included in Bleak. Sorry.

closed 2021-01-07 by hbldh

xloem, 2021-01-07:
Thanks for your quick response.  I made an async-to-sync class object wrapper that's 47 lines long,  I've never made a pypi package before but it can't be too hard.

xloem, 2021-01-07:
=) https://pypi.org/project/async-to-sync/