hbldh/bleak #416: Track value handles

xloem, 2021-01-07:
bleak provides for accessing characteristics based on their declaration handle.  It can also be nice to access them based on the direct value handle they have.  Since these never overlap, supporting both is possible using the same interface.

hbldh, 2021-01-12:
I remember value handles from my use of `pygatt` and `pygattlib`, but I have not seen them in any of the backends that Bleak uses. I do not think this is possible to implement with the APIs I have. Sorry.

closed 2021-01-12 by hbldh

xloem, 2021-01-12:
well this is a pretty microscopic thing, but if anybody ever needs this in the future, for e.g. interoperating with pygatt, it seems it would be as easy as adding another property to the characteristic class, and throwing an error or returning a dummy value by default.

hbldh, 2021-01-12:
The value is not present in the backend APIs, so there is nothing to fill the added property with.

xloem, 2021-01-12:
good point.  my concern would move to bluez.

hbldh, 2021-01-12:
I honestly did not think they were necessary. I thought a write to some descriptor under the characteristic would be identical to writing with the value handle.

xloem, 2021-01-12:
bluez translates the characteristic handle to the value handle when the value access is done.  the code i was working with had the device interface documented based on value handles instead of characteristic handles, because they had used pygatt, but learning about the difference really helped me understand bluetooth services better.