hbldh/bleak #417: how to import and use bleak in uwp app

barron9, 2021-01-07:


hbldh, 2021-01-11:
Note that it is not guaranteed that you will successfully be able to pair with all devices you discover. It might work to pair using Bleak, given that the device does not need any additional encryption and no confirmations.

There is a bug in the current version of Bleak regarding pairing, but if you install the version found in the `feature/pairing-improvement-dotnet`. Try installing that and then run this code and report the results here:

```python
import asyncio
from bleak import BleakScanner, BleakClient

async def run():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d.address + d.name)
        async with BleakClient(d) as client:
            print(f"CanPair: {client._requester.DeviceInformation.Pairing.CanPair}")
            print(f"IsPaired: {client._requester.DeviceInformation.Pairing.IsPaired}")
            res = await client.pair(protection_level=None)
            print(res)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```

NikitaKozlovtcev, 2021-02-18:
I also cannot connect to the BLE device. The code above gives this error. 
```python
Traceback (most recent call last):
  File "C:/REP/Development/Tools/BLE/main.py", line 41, in <module>
    loop.run_until_complete(run())
  File "C:\Users\Nikita_Kozlovtsev\AppData\Local\Programs\Python\Python38-32\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:/REP/Development/Tools/BLE/main.py", line 22, in run
    async with BleakClient(d) as client:
  File "C:\Users\Nikita_Kozlovtsev\AppData\Local\Programs\Python\Python38-32\lib\site-packages\bleak\backends\client.py", line 59, in __aenter__
    await self.connect()
  File "C:\Users\Nikita_Kozlovtsev\AppData\Local\Programs\Python\Python38-32\lib\site-packages\bleak\backends\dotnet\client.py", line 231, in connect
    await asyncio.wait_for(event.wait(), timeout=10)
  File "C:\Users\Nikita_Kozlovtsev\AppData\Local\Programs\Python\Python38-32\lib\asyncio\tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
```

hbldh, 2021-02-19:
Try a longer timeout: `async with BleakClient(d, timeout=30) as client:`

NikitaKozlovtcev, 2021-02-19:
> Try a longer timeout: `async with BleakClient(d, timeout=30) as client:`

it really makes mistakes less. But requests for 30 seconds are very long. I wrote more about my case here
https://github.com/hbldh/bleak/issues/452

closed 2021-02-19 by hbldh