hbldh/bleak #418: Get extra ManufacturerData with a Scan Response

matteobertozzi, 2021-01-08:
* bleak version: 0.10.0
* Python version: 3.7.3
* Operating System: Linux (Debian 10.7)
* BlueZ version (`bluetoothctl -v`): 5.50

I have a BLE device that provides extra information (extra ManufacturerData) in the Scan Response message.
but I did not see how to get this data using discover(). 

Is there a way to send a scan request to get the extra information?

Thanks!

dlech, 2021-01-08:
Have you tried [register_detection_callback](https://bleak.readthedocs.io/en/latest/api.html#bleak.backends.scanner.BaseBleakScanner.register_detection_callback)?

matteobertozzi, 2021-01-08:
> Have you tried [register_detection_callback](https://bleak.readthedocs.io/en/latest/api.html#bleak.backends.scanner.BaseBleakScanner.register_detection_callback)?

for the same device I always get the same AdvertisementData. 
I just tried it again with the simple https://github.com/hbldh/bleak/blob/master/examples/detection_callback.py
is there a flag to pass to the scanner or should it work just like this? I immagine that the callback should be called "twice" with the two different AdvertisementData

dlech, 2021-01-08:
Please try the `dbus-next-2` branch.

    pip install --force-reinstall https://github.com/hbldh/bleak/archive/dbus-next-2.zip#egg=bleak

matteobertozzi, 2021-01-08:
```
>>> import bleak
>>> bleak.__version__
'0.11.0a1'
```

For the devices I'm interested in nothing changed, but for a device that is around the area (not sure what it is)
I got two different calls one with the manufacturer_data the other without
```
4B:1E:F4:70:B8:50 RSSI: -87 AdvertisementData(manufacturer_data={224: b'\x018\xca\x81M '}, service_data={'0000fe9f-0000-1000-8000-00805f9b34fb': b'\x02k23ucAUHtdY\x00\x00\x01v\xe3\x9e\xbf|'}, service_uuids=['0000fe9f-0000-1000-8000-00805f9b34fb'])
4B:1E:F4:70:B8:50 RSSI: -87 AdvertisementData(service_data={'0000fe9f-0000-1000-8000-00805f9b34fb': b'\x02k23ucAUHtdY\x00\x00\x01v\xe3\x9e\xbf|'}, service_uuids=['0000fe9f-0000-1000-8000-00805f9b34fb'])
```

dlech, 2021-01-08:
Sounds like a problem with the device or with BlueZ then. If BlueZ doesn't think that the data changed, it won't send any property changes events to Bleak.

matteobertozzi, 2021-01-14:
I've played around a bit more to understand what is going on.

using the Android app "BLE Scanner" i can see the raw advertisements
and then I have a nice table splitting the advertisement with Types
```
0x01 (Flags) 0x06
0xFF (Manufacturer data, which is the ibeacon advertisement) 4C000215...
0x03 (Complete 16b Services) 0x80FF
0x16 (16b Service Data) 0xBBAA60000000
0x09 (Complete Local Name) 
```

using bleak in device.details I see
```{
  'props': {
    'Name': ...
    'Alias': ...
    'UUIDs': ['0000ff80-0000-1000-8000-00805f9b34fb'], 
    'ManufacturerData': {76: 0x0215...}, 
    'ServiceData': {'0000aabb-0000-1000-8000-00805f9b34fb': 0x60000000}, 
  }
}
```

so it looks like all the data is available but is not clear to me the format. 
- UUIDs contains what it looks like the 0x03 (Complete 16b Services), is there any explanation on how it is converted?
 - ManufacturerData is clear, and i guess that 76 (0x4c) is the first byte of the manufacturer data, but the second byte 00 is not present in the manufacturer data and that contains just the ibeacon packet
 - Service data looks like the 0x16 service data 0xBBAA600000000 but I don't understand why it is split as uid:value
 - alias or name i guess it contains the 0x09 complete local name

is there any information that you can point me to that describe the format in device.details? and how to reason about the advertisement packet and the UUIDs, ServiceData that i receive?

thanks!

dlech, 2021-01-14:
> * UUIDs contains what it looks like the 0x03 (Complete 16b Services), is there any explanation on how it is converted?

In Bleak (and BlueZ, I guess), all UUIDs are converted to 128-bit UUID strings. So 16-bit UUIDs become `0000????-0000-1000-8000-00805f9b34fb` where `????` is the 16-bit UUID.

> * ManufacturerData is clear, and i guess that 76 (0x4c) is the first byte of the manufacturer data, but the second byte 00 is not present in the manufacturer data and that contains just the ibeacon packet

The second byte of the manufacturer data is also part of the vendor ID. It is a 16-bit little endian value.

> * Service data looks like the 0x16 service data 0xBBAA600000000 but I don't understand why it is split as uid:value

This makes for easy lookup of values.

> * alias or name i guess it contains the 0x09 complete local name

Alias can be overridden, so we use Name for the LocalName.

matteobertozzi, 2021-01-14:
cool, now it's way more clear.
thank you for taking the time to help out.

closed 2021-01-14 by matteobertozzi