hbldh/bleak #419: Interface Concern: Notification handles

xloem, 2021-01-09:
* bleak version: 364f76a34f06ccd03eb8119f774fc3d3fdfa8ddf

### Description

The current API [provides the characteristic handle as the first argument to notification callbacks](https://github.com/hbldh/bleak/blob/develop/bleak/backends/client.py#L211).

One of the things I'm experimenting with is porting Bleak to Android, and on Android there is unfortunately no access to the characteristic handles, only the UUIDs.

It seems it would make sense to provide the BleakGATTCharacteristic object here, instead of the integer handle.  Then the user could ask for the handle or the uuid as they needed.  Any thoughts?

dlech, 2021-01-09:
How about using the the instance ID. My guess is that it is the handle value. But even if it isn't it serves the same purpose.

https://developer.android.com/reference/android/bluetooth/BluetoothGattCharacteristic#getInstanceId()

xloem, 2021-01-09:
Thanks.  The documentation doesn't mention handles, but searching quickly I eventually found https://android.googlesource.com/platform/packages/apps/Bluetooth/+/master/src/com/android/bluetooth/gatt/GattService.java#3080 which implies they are indeed handles.

closed 2021-01-09 by xloem

xloem, 2021-01-10:
[work is at https://github.com/xloem/bleak/tree/p4a .  taking a break from it for now, some bug with notifications, been working on this for days.]

xloem, 2021-01-14:
@dlech i'm noticing there's some ambiguity around what to do when multiple scanner or client objects are made.  Is there a norm on whether this should throw and error or broadcast events to all objects, and if there's an error whether it should be in the constructor or in the connect/scan methods?  [for now i will implement errors in connect/scan methods]

dlech, 2021-01-14:
Hmm... I think for now that is undefined behavior.