hbldh/bleak #423: QWindowsContext: OleInitialize() failed:  "COM error 0xffffffff80010106 RPC_E_CHANGED_MODE (Unknown error 0x080010106)"

edward-jw-yu, 2021-01-16:
* bleak version: 0.9.1
* Python version: 3.7
* Operating System: win10
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description
I am using bleak in PyQt5 projects on win10
and after import bleak, the console printed 'QWindowsContext: OleInitialize() failed:  "COM error 0xffffffff80010106 RPC_E_CHANGED_MODE (Unknown error 0x080010106)"' when runing the PyQt5 project.
This caused my program failed to open the QFiledialog with Native windows. The program just hanged there like dead.

### What I Did

It is preferable if an issue contains a [Miminal Workable Example](https://stackoverflow.com/help/minimal-reproducible-example).
This will otherwise be one of the first questions you will get as a response.

It is also preferable if that example is not dependent on a specific peripheral device, but can be run and
reproduced with other BLE peripherals as well.

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here as well.
```

hbldh, 2021-01-18:
I cannot help you much here, given that I never tried using Qt with Bleak. But I expect that it does not play nice with `asyncio`. Try integrating with [`qasync`](https://pypi.org/project/qasync/) to make it work.

closed 2021-02-01 by edward-jw-yu

JBrakefield, 2021-06-18:
I have this same issue happening now after reverting from Python 3.9 and Python 3.8. I get this same annoying warning if PyQt5 and bleak are both imported. My program ran fine on Python 3.9, but when trying to share my program and get the modules installed for someone else it was too difficult (pynnet installation issues and so forth). I can't remember how to get around them for python 3.9, and it is so much simpler to just use python 3.8 to get around that. However, this new warning/error is seemingly impossible to fix. I've tried everything including suppressing it with 'warnings' module (didn't work).

We either need a fix for this issue or a way to make things install more smoothly with python 3.9.

dlech, 2021-06-18:
Related issue (for pywinrt backend): https://github.com/microsoft/xlang/issues/690

I think the solution is probably to make sure Qt calls and Bleak calls each stay on their own thread.

dlech, 2021-06-25:
Also see https://github.com/hbldh/bleak/issues/580#issuecomment-867928327 for another possible workaround.

JBrakefield, 2021-06-25:
> Also see [#580 (comment)](https://github.com/hbldh/bleak/issues/580#issuecomment-867928327) for another possible workaround.

I tried adding `from winrt import _winrt

_winrt.unint_apartment()`

but python says "AttributeError: module 'winrt._winrt' has no attribute 'unint_apartment'"

dlech, 2021-06-25:
Looks like I spelled "uninit" wrong.

JBrakefield, 2021-06-25:
ahh thanks! I think this fixed it!

bstarbuck3, 2021-12-25:
from bleak_winrt import _winrt
_winrt.uninit_apartment()

dlech, 2022-01-26:
FYI, this workaround is no longer needed since Bleak v0.14.x.