hbldh/bleak #425: What happens to the response?

littleblackfish, 2021-01-17:
I am new to BLE so excuse me if I'm missing a point here.

I understand there is a parameter so I can do `write_gatt_char(uuid, value, response=True)` to do a "write-with-response", but how do I actually get the response? 

No return, and I tried reading the same character back, but the expected response (per specs of this device) is not there.

dlech, 2021-01-17:
The response is handled by the OS. The async function will not return until the response is received. The response doesn't contain any useful data. So it is probably better to think of it as "wait for acknowledgement of receipt before continuing".

littleblackfish, 2021-01-17:
Thank, you. I am trying to map the attached scheme from the specs of my device of interest to this library. I am writing [01 00] to the mentioned PMD control point, but since `write_gatt_char()` does not return anything, where would you suggest I look for the response? 

[image]

Carglglz, 2021-01-18:
@littleblackfish I think you need to read the response from PMD control point `read_gatt_char(PMD_CHAR_UUID)` after writing to it.

littleblackfish, 2021-01-18:
Thank you for the input, this is what I thought, although it appears not to be the case. The same documentation specifies what happens when your read the PMD_CHAR, and that respond does not appear to change after writing to this UUID.

[image]

dlech, 2021-01-18:
did you enable notifications?

littleblackfish, 2021-01-18:
Ah, thank you, that sounds like a very viable solution. I will report back as soon as I can try (device is not with me at the moment). 

I realize this is primarily my lack of experience with BLE rather than a library issue but guidance is much appreciated.

littleblackfish, 2021-01-19:
Yes! This was the way. Thank you.

closed 2021-01-25 by hbldh