hbldh/bleak #427: importing bleak exits script

andrew-playbrush, 2021-01-18:
* bleak version: 0.10.0
* Python version:3.8.5
* Operating System:buntu 20.04
* BlueZ version (`bluetoothctl -Uv`) in case of Linux:5.53

### Description

### What I Did

Example code:

``` 
#!/usr/bin/env python3

print("Here")

from bleak import discover

print("Not here")

```
Here

```
Exit code is 0

hbldh, 2021-01-18:
Strange. Could you try running the `discover.py` example script and see if that works?

andrewtholt, 2021-01-18:
It does.  I can run this ok on another machine, I checked the version of python, etc as you listed them and it's the same and works.

To my, the weird thing is the complete absence of any messages.  

Thanks for the rapid response.

hbldh, 2021-01-18:
Maybe it just fails to flush the stdout buffer before closing the script? Try
```python
#!/usr/bin/env python3

print("Here", flush=True)

from bleak import discover

print("Not here", flush=True)
```

andrewtholt, 2021-01-19:
Forgot to mention, I have it working fine in a venv on another machine.

Python version 3.7.3

andrewtholt, 2021-01-19:
Cured.  I wont say found.  Basically something in my environment changed, possibly caused by an upgrade over the weekend.

Beginning to appreciate python venv

Thanks for your time

Regards,
Andrew

closed 2021-01-19 by hbldh

andrewtholt, 2021-01-22:
I have found is.  Some time in the past I must have had a python script called uuid.py.  I found and removed the associated .pyc 

That seems to have fixed it