hbldh/bleak #428: Bytearray parsing difference on linux vs windows

trflorian, 2021-01-20:
* bleak version: 0.11.0a2
* Python version: 3.8.5
* Operating System: Ubuntu 20.04 LTS
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.54

### Description

There is inconsistent handling of the data byte array between Linux and Windows. The same code snippet runs on Windows just fine but on linux it doesn't work.

### What I Did

1) Setup client connection
2) Request services
3) Write to any valid gatt characteristic
`await client.write_gatt_char(characteristic,bytearray("Hello World","utf-8"),response=True)`
`await client.write_gatt_char(characteristic,b"Hello World",response=True)`

On windows both snippets work fine, but in the linux backend the second one raises an exception.

Expected behavior: either it can on both systems throw an exception or implicitly convert the bytes to a bytearray on both os.

```
sphero_mini.py 124 _send
await self.client.write_gatt_char(characteristic,output,response=True)

client.py 616 write_gatt_char
await self._bus.callRemote(

txdbus.error.MarshallingError:
List, Tuple, Bytearray, or Dictionary required for DBus array.  Received: b'\x8d\n\x13\r\x00\xd5\xd8'
```
(the data received in the log does not correspond to the Hello World string, that's just from debugging)

dlech, 2021-01-20:
FYI, this should be fixed already in the `dbus-next-2` branch.

trflorian, 2021-01-20:
thank you!
closed due to duplicate #408

closed 2021-01-20 by trflorian