hbldh/bleak #429: Connection gets lost before getting all the services from the device.

chinmayjoshi1992, 2021-01-22:
I've a device which has 10-15 services.

When I send a get_services() request it only gives me services which I get in 5 seconds after that connection losts.
Is there a way to get all the services in one go?

2021-01-22 15:55:44,225 bleak.backends.dotnet.scanner DEBUG: Received 80:E1:26:08:33:82: iPPE.
2021-01-22 15:55:44,233 bleak.backends.dotnet.scanner DEBUG: Received 80:E1:26:08:33:82: Unknown.
2021-01-22 15:55:44,233 bleak.backends.dotnet.client DEBUG: Connecting to BLE device @ 80:E1:26:08:33:82
2021-01-22 15:55:44,510 bleak.backends.dotnet.client DEBUG: _ConnectionStatusChanged_Handler: 1
2021-01-22 15:55:44,510 bleak.backends.dotnet.client DEBUG: Get Services...
2021-01-22 15:55:44,800 bleak.backends.dotnet.client INFO: Services resolved for BleakClientDotNet (80:E1:26:08:33:82)
Services:
00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile
00001800-0000-1000-8000-00805f9b34fb: Generic Access Profile
00005003-0000-1000-8000-00805f9b34fb: Vendor specific
00005002-0000-1000-8000-00805f9b34fb: Vendor specific
23586fe6-e787-11ea-adc1-0242ac120002: Unknown
00005005-0000-1000-8000-00805f9b34fb: Vendor specific
2021-01-22 15:55:44,800 bleak.backends.dotnet.client DEBUG: Disconnecting from BLE device...
2021-01-22 15:55:47,848 bleak.backends.dotnet.client DEBUG: _ConnectionStatusChanged_Handler: 0

hbldh, 2021-01-22:
Please include the code you run to get that result. I cannot help you otherwise.

chinmayjoshi1992, 2021-01-25:
Actually, I'm running the service_explorer.py example file.

So the scenario is I have one embedded device running BLE. I frequently add-Remove characteristics from the device. Earlier the script was showing the old parameters. So I ran the same script on different machines and IT was showing the correct characteristics.

Is the lib store the characteristics somewhere? If so how can I clear the same? 

Also there is one chara for which it is showing Error 0x0b.

hbldh, 2021-01-25:
The `service_explorer.py` script is designed to fetch all services, print them and then disconnect. Thus, that the connection is lost is by design.

If you check out the ``feature/caching-improvement-windows`` branch and run the script using that instead, and also changing to
```python
async with BleakClient(address, use_cached=False) as client:
    [...]
```
instead, does that solve your problem?

chinmayjoshi1992, 2021-01-25:
I checked out to feature/caching-improvement-windows branch and getting below error.

bleak.exc.BleakDotNetTaskError: System.Exception: Catastrophic failure (Exception from HRESULT: 0x8000FFFF (E_UNEXPECTED))

Also I tried use_cached=False in the earlier code still no success.

One thing I observed is, when I'm reading the characteristic value the value is getting updated correctly but there are two issues :
- The uuid of chara is wrong
- the service currently has 2 chara but it is showing 4 (retaining the previous values.)

hbldh, 2021-01-26:
I'm sorry, but the code in `feature/caching-improvement-windows` works fine for me, on all devices I have to test with.

Are you using Windows 10 version 2004? Then the crash problem might be related to #257 and not to the caching.

When it comes to the uuids of your characteristics, Bleak only relays what it gets from the OS systems, it does not change anything. What you get is what Windows reads from either your device or from its cache.

chinmayjoshi1992, 2021-01-27:
@hbldh this worked for me. I'm now able to see the new characteristics. Thanks for the help.

One thing want to mention, when I'm trying to read the chara value I'm getting this error now.

Could not get GATT characteristics for d08ec892-f569-4c10-a617-76a06781954e: ProtocolError (Error: 0x0B: Connection Already Exists)

For others script is able to get the values.

hbldh, 2021-01-27:
Good to hear that it was resolved.

The maximum length is highly dependent on the device and there is currently no way of finding out how long it is in Bleak by any other means than experimenting...

chinmayjoshi1992, 2021-01-28:
ok, thanks for the help. closing the issue.

closed 2021-01-28 by chinmayjoshi1992