hbldh/bleak #432: Error: 0x03: Hardware Failure when using write_gatt_char

pledi, 2021-01-25:
* bleak version: 0.10
* Python version: 3.7
* Operating System: Windows 10

### Description

I'm trying to write a temperature (5900) to a characteristic.
The data must fit to the following spec: 

 ```
 Size: 2 Bytes 
 Type: UINT16 Little Endian
```

I extracted these from nRF Connect and with the nRF Connect app, 
I'm also able to send a UINT 5900 as write **command** [(Screenshot)](https://user-images.githubusercontent.com/9287613/105772398-62907480-5f62-11eb-832a-529cd2827ad2.jpg) and the mug I want to control sets the temperature. 
(but it fails if I send it as write request)

### What I Did

```python
TARGET_TEMP = "fc540003-236c-4c94-8fa9-944a3e5353fa"
# temp = 5900

async def setToTemp(self, temp):
    if await self.connectedClient.is_connected():
        print("try setting the target temperature")
        newtarget = temp.to_bytes(2, 'little')
        await self.connectedClient.write_gatt_char(TARGET_TEMP, newtarget,True)
    else:
        print("not connected")
```
### Crash log: 

```
Task exception was never retrieved
future: <Task finished coro=<MainWindow.setToTemp() done, defined at C:\Users\koolb\OneDrive\Dokumente\QTEmberQuick\main.py:84> 
exception=BleakError("Could not write value b'\\x0c\\x17' to characteristic fc540003-236c-4c94-8fa9-944a3e5353fa: ProtocolError (Error: 0x03: Hardware Failure)")>
Traceback (most recent call last):
  File "C:\Users\koolb\OneDrive\Dokumente\QTEmberQuick\main.py", line 91, in setToTemp
    await self.connectedClient.write_gatt_char(TARGET_TEMP, newtarget,True)
  File "C:\Users\koolb\AppData\Local\Programs\Python\Python37-32\lib\site-packages\bleak\backends\dotnet\client.py", line 686, in write_gatt_char
    write_result.ProtocolError, "Unknown"
bleak.exc.BleakError: Could not write value b'\x0c\x17' to characteristic fc540003-236c-4c94-8fa9-944a3e5353fa: ProtocolError (Error: 0x03: Hardware Failure).
```

`\\x0c\\x17` seems to be right, since 0x170C is 5900 as UINT16.
I've also tried using `struct.pack("<H",5900)` with no luck.

hbldh, 2021-01-26:
In Windows, try [pairing with your device first, with encryption](https://bleak.readthedocs.io/en/latest/api.html#bleak.backends.dotnet.client.BleakClientDotNet.pair):

```python
async with BleakClient(address) as client:
    is_connected = await client.is_connected()
    print(f"Connected: {is_connected}")
    is_paired = await client.pair(protection_level=2)
    print(f"Paired: {is_paired}")
    # Call setToTemp here
```

In macOS, there is no pairing functionality in Bleak yet.

pledi, 2021-01-26:
Hey there, thanks for the quick reply. 
I've added the protection_level=2 parameter to the pair function, but I still get the same error. 
[Link to main.py](https://github.com/pledi/EmberControl/blob/e5f7eddbfd0b618cf294c83d21305b636a36deb5/main.py#L136)

closed 2021-01-26 by pledi

hbldh, 2021-01-27:
@pledi Look at #425 for more explanation about the "response". It is not "getting a value back" but rather a "wait for the device to acknowledge the reception of data".

mazafra1, 2024-11-17:
@hbldh Hi there!! 

It'd be super nice to get this working on macOS. Could you add it as a feature? 

Besides, does it for in Linux?
Thanks for your awesome work.