hbldh/bleak #433: Descriptor with handle 00002902-0000-1000-8000-00805f9b34fb was not found!

chinmayjoshi1992, 2021-01-27:
* bleak version: 0.10.0
* Python version: python 3.7
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

bleak.exc.BleakError: Descriptor with handle 00002902-0000-1000-8000-00805f9b34fb was not found!

### What I Did

In my device, there is one characteristic that has a descriptor with UUID 00002902-0000-1000-8000-00805f9b34fb. This basically enables the notification for that particular characteristic.

below is the code I'm running.

import asyncio
import binascii

from bleak import BleakClient

address = "80:E1:26:08:33:82"
write_desc_uuid = "00002902-0000-1000-8000-00805f9b34fb"
write_uuid = "1f45038d-3636-4e54-a018-a817b8202069"

async def run(address, debug=False) :
    if debug :
        import sys

        l = logging.getLogger("asyncio")
        l.setLevel(logging.DEBUG)
        h = logging.StreamHandler(sys.stdout)
        h.setLevel(logging.DEBUG)
        l.addHandler(h)
        logger.addHandler(h)

    async with BleakClient(address) as client :
        x = await client.is_connected()
        print("Connected: {0}".format(x))
        write_val= bytearray(b'\x01\x00')
        await client.write_gatt_descriptor(write_desc_uuid, write_val)
        await asyncio.sleep(0.2)
        print("Data Sent to device.")

loop = asyncio.get_event_loop()
loop.run_until_complete(run(address))

after running the code I'm getting bleak.exc.BleakError: Descriptor with handle 00002902-0000-1000-8000-00805f9b34fb was not found! this error.

I also ran the get all services script from the example in which I can see the same descriptor available.

chinmayjoshi1992, 2021-01-27:
I used the handle ID instead of UUID and it worked.

closed 2021-01-27 by chinmayjoshi1992

hbldh, 2021-01-27:
Ah, yes. In the Bleak code, GATT descriptors can only be [accessed by integer handle](https://github.com/hbldh/bleak/blob/develop/bleak/backends/service.py#L174-L176).

takeraparterer, 2023-09-21:
how do i get the handle id?

dlech, 2023-09-21:
The best way is to iterate the descriptor dict to find the one you need.