hbldh/bleak #436: Can't disconnect properly

diogotecelao, 2021-01-31:
* bleak version: 0.9.1
* Python version: 3.7.4
* Operating System: Windows 10

### Description

Hi all! First of all, many thanks for this awesome library!

I created a BLEInterface wrapper class which aims to make to make the Bluetooth connections easy and straight forward for my use-case. Everything works extremely well (establish connection, send and receive commands, etc), but when I try to close the connection, I fail to do so and Bleak returns a timeout error. I have debugged my Bluetooth device's firmware and it receives the disconnection event from the computer. However, apparently Bleak isn't acknowledging that and times out. This only happens with my code. I have tried to run a simpler example code and the problem doesn't exist. Hence, I assume I'm messing something up regarding asyncio.

I have created a Minimal Workable Example which should work with any Bluetooth device. It just establishes connection and then closes it.

Can anyone help me?

```
import asyncio
from threading import Thread
from bleak import BleakClient

class BLEInterface:

    def __init__(self, communication_address: str):
        """
        Initialises and sets the Bluetooth connection and runs a loop in a different thread to keep the BLEInterface
        instance running until the user closes the connection

        :param communication_address: communication address of the Bluetooth device we want to connect with
        """
        self.loop = asyncio.new_event_loop()
        # Establish the Bluetooth connection and keep the object reference for further use
        self.connection = self.loop.run_until_complete(self.establish_connection(communication_address))
        # Set the connection has open
        self.connection_open = True
        # Run the message reader loop thread
        Thread(target=self.message_reader_loop).start()

    def message_reader_loop(self):
        """
        Creates a new loop which keeps the code running until the connection is closed

        """
        # Create a new loop (running in a separate thread) to listen for messages
        loop = asyncio.new_event_loop()
        loop.run_until_complete(self.message_reader_loop_async())

    async def message_reader_loop_async(self):
        """
        Keeps the code running until the connection is closed

        """
        while self.connection_open:
            await asyncio.sleep(0.01)

    def on_incoming_bth_message(self, sender: int, data: bytearray) -> None:
        print('Received message: ', data)

    async def establish_connection(self, communication_address: str):
        """
        Establishes the Bluetooth connection with the device

        :param communication_address: communication address of the Bluetooth device we want to connect with
        """
        print('Establishing connection with the device {mac}'.format(mac=communication_address))
        # Establish connection with the device
        client = BleakClient(communication_address, loop=self.loop)
        await client.connect()
        # Set on disconnect callback
        client.set_disconnected_callback(self.on_bth_disconnect)
        print('Connection successfully established!')
        return client

    def on_bth_disconnect(self, client: BleakClient) -> None:
        self.close_connection()

    def close_connection(self):
        self.connection_open = False
        asyncio.get_event_loop().run_until_complete(self.close_device_connection_async())

    async def close_device_connection_async(self):
        try:
            print('Disconnecting from device')
            await self.connection.disconnect()
            print('Successfully disconnected from device')
        except (ValueError, Exception) as e:
            print('Error disconnecting to device')
            raise Exception(e)

if __name__ == '__main__':
    print()
    ble_interface = BLEInterface('FA:FF:5C:B3:1E:2B')
    ble_interface.close_connection()
```

diogotecelao, 2021-02-06:
Was able to fix this by myself. I'm publishing the code below. Basically, what I believe it was happening was the following. The connection with the device was being created in a given event loop A, while the disconnection was being requested on event loop B. When the device replied back with telling the disconnection was successfully, that event was received in event loop A and not in event loop B. The event loop B was waiting for the reply event and, therefore, raised the timeout.

```
import asyncio
from bleak import BleakClient

class BLEInterface:

    def __init__(self, communication_address: str):
        """
        Initialises and sets the Bluetooth connection and runs a loop in a different thread to keep the BLEInterface
        instance running until the user closes the connection

        :param communication_address: communication address of the Bluetooth device we want to connect with
        """
        self.loop = asyncio.new_event_loop()
        # Establish the Bluetooth connection and keep the object reference for further use
        self.connection = self.loop.run_until_complete(self.establish_connection(communication_address))

    def on_incoming_bth_message(self, sender: int, data: bytearray) -> None:
        print('Received message: ', data)

    async def establish_connection(self, communication_address: str):
        """
        Establishes the Bluetooth connection with the device

        :param communication_address: communication address of the Bluetooth device we want to connect with
        """
        print('Establishing connection with the device {mac}'.format(mac=communication_address))
        # Establish connection with the device
        client = BleakClient(communication_address, loop=self.loop)
        await client.connect()
        # Set on disconnect callback
        client.set_disconnected_callback(self.on_bth_disconnect)
        print('Connection successfully established!')
        return client

    def on_bth_disconnect(self, client: BleakClient) -> None:
        print('On Bluetooth disconnect')

    def close_connection(self):
        print('Closing the connection')
        self.loop.run_until_complete(self.close_device_connection_async())

    async def close_device_connection_async(self):
        try:
            print('Disconnecting from device')
            await self.connection.disconnect()
            print('Successfully disconnected from device')
        except (ValueError, Exception) as e:
            print('Error disconnecting to device')
            raise Exception(e)

if __name__ == '__main__':
    ble_interface = BLEInterface('FA:FF:5C:B3:1E:2B')
    ble_interface.close_connection()
```

closed 2021-02-06 by diogotecelao

savejeff, 2022-04-16:
I copied the code and ran it, and i get a timeout error. 

10:47:32: Services resolved for BleakClientWinRT (94:B9:7E:FE:76:8E)
Error disconnecting to device
Traceback (most recent call last):
  File "XXX/BLEHelp.py", line 422, in close_device_connection_async
    await self.connection.disconnect()
  File "C:\Users\XXX\AppData\Local\Programs\Python\Python38\lib\site-packages\bleak\backends\winrt\client.py", line 313, in disconnect
    await asyncio.wait_for(event.wait(), timeout=10)
  File "C:\Users\XXX\AppData\Local\Programs\Python\Python38\lib\asyncio\tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:
```

Traceback (most recent call last):
  File "XXX/BLEHelp.py", line 456, in <module>
    ble_interface.close_connection()
  File "XXX/BLEHelp.py", line 417, in close_connection
    self.loop.run_until_complete(self.close_device_connection_async())
  File "C:\Users\XXX\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "XXX/BLEHelp.py", line 426, in close_device_connection_async
    raise Exception(e)
Exception

```

I'm using 
bleak 0.14.2
bleak-winrt 1.1.1
windows 10
python 3.8

if have yet to find a working way to disconnect from a device without getting a timeout error

arist0v, 2022-07-12:
it's not the better way, but i use try/except to handle the timeout exception(asyncio.exceptions.TimeoutError) and just pass on it. it allow me to keep working

savejeff, 2022-07-13:
Yeah the problem is not that citical. it only happens when you are connected for not enough time. after some time disconencts work fine. seems to be a windows problem not a bleak problem

arist0v, 2022-07-13:
I work on linux amd have the issue