hbldh/bleak #438: python crash on macOS 11.0 Big Sur

lightingghost, 2021-02-04:
* bleak version: 0.10.0
* Python version: 3.8.3
* Operating System: macOS 11.0.1 Big Sur
* BlueZ version (`bluetoothctl -v`) in case of Linux: n/a

### Description

Python crashed when running discover()

### What I Did

```python
$ ipython
Python 3.8.3 (default, Jul  2 2020, 11:26:31)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.16.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import bleak as bl

In [2]: import asyncio

In [3]: devices = await bl.discover()
[1]    64147 abort      ipython
```

dlech, 2021-02-04:
What is the stack trace from the crash?

lightingghost, 2021-02-04:
> What is the stack trace from the crash?

Thank you for your reply. I was trying to attach lldb to the ipython process and the stack trace is

```
Process 80553 resuming
Process 80553 stopped
* thread #3, queue = 'com.apple.root.default-qos', stop reason = signal SIGABRT
    frame #0: 0x00007fff20333792 libsystem_kernel.dylib`__abort_with_payload + 10
libsystem_kernel.dylib`__abort_with_payload:
->  0x7fff20333792 <+10>: jae    0x7fff2033379c            ; <+20>
    0x7fff20333794 <+12>: movq   %rax, %rdi
    0x7fff20333797 <+15>: jmp    0x7fff203126a1            ; cerror_nocancel
    0x7fff2033379c <+20>: retq
(lldb) bt
* thread #3, queue = 'com.apple.root.default-qos', stop reason = signal SIGABRT
  * frame #0: 0x00007fff20333792 libsystem_kernel.dylib`__abort_with_payload + 10
    frame #1: 0x00007fff203351d9 libsystem_kernel.dylib`abort_with_payload_wrapper_internal + 80
    frame #2: 0x00007fff2033520b libsystem_kernel.dylib`abort_with_payload + 9
    frame #3: 0x00007fff24b9b1b4 TCC`__TCC_CRASHING_DUE_TO_PRIVACY_VIOLATION__ + 163
    frame #4: 0x00007fff24b9b9e6 TCC`__TCCAccessRequest_block_invoke.128 + 498
    frame #5: 0x00007fff24b9960a TCC`__tccd_send_message_block_invoke + 573
    frame #6: 0x00007fff2008eab8 libxpc.dylib`_xpc_connection_reply_callout + 36
    frame #7: 0x00007fff2008ea40 libxpc.dylib`_xpc_connection_call_reply_async + 69
    frame #8: 0x00007fff2019b831 libdispatch.dylib`_dispatch_client_callout3 + 8
    frame #9: 0x00007fff201b4010 libdispatch.dylib`_dispatch_mach_msg_async_reply_invoke + 375
    frame #10: 0x00007fff201ab564 libdispatch.dylib`_dispatch_kevent_worker_thread + 1390
    frame #11: 0x00007fff203434e1 libsystem_pthread.dylib`_pthread_wqthread + 386
    frame #12: 0x00007fff20342467 libsystem_pthread.dylib`start_wqthread + 15
```

hbldh, 2021-02-19:
@lightingghost Questions: 
1. Does it happen consistently every time? 
2. Do you have an M1 Mac?
3. Have you successfully used Bleak before on the same computer with older macOS version?

lightingghost, 2021-02-19:
1. Yes
2. No
3. Bleak previously worked on the same Mac with 10.15

hbldh, 2021-02-19:
@lightingghost Did you install Python when on 10.15? Did you install Bleak then as well? 
If so, could you first try to create a new Bleak installation with a new pyobj installation. If that still does not work, could you try to install Python anew and use that for running you program?

lightingghost, 2021-02-22:
> @lightingghost Did you install Python when on 10.15? Did you install Bleak then as well?
> If so, could you first try to create a new Bleak installation with a new pyobj installation. If that still does not work, could you try to install Python anew and use that for running you program?

Thank you for the reply. I tried reinstall python and then bleak, it still throws the same error.

hbldh, 2021-02-25:
@lightingghost Could you please run the `discover.py` example script instead of calling it from ipython? Does it still throw the same exception?

iserko, 2021-02-27:
@hbldh
using bleak installed from the latest `develop` branch
on OSX 11.2.2 (Big Sur latest patch)
Python 3.8.2 (default system Python3 version)

```
% /usr/bin/python3 -V
Python 3.8.2
% /usr/bin/python3 -m venv .venv
% git rev-parse HEAD  # latest develop branch
3e79f4a718a208324f85ade61ba49982c0df6a7f
% .venv/bin/pip install -e .
Obtaining file:///Users/igor/workspace/bleak
Collecting pyobjc-core
  Using cached pyobjc_core-7.1-cp38-cp38-macosx_10_9_x86_64.whl (292 kB)
Collecting pyobjc-framework-CoreBluetooth
  Using cached pyobjc_framework_CoreBluetooth-7.1-cp36-abi3-macosx_10_9_x86_64.whl (9.8 kB)
Collecting pyobjc-framework-libdispatch
  Using cached pyobjc_framework_libdispatch-7.1-cp38-cp38-macosx_10_9_x86_64.whl (12 kB)
Collecting pyobjc-framework-Cocoa>=7.1
  Using cached pyobjc_framework_Cocoa-7.1-cp38-cp38-macosx_10_9_x86_64.whl (261 kB)
Installing collected packages: pyobjc-core, pyobjc-framework-Cocoa, pyobjc-framework-libdispatch, pyobjc-framework-CoreBluetooth, bleak
  Running setup.py develop for bleak
Successfully installed bleak pyobjc-core-7.1 pyobjc-framework-Cocoa-7.1 pyobjc-framework-CoreBluetooth-7.1 pyobjc-framework-libdispatch-7.1

# now we actually run the example script
% .venv/bin/python -X faulthandler examples/discover.py
Fatal Python error: Aborted

Thread 0x000000011413be00 (most recent call first):
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/threading.py", line 306 in wait
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/threading.py", line 558 in wait
  File "/Users/igor/workspace/bleak/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 91 in init
  File "/Users/igor/workspace/bleak/bleak/backends/corebluetooth/scanner.py", line 37 in __init__
  File "/Users/igor/workspace/bleak/bleak/backends/scanner.py", line 83 in discover
  File "examples/discover.py", line 16 in run
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/asyncio/events.py", line 81 in _run
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/asyncio/base_events.py", line 1859 in _run_once
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/asyncio/base_events.py", line 570 in run_forever
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/asyncio/base_events.py", line 603 in run_until_complete
  File "examples/discover.py", line 22 in <module>
[1]    37081 abort      .venv/bin/python -X faulthandler examples/discover.py
```

iserko, 2021-02-27:
After a bit of digging I managed to figure it out. @lightingghost had `__TCC_CRASHING_DUE_TO_PRIVACY_VIOLATION__` in the Apple error report (I had the same). So basically Python didn't have permission to access Bluetooth.

What I did to fix it was to add my Terminal app (in my case iTerm2.app) into System Preferences -> Security & Privacy -> Bluetooth and I just drag and dropped the app there.

Worked like a charm and the example now works.

lightingghost, 2021-02-27:
> After a bit of digging I managed to figure it out. @lightingghost had `__TCC_CRASHING_DUE_TO_PRIVACY_VIOLATION__` in the Apple error report (I had the same). So basically Python didn't have [...]

Thanks for sharing! This works for me!

hbldh, 2021-03-01:
@iserko This is marvelous news! Thank you for reporting this back. I will include notes on this when I do #266! Will close this issue in the meantime.

closed 2021-03-01 by hbldh

paaguti, 2021-12-02:
Thanks a ton!!!