hbldh/bleak #442: InProgress error in Service Explorer + more

mileseverett, 2021-02-12:
* bleak version: 0.10.0
* Python version: 3.7.3
* Operating System:
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Attempting to pull data from a sensor in a Schwinn IC8 spin bike. Having trouble reading the data (is there any examples of reading a notify once every 0.1s?) so tried to get info on all services for an issue, ended up with this issue. 

Bluetooth device is on and no other devices are connected. I have posted a list of all of the UUIDs which I was able to pull using a different program, but unsure what the issue is. 

### What I Did

```
"""
Service Explorer
----------------
An example showing how to access and print out the services, characteristics and
descriptors of a connected GATT server.
Created on 2019-03-25 by hbldh <henrik.blidh@nedomkull.com>
"""
import platform
import asyncio
import logging

from bleak import BleakClient

async def run(address, debug=False):
    log = logging.getLogger(__name__)
    if debug:
        import sys

        log.setLevel(logging.DEBUG)
        h = logging.StreamHandler(sys.stdout)
        h.setLevel(logging.DEBUG)
        log.addHandler(h)

    async with BleakClient(address) as client:
        x = await client.is_connected()
        log.info("Connected: {0}".format(x))

        for service in client.services:
            log.info("[Service] {0}: {1}".format(service.uuid, service.description))
            for char in service.characteristics:

                if "read" in char.properties:
                    try:
                        value = bytes(await client.read_gatt_char(char.uuid))
                    except Exception as e:
                        value = str(e).encode()
                else:
                    value = None
                log.info(
                    "\t[Characteristic] {0}: (Handle: {1}) ({2}) | Name: {3}, Value: {4} ".format(
                        char.uuid,
                        char.handle,
                        ",".join(char.properties),
                        char.description,
                        value,
                    )
                )
                for descriptor in char.descriptors:
                    value = await client.read_gatt_descriptor(descriptor.handle)
                    log.info(
                        "\t\t[Descriptor] {0}: (Handle: {1}) | Value: {2} ".format(
                            descriptor.uuid, descriptor.handle, bytes(value)
                        )
                    )

if __name__ == "__main__":
    address = (
        "E5:AA:67:2B:CB:3C"
        if platform.system() != "Darwin"
        else "B9EA5233-37EF-4DD6-87A8-2A875E821C46"
    )
    loop = asyncio.get_event_loop()
    loop.set_debug(True)
    loop.run_until_complete(run(address, True))

```

```
Connected: True
[Service] 00001826-0000-1000-8000-00805f9b34fb: Fitness Machine
	[Characteristic] 00002ad9-0000-1000-8000-00805f9b34fb: (Handle: 53) (write,indicate) | Name: Fitness Machine Control Point, Value: None 
Traceback (most recent call last):
  File "/home/pi/bluetooth8.py", line 66, in <module>
    loop.run_until_complete(run(address, True))
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "/home/pi/bluetooth8.py", line 50, in run
    value = await client.read_gatt_descriptor(descriptor.handle)
  File "/usr/local/lib/python3.7/dist-packages/bleak/backends/bluezdbus/client.py", line 546, in read_gatt_descriptor
    ).asFuture(asyncio.get_event_loop())
txdbus.error.RemoteError: org.bluez.Error.InProgress: In Progress
```

When I print out all characteristics + their properties/descriptors

```
======= 00001826-0000-1000-8000-00805f9b34fb: Fitness Machine ===========
00002ad9-0000-1000-8000-00805f9b34fb: Fitness Machine Control Point ['write', 'indicate']
00002ad3-0000-1000-8000-00805f9b34fb: Training Status ['read', 'notify']
00002ad2-0000-1000-8000-00805f9b34fb: Indoor Bike Data ['notify']
00002acc-0000-1000-8000-00805f9b34fb: Fitness Machine Feature ['read']
00002ada-0000-1000-8000-00805f9b34fb: Fitness Machine Status ['notify']
======= 00001816-0000-1000-8000-00805f9b34fb: Cycling Speed and Cadence ===========
00002a55-0000-1000-8000-00805f9b34fb: SC Control Point ['write', 'indicate']
00002a5d-0000-1000-8000-00805f9b34fb: Sensor Location ['read']
00002a5c-0000-1000-8000-00805f9b34fb: CSC Feature ['read']
00002a5b-0000-1000-8000-00805f9b34fb: CSC Measurement ['notify']
======= 0000180d-0000-1000-8000-00805f9b34fb: Heart Rate ===========
00002a38-0000-1000-8000-00805f9b34fb: Body Sensor Location ['read']
00002a37-0000-1000-8000-00805f9b34fb: Heart Rate Measurement ['notify']
======= 0000180a-0000-1000-8000-00805f9b34fb: Device Information ===========
00002a50-0000-1000-8000-00805f9b34fb: PnP ID ['read']
00002a28-0000-1000-8000-00805f9b34fb: Software Revision String ['read']
00002a26-0000-1000-8000-00805f9b34fb: Firmware Revision String ['read']
00002a27-0000-1000-8000-00805f9b34fb: Hardware Revision String ['read']
00002a24-0000-1000-8000-00805f9b34fb: Model Number String ['read']
00002a29-0000-1000-8000-00805f9b34fb: Manufacturer Name String ['read']
======= 00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile ===========
```

Above was generated with the following code:

```
"""
Service Explorer
----------------
An example showing how to access and print out the services, characteristics and
descriptors of a connected GATT server.
Created on 2019-03-25 by hbldh <henrik.blidh@nedomkull.com>
"""
import platform
import asyncio
import logging

from bleak import BleakClient

async def run(address, debug=False):
    log = logging.getLogger(__name__)
    if debug:
        import sys

        log.setLevel(logging.DEBUG)
        h = logging.StreamHandler(sys.stdout)
        h.setLevel(logging.DEBUG)
        log.addHandler(h)

    async with BleakClient(address) as client:
        x = await client.is_connected()
        log.info("Connected: {0}".format(x))

        for service in client.services:
            log.info("[Service] {0}: {1}".format(service.uuid, service.description))
            for char in service.characteristics:

                if "read" in char.properties:
                    try:
                        value = bytes(await client.read_gatt_char(char.uuid))
                    except Exception as e:
                        value = str(e).encode()
                else:
                    value = None
                log.info(
                    "\t[Characteristic] {0}: (Handle: {1}) ({2}) | Name: {3}, Value: {4} ".format(
                        char.uuid,
                        char.handle,
                        ",".join(char.properties),
                        char.description,
                        value,
                    )
                )
                for descriptor in char.descriptors:
                    value = await client.read_gatt_descriptor(descriptor.handle)
                    log.info(
                        "\t\t[Descriptor] {0}: (Handle: {1}) | Value: {2} ".format(
                            descriptor.uuid, descriptor.handle, bytes(value)
                        )
                    )

if __name__ == "__main__":
    address = (
        "E5:AA:67:2B:CB:3C"
        if platform.system() != "Darwin"
        else "B9EA5233-37EF-4DD6-87A8-2A875E821C46"
    )
    loop = asyncio.get_event_loop()
    loop.set_debug(True)
    loop.run_until_complete(run(address, True))
```

hbldh, 2021-02-19:
@mileseverett Could you wrap the descriptor reading in `service_explorer.py` in a try clause and see if the InProgress error is present on only one descriptor:

```
try:
    value = await client.read_gatt_descriptor(descriptor.handle)
except Exception as e:
    value = str(e).encode()
```

dlech, 2021-05-19:
I was getting an "In Progress" error when trying to start notifications just now. I removed the device using `bluetoothctl` then Bleak started to work again.

hbldh, 2021-08-10:
Closing this old issue since there is nothing more to do right now.

closed 2021-08-10 by hbldh