hbldh/bleak #443: Can I discover service's characteristic UUID ?

andrealmeida98, 2021-02-14:
OS : Linux
Python: Python3

This is more a question than an issue. 
This code will print the mac and the UUID of the service. Can I use that UUID to discover its characteristics' UUIDs ?

```python
import asyncio
from bleak import BleakScanner

async def run():
    devices = await BleakScanner.discover()
    for d in devices:
            print(d)
            
loop = asyncio.get_event_loop()
loop.run_until_complete(run())

```

koenvervloesem, 2021-02-14:
I think you're looking for the [service explore example](https://github.com/hbldh/bleak/blob/develop/examples/service_explorer.py).

dlech, 2021-02-14:
> This is more a question than an issue.

FYI, there is a new [Discussions](https://github.com/hbldh/bleak/discussions) area for this sort of thing.

closed 2021-02-14 by dlech