hbldh/bleak #445: bleak.exc.BleakError: This service is already present in this BleakGATTServiceCollection!

GeorgeHowardSkub, 2021-02-17:
* bleak version: 0.10.0
* Python version: 3.7.9
* Operating System: Windows 10

### Description

I am trying to connect to my esp32 through BLE, the esp is running micropython with the standard micropython BLE library. I could already connect the device with my phone using Serial bluetooth interface app (https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=nl&gl=US) and send data back and forth.

Now I am trying to connect it to my pc using Bleak. I got some errors at my first attempts saying that the write and read characteristics UUID were wrong. Now I tried the service_explorer example (https://github.com/hbldh/bleak/blob/develop/examples/service_explorer.py) to find these UUID's. But the error: This service is already present in this BleakGATTServiceCollections comes up, this error also appeared in my first attempts to connect the devices. I searched for this error but didn't find any mentions of it.

### What I Did

This is basically the service_explorer example with the esp32 address filled in.
I```
async def run(address, debug=False):
    log = logging.getLogger(__name__)
    if debug:
        import sys

        log.setLevel(logging.DEBUG)
        h = logging.StreamHandler(sys.stdout)
        h.setLevel(logging.DEBUG)
        log.addHandler(h)

    async with BleakClient(address) as client:
        x = await client.is_connected()
        log.info("Connected: {0}".format(x))
        for service in client.services:
            log.info("[Service] {0}: {1}".format(service.uuid, service.description))
            for char in service.characteristics:
                if "read" in char.properties:
                    try:
                        value = bytes(await client.read_gatt_char(char.uuid))
                    except Exception as e:
                        value = str(e).encode()
                else:
                    value = None
                log.info(
                    "\t[Characteristic] {0}: (Handle: {1}) ({2}) | Name: {3}, Value: {4} ".format(
                        char.uuid,
                        char.handle,
                        ",".join(char.properties),
                        char.description,
                        value,
                    )
                )
                for descriptor in char.descriptors:
                    value = await client.read_gatt_descriptor(descriptor.handle)
                    log.info(
                        "\t\t[Descriptor] {0}: (Handle: {1}) | Value: {2} ".format(
                            descriptor.uuid, descriptor.handle, bytes(value)
                        )
                    )

if __name__ == "__main__":
    address = (
        "BC:DD:C2:C7:93:8E"
        if platform.system() != "Darwin"
        else "B9EA5233-37EF-4DD6-87A8-2A875E821C46"
    )
    print(address)
    loop = asyncio.get_event_loop()
    loop.set_debug(True)
    loop.run_until_complete(run(address, True))
```

```
PS C:> python .\service_explorer.py
BC:DD:C2:C7:93:8E
Traceback (most recent call last):
  File ".\service_explorer.py", line 58, in <module>
    loop.run_until_complete(run(address, True))
  File "C:\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 587, in run_until_complete
    return future.result()
  File ".\service_explorer.py", line 18, in run
    async with BleakClient(address) as client:
  File "C:\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\client.py", line 59, in __aenter__
    await self.connect()
  File "C:\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\dotnet\client.py", line 238, in connect
    await self.get_services()
  File "C:\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\dotnet\client.py", line 440, in get_services
    self.services.add_service(BleakGATTServiceDotNet(service))
  File "C:\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\service.py", line 116, in add_service
    "This service is already present in this BleakGATTServiceCollection!"
bleak.exc.BleakError: This service is already present in this BleakGATTServiceCollection!
```
Thank you in advance!

dlech, 2021-02-17:
Probably related to #362.

hbldh, 2021-02-19:
Yes, this should be solved, I will get on this as soon as possible.

commit e857fd63 (2021-02-25): Changed Service identification to handles for Windows backend.

Fixes #445 for Windows
Fixes #362 for Windows
PR #449 (2021-03-01): Changed Service identification to handles instead of UUID
Identifying Services by UUID is wrong, since a peripheral can have multiple services with the same UUID. The proper identifier should be the handle instead. From previously having thought the service handles was not present in the APIs that Bleak use, I have now done a depper investigation and foudn that I was wrong, they were there and they were quite easily obtained as well.

This should once and for all prevent the `This service is already present in this BleakGATTServiceCollection!` error from occuring for users.

- [X] Windows back [... cut]
commit 4a3f8620 (2021-03-01): Changed Service identification to handles instead of UUID (#449)

Changed Service identification to handles for Windows pythonnet backend.
Changed Service identification to handles for BlueZ txdbus backend
Implement service handle for Core Bluetooth
Updated CHANGELOG.rst

Fixes #445
Fixes #362
closed 2021-03-01 by hbldh