hbldh/bleak #466: Feature request: BLE peripheral / advertising

mrquincle, 2021-02-26:
* bleak version: n/a
* Python version: 3.8.5
* Operating System: Linux V 5.4.0-65-generic
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.53

### Description

This is an awesome library! What would be a nice is either code for or a statement on the support for playing the role of BLE peripheral. Foremost, is it in the foreseeable future that the lib would support advertising/broadcasting.

On Linux that would be via the dbus/bluez backend through the `org.bluez.LEAdvertisingManager1` interface. 

### What I Did

I have looked around in the code and the issues, but I haven't seen a statement on support for or against this feature. Hopefully, you'll be in favor, but I can understand if you're not. It's of course quite some work and I'm not sure how easy it is to build it cross-platform. A statement would be nice though. :-)

dlech, 2021-02-26:
See #81

mrquincle, 2021-02-26:
Ah, thanks! Hopefully this question has then some keywords so people won't ask again. :-)

And who knows! You might find a use case some time.

closed 2021-02-26 by mrquincle