hbldh/bleak #468: Connect on raspberry pi causes an exception

Szymon-Gesicki, 2021-02-28:
* bleak version: 0.10.0
* Python version: 3.7.3
* Operating System: Raspbian GNU/Linux 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

When I try to connect to some ble device I get this exception;
```
  File "test.py", line 9, in print_services
    async with BleakClient(device) as client:
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/client.py", line 59, in __aenter__
    await self.connect()
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 150, in connect
    raise BleakError(str(e))
bleak.exc.BleakError: org.bluez.Error.Failed: Input/output error
```.  

I can discover all devices but i can't connect to any device.

hbldh, 2021-03-02:
Can you connect through any other means? `gatttool` or `bluetoothctl`?

Also, if you have paired with devices [this](https://raspberrypi.stackexchange.com/questions/92926/trouble-connecting-bluetooth-keyboard-after-updating-and-upgrading-raspbian) might be of help?

Szymon-Gesicki, 2021-03-03:
the problem turned out to be bluetooth on raspberry 400

closed 2021-03-03 by Szymon-Gesicki