hbldh/bleak #472: BluezDbus backend: _cleanup_all() on disconnected device prints errors to log

devbis, 2021-03-03:
* bleak version: develop
* Python version: 3.7.9
* Operating System: OpenWrt
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

One of my devices disconnects after several seconds after connection. I connect and subscribe to a char, and after some time it disconnects. It is the expected behavior. But every time bleak runs client.stop_notify() it prints
```
ERROR:bleak.backends.bluezdbus.client:Failed to stop notifications on characteristic 53 (/org/bluez/hci0/dev_A4_C1_38_71_CE_E7): Not connected
```
I suggest avoid stop_notify() calls on disconnected devices because there is no way to avoid these errors in the log.

hbldh, 2021-03-03:
We will have to investigate this for a bit. I do not want the notifications callbacks to linger in the BlueZ backend so that they are present when reconnecting, so we have to see if we can still remove it somehow.

dlech, 2021-03-03:
I had a feeling that might cause an issue. We should probably allow `stop_notify()` to be called at any time and multiple times the same way we do for `disconnect()` without raising an error.

dlech, 2021-03-11:
I've been digging into BlueZ StopNotify more (unrelated to Bleak). I've learned that BlueZ "remembers" which notifications were enabled when a device is disconnected and starts those same notifications again when the device is connected again (even before ServicesResolved is set to True!).

But since Bleak closes the D-Bus message bus, this resets the state in BlueZ and we don't get the behavior described above. As such, we can just drop the call to `stop_notify()` from the cleanup code.

dlech, 2021-03-11:
>     * BlueZ version (`bluetoothctl -v`) in case of Linux:

What version of BlueZ do you have? I don't see that the reported error is possible for StopNotify.

devbis, 2021-03-11:
It is `5.50`

dlech, 2021-03-11:
Thanks. I forgot that the error was coming from Bleak... BlueZ has an error with the exact same text.

PR #479 (2021-03-16): BlueZ: Fixed stop_notify() error logged on device disconnect.
This removes the call to stop_notify() in the cleanup code path in the BlueZ backend. This was causing an error message to always be logged when notifications were enabled and a device was disconnected.

As described in the code comment. This call is not necessary and can be removed.

Fixes: #472

(No changelog since this bug was introduced in changes since the previous release)
commit 24003630 (2021-03-16): Merge pull request #479 from hbldh/bug/bluez-stop-notify

This removes the call to stop_notify() in the cleanup code path in the
BlueZ backend. This was causing an error message to always be logged
when notifications were enabled and a device was disconnected.

As described in the code comment. This call is not necessary and can be
removed.

Fixes #472
commit e0a42e4d (2021-03-16): BlueZ: Fixed stop_notify() error logged on device disconnect.

This removes the call to stop_notify() in the cleanup code path in the
BlueZ backend. This was causing an error message to always be logged
when notifications were enabled and a device was disconnected.

As described in the code comment. This call is not necessary and can be
removed.

Fixes: #472

(No changelog since this bug was introduced in changes since the
previous release)
closed 2021-03-16 by hbldh