hbldh/bleak #473: Missing service in service discovery on Windows versus Linux

liamw9534, 2021-03-03:
I am seeing a difference in service discovery behaviour between Windows 10 (.net) and Linux (dbus) platforms.  Running the script ``service_explorer.py``, I am obtaining the following output in Linux against my target device:

```
Connected: <bound method BleakClientBlueZDBus.is_connected of <BleakClientBlueZDBus, CE:BD:65:D5:4F:D2, <bleak.backends.bluezdbus.client.BleakClientBlueZDBus object at 0x7fd112c31da0>>>
[Service] 0000fe20-cc7a-482a-984a-7f2ed5b3e58f: Emerson
	[Characteristic] 0000fe24-cc7a-482a-984a-7f2ed5b3e58f: August Home Inc (write), Value: None
	[Characteristic] 0000fe23-cc7a-482a-984a-7f2ed5b3e58f: Zoll Medical Corporation (read,notify), Value: b'\x0f'
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: Client Characteristic Configuration) | Value: b'\x00\x00'
	[Characteristic] 0000fe22-cc7a-482a-984a-7f2ed5b3e58f: Zoll Medical Corporation (read,write), Value: b'\x00\x00\x00\x00'
[Service] 6e400001-b5a3-f393-e0a9-e50e24dcca9e: Nordic UART Service
	[Characteristic] 6e400003-b5a3-f393-e0a9-e50e24dcca9e: Nordic UART TX (notify), Value: None
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: Client Characteristic Configuration) | Value: b'\x00\x00'
	[Characteristic] 6e400002-b5a3-f393-e0a9-e50e24dcca9e: Nordic UART RX (write-without-response,write), Value: None
[Service] 00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile
```
Running the same script under Windows you will note that the 0000fe20-cc7a-482a-984a-7f2ed5b3e58f service is missing:

```
[Service] 00001800-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Access Profile
        [Characteristic] 00002a00-0000-1000-8000-00805f9b34fb (Handle: 2): Device Name (read,write), Value: b'GenTracker'
        [Characteristic] 00002a01-0000-1000-8000-00805f9b34fb (Handle: 4): Appearance (read), Value: b'\x00\x00'
        [Characteristic] 00002a04-0000-1000-8000-00805f9b34fb (Handle: 6): Peripheral Preferred Connection Parameters (read), Value: b'\x10\x00<\x00\x00\x00\x90\x01'
        [Characteristic] 00002aa6-0000-1000-8000-00805f9b34fb (Handle: 8): Central Address Resolution (read), Value: b'\x01'
[Service] 00001801-0000-1000-8000-00805f9b34fb (Handle: 10): Generic Attribute Profile
[Service] 6e400001-b5a3-f393-e0a9-e50e24dcca9e (Handle: 11): Nordic UART Service
        [Characteristic] 6e400002-b5a3-f393-e0a9-e50e24dcca9e (Handle: 12): Nordic UART RX (write-without-response,write), Value: None
        [Characteristic] 6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 14): Nordic UART TX (notify), Value: None
                [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 16): Client Characteristic Configuration) | Value: b'\x00\x00'
```
I have tried using both `master` and `develop` branches of `bleak` on Windows (even with caching disabled) and it didn't help.

liamw9534, 2021-03-03:
I suspect the issue is in the .net framework since I used Bluetooth LE Explorer and got the same result:

[image]

hbldh, 2021-03-03:
The Linux backend does things a bit different than other BLE solutions. It internalizes some services, for example the battery service and the device information in the `00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile` as well I think (@dlech might know more about this).

Or are you looking for the stuff in the `0000fe20-cc7a-482a-984a-7f2ed5b3e58f: Emerson` service specifically? Why that isn't present I do not know...

hbldh, 2021-03-03:
Could you please run the service explorer (in Linux) with the current `develop` branch and publish the printout there as well, so the handles of all components are visible.

liamw9534, 2021-03-03:
Hi there,

It isn't a "bleak" issue from what I can tell.  The same issue arises using
a windows BLE explorer tool which must be using the same underlying Windows
BLE API.  Your shim layer is just reporting up whatever the windows
framework finds.

Liam.

liamw9534, 2021-03-03:
Just FYI - the aforementioned service was being added as a secondary service in our target device rather than primary.  It looks like the respective bluetooth stacks handle this differently between Windows and Linux.  Changing it to primary and we can now discover it on Windows too.

closed 2021-03-03 by liamw9534