hbldh/bleak #476: Error with Bleak and Kivy [TypeError: __import__() takes at least 1 argument (0 given)]

BioMycoBit, 2021-03-07:
* bleak version: 0.11.0a3
* Python version: 3.8.6
* Operating System: Windows 10

Henrik:

I came across this error today while trying to combine kivy and bleak.

Example Code
https://raw.githubusercontent.com/kivy/kivy/master/examples/async/asyncio_basic.py

The below code works fine

```
import asyncio

from kivy.app import async_runTouchApp
from kivy.lang.builder import Builder

kv = '''
BoxLayout:
    orientation: 'vertical'
    Button:
        id: btn
        text: 'Press me'
    BoxLayout:
        Label:
            id: label
            text: 'Button is "{}"'.format(btn.state)
'''

async def run_app_happily(root, other_task):
    '''This method, which runs Kivy, is run by the asyncio loop as one of the
    coroutines.
    '''
    # we don't actually need to set asyncio as the lib because it is the
    # default, but it doesn't hurt to be explicit
    await async_runTouchApp(root, async_lib='asyncio')  # run Kivy
    print('App done')
    # now cancel all the other tasks that may be running
    other_task.cancel()

async def waste_time_freely():
    '''This method is also run by the asyncio loop and periodically prints
    something.
    '''
    try:
        while True:
            print('Sitting on the beach')
            await asyncio.sleep(2)
    except asyncio.CancelledError as e:
        print('Wasting time was canceled', e)
    finally:
        # when canceled, print that it finished
        print('Done wasting time')

if __name__ == '__main__':
    def root_func():
        '''This will run both methods asynchronously and then block until they
        are finished
        '''
        root = Builder.load_string(kv)  # root widget
        other_task = asyncio.ensure_future(waste_time_freely())
        return asyncio.gather(run_app_happily(root, other_task), other_task)

    loop = asyncio.get_event_loop()
    loop.run_until_complete(root_func())
    loop.close()
```

**TypeError: import() takes at least 1 argument (0 given) is thrown when bleak is imported**

```
import asyncio
import kivy

from kivy.app import async_runTouchApp
from kivy.lang.builder import Builder

from bleak import BleakClient, BleakError

kivy.require("1.10.1")

kv = '''
BoxLayout:
    orientation: 'vertical'
    Button:
        id: btn
        text: 'Press me'
    BoxLayout:
        Label:
            id: label
            text: 'Button is "{}"'.format(btn.state)
'''

async def run_app_happily(root, other_task):
    '''This method, which runs Kivy, is run by the asyncio loop as one of the
    coroutines.
    '''
    # we don't actually need to set asyncio as the lib because it is the
    # default, but it doesn't hurt to be explicit
    await async_runTouchApp(root, async_lib='asyncio')  # run Kivy
    print('App done')
    # now cancel all the other tasks that may be running
    other_task.cancel()

async def waste_time_freely():
    '''This method is also run by the asyncio loop and periodically prints
    something.
    '''
    try:
        while True:
            print('Sitting on the beach')
            await asyncio.sleep(2)
    except asyncio.CancelledError as e:
        print('Wasting time was canceled', e)
    finally:
        # when canceled, print that it finished
        print('Done wasting time')

if __name__ == '__main__':
    def root_func():
        '''This will run both methods asynchronously and then block until they
        are finished
        '''
        root = Builder.load_string(kv)  # root widget
        other_task = asyncio.ensure_future(waste_time_freely())
        return asyncio.gather(run_app_happily(root, other_task), other_task)

    loop = asyncio.get_event_loop()
    loop.run_until_complete(root_func())
    loop.close()
```
```
C:\repos\BioController\venv\Scripts\python.exe C:/repos/BioController/ex_kivy_async.py
[INFO   ] [Logger      ] Record log in C:\Users\Brandon\.kivy\logs\kivy_21-03-06_27.txt
[INFO   ] [deps        ] Successfully imported "kivy_deps.angle" 0.3.0
[INFO   ] [deps        ] Successfully imported "kivy_deps.glew" 0.3.0
[INFO   ] [deps        ] Successfully imported "kivy_deps.sdl2" 0.3.1
[INFO   ] [Kivy        ] v2.0.0
[INFO   ] [Kivy        ] Installed at "C:\repos\BioController\venv\lib\site-packages\kivy\__init__.py"
[INFO   ] [Python      ] v3.8.6 (tags/v3.8.6:db45529, Sep 23 2020, 15:52:53) [MSC v.1927 64 bit (AMD64)]
[INFO   ] [Python      ] Interpreter at "C:\repos\BioController\venv\Scripts\python.exe"
[INFO   ] [Factory     ] 186 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil (img_ffpyplayer ignored)
 Traceback (most recent call last):
   File "C:/repos/BioController/ex_kivy_async.py", line 67, in <module>
     loop.run_until_complete(root_func())
   File "C:/repos/BioController/ex_kivy_async.py", line 61, in root_func
     root = Builder.load_string(kv)  # root widget
   File "C:\repos\BioController\venv\lib\site-packages\kivy\lang\builder.py", line 404, in load_string
     widget = Factory.get(parser.root.name)(__no_builder=True)
   File "C:\repos\BioController\venv\lib\site-packages\kivy\factory.py", line 153, in __getattr__
     module = __import__(
 TypeError: __import__() takes at least 1 argument (0 given)

Process finished with exit code 1
```

BioMycoBit, 2021-03-08:
I found a thread on the issue

[https://www.reddit.com/r/kivy/comments/fxdqos/not_possible_to_use_bleak_module_with_kivy/fmts96w?utm_source=share&utm_medium=web2x&context=3](https://www.reddit.com/r/kivy/comments/fxdqos/not_possible_to_use_bleak_module_with_kivy/fmts96w?utm_source=share&utm_medium=web2x&context=3)

_Kivy is trying to decide what window provider to use from the ones available, and it fails because _ _ import _ _ () needs an argument. That is often because one of the libraries (maybe here, Bleak) overwrites _ _ import _ _ () confusing the rest of the code that relies on the standard implementation._

BioMycoBit, 2021-03-08:
I found these two prior threads on the issue of integrating kivy and bleak.

**Error when using bleak with kivy #176**

**Kivy app not created when importing bleak module #6816**

_It is a bit nasty to have an import line in the middle of your but it is the only way that I could make it work so far. Let me know if you have already found another way._
#6816 included a "bit nasty" hack by @bardiabarabadi

Both #176 and #6816 are closed so I'll close this one as well.

closed 2021-03-08 by BioMycoBit

hbldh, 2021-03-16:
@BioMycoBit Thank you! I will try rework this to use in #266 as a Kivy example.

BioMycoBit, 2021-03-19:
@hbldh glad I could help you out.