hbldh/bleak #484: BluezDbus backend: 2nd call to start_notify() hangs.

vliedel, 2021-03-16:
* bleak version: 0.10.0
* Python version: 3.7
* Operating System: Ubuntu 18.04
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.48

### Description

`start_notify()` hangs.

### What I Did

Simply try to start, stop, and start notifications on the same characteristic. I haven't tried this with other devices, nor did i try workarounds yet.

```python
import asyncio
import logging

from bleak import BleakClient

address = "f4:60:ef:35:22:ec"
characteristic = "24f0000d-7d10-4805-bfc1-7663a01c3bff"

logging.basicConfig(format='%(levelname)-7s: %(message)s', level=logging.DEBUG)

def handleNotification(cHandle, data):
	print(f"cHandle={cHandle} data={data}")

async def run(address):
	async with BleakClient(address) as client:
		print("connect")
		await client.connect()

		print("get_services")
		await client.get_services()

		print("start_notify")
		await client.start_notify(characteristic, handleNotification)

		print("stop_notify")
		await client.stop_notify(characteristic)

		print("start_notify")
		await client.start_notify(characteristic, handleNotification)

		print("stop_notify")
		await client.stop_notify(characteristic)

loop = asyncio.get_event_loop()
loop.run_until_complete(run(address))
print("Done")
```

Output:
```
$ python3.7 bleak-test.py 
DEBUG  : Using selector: EpollSelector
DEBUG  : 20-D1-7A-A8-DD-FB, 20:D1:7A:A8:DD:FB (-88 dBm), Object Path: /org/bluez/hci0/dev_20_D1_7A_A8_DD_FB
DEBUG  : 4B-39-40-A8-FC-C5, 4B:39:40:A8:FC:C5 (-87 dBm), Object Path: /org/bluez/hci0/dev_4B_39_40_A8_FC_C5
DEBUG  : FB-AA-1D-2E-DC-C5, FB:AA:1D:2E:DC:C5 (-93 dBm), Object Path: /org/bluez/hci0/dev_FB_AA_1D_2E_DC_C5
DEBUG  : 48-9D-B4-E3-2E-BB, 48:9D:B4:E3:2E:BB (-92 dBm), Object Path: /org/bluez/hci0/dev_48_9D_B4_E3_2E_BB
DEBUG  : F4-60-EF-35-22-ED, F4:60:EF:35:22:ED (-53 dBm), Object Path: /org/bluez/hci0/dev_F4_60_EF_35_22_ED
DEBUG  : 41-5D-6D-99-E2-69, 41:5D:6D:99:E2:69 (-69 dBm), Object Path: /org/bluez/hci0/dev_41_5D_6D_99_E2_69
DEBUG  : 74-3E-7B-52-7C-33, 74:3E:7B:52:7C:33 (-91 dBm), Object Path: /org/bluez/hci0/dev_74_3E_7B_52_7C_33
DEBUG  : BART, F4:60:EF:35:22:EC (-52 dBm), Object Path: /org/bluez/hci0/dev_F4_60_EF_35_22_EC
DEBUG  : 20-D1-7A-A8-DD-FB, 20:D1:7A:A8:DD:FB (-88 dBm), Object Path: /org/bluez/hci0/dev_20_D1_7A_A8_DD_FB
DEBUG  : InterfacesRemoved, org.freedesktop.DBus.ObjectManager (/): ['/org/bluez/hci0/dev_20_D1_7A_A8_DD_FB', ['org.freedesktop.DBus.Properties', 'org.freedesktop.DBus.Introspectable', 'org.bluez.Device1']]
DEBUG  : None, None (None dBm), Object Path: None
DEBUG  : 4B-39-40-A8-FC-C5, 4B:39:40:A8:FC:C5 (-87 dBm), Object Path: /org/bluez/hci0/dev_4B_39_40_A8_FC_C5
DEBUG  : InterfacesRemoved, org.freedesktop.DBus.ObjectManager (/): ['/org/bluez/hci0/dev_4B_39_40_A8_FC_C5', ['org.freedesktop.DBus.Properties', 'org.freedesktop.DBus.Introspectable', 'org.bluez.Device1']]
DEBUG  : None, None (None dBm), Object Path: None
[... 142 lines cut ...]
	/org/bluez/hci0/dev_F4_60_EF_35_22_EC/service000a/char000b/desc000d
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
connect
DEBUG  : Connecting to BLE device @ f4:60:ef:35:22:ec with hci0
DEBUG  : Connection successful.
get_services
start_notify 1
DEBUG  : DBUS: path: /org/bluez/hci0/dev_F4_60_EF_35_22_EC/service0017/char001c, domain: org.bluez.GattCharacteristic1, body: {'Notifying': True}
DEBUG  : GATT Char Properties Changed: /org/bluez/hci0/dev_F4_60_EF_35_22_EC/service0017/char001c | [{'Notifying': True}, []]
DEBUG  : DBUS: path: /org/bluez/hci0/dev_F4_60_EF_35_22_EC/service0017/char001c, domain: org.bluez.GattCharacteristic1, body: {'Notifying': True}
DEBUG  : GATT Char Properties Changed: /org/bluez/hci0/dev_F4_60_EF_35_22_EC/service0017/char001c | [{'Notifying': True}, []]
stop_notify 1
DEBUG  : DBUS: path: /org/bluez/hci0/dev_F4_60_EF_35_22_EC/service0017/char001c, domain: org.bluez.GattCharacteristic1, body: {'Notifying': False}
DEBUG  : GATT Char Properties Changed: /org/bluez/hci0/dev_F4_60_EF_35_22_EC/service0017/char001c | [{'Notifying': False}, []]
DEBUG  : DBUS: path: /org/bluez/hci0/dev_F4_60_EF_35_22_EC/service0017/char001c, domain: org.bluez.GattCharacteristic1, body: {'Notifying': False}
DEBUG  : GATT Char Properties Changed: /org/bluez/hci0/dev_F4_60_EF_35_22_EC/service0017/char001c | [{'Notifying': False}, []]
start_notify 2
DEBUG  : DBUS: path: /org/bluez/hci0/dev_F4_60_EF_35_22_EC/service0017/char001c, domain: org.bluez.GattCharacteristic1, body: {'Notifying': True}
DEBUG  : GATT Char Properties Changed: /org/bluez/hci0/dev_F4_60_EF_35_22_EC/service0017/char001c | [{'Notifying': True}, []]
DEBUG  : DBUS: path: /org/bluez/hci0/dev_F4_60_EF_35_22_EC/service0017/char001c, domain: org.bluez.GattCharacteristic1, body: {'Notifying': True}
DEBUG  : GATT Char Properties Changed: /org/bluez/hci0/dev_F4_60_EF_35_22_EC/service0017/char001c | [{'Notifying': True}, []]
```

After that nothing happens, except that my device forcefully disconnects idle connections.

dlech, 2021-03-16:
`async with BleakClient(address) as client:` implicitly calls `client.connect()`, so you are connecting twice, as can be seen in the log. I would have expected the 2nd connection attempt to fail since the device was already connected. Also connecting implicitly discovers services, so `await client.get_services()` doesn't really do anything since you aren't using the return value.

Does the issue still happen once the 2nd connection is removed?

dlech, 2021-03-16:
And actually, we recently reworked the BlueZ backend but haven't released the changes yet. So could you please test with the `develop` branch?

vliedel, 2021-03-17:
Hi, thanks for the response.

I removed the `client.connet()` call, that didn't help.
Then i removed bleak, and installed from the develop branch, and that still hangs at the same point.

Output:
```
$ python3.7 bleak-test.py 
DEBUG  : Using selector: EpollSelector
DEBUG  : Connecting to device @ f4:60:ef:35:22:ec with hci0
DEBUG  : cached devices: {'/org/bluez/hci0/dev_CB_AD_B6_FC_24_DD': {'Address': 'CB:AD:B6:FC:24:DD', 'AddressType': 'random', 'Alias': 'CB-AD-B6-FC-24-DD', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x02\x15\xe4\xb6\xf6\xe9\xd0\xc4A\xa7\xa7\x8f\xd2\x99\x98\xa1\x0b%\x92\x86\xe9\x19\xc4'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_F4_60_EF_35_22_ED': {'Address': 'F4:60:EF:35:22:ED', 'AddressType': 'random', 'Alias': 'F4-60-EF-35-22-ED', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x02\x15\x18CB>\xe1uJ\xf0\xa2\xe41\xe3/r\x9a\x8aY.\x13o\xc4'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_C6_22_E2_A1_AC_53': {'Address': 'C6:22:E2:A1:AC:53', 'AddressType': 'random', 'Name': 'CRWN', 'Alias': 'CRWN', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ServiceData': {'0000c001-0000-1000-8000-00805f9b34fb': b'\x07\x01\xb6\xde*<n\xd8\xdfi\xf9\xfb\xd6\x91;Y\x94M'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_FB_AA_1D_2E_DC_C5': {'Address': 'FB:AA:1D:2E:DC:C5', 'AddressType': 'random', 'Alias': 'FB-AA-1D-2E-DC-C5', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x02\x15\xe4\xb6\xf6\xe9\xd0\xc4A\xa7\xa7\x8f\xd2\x99\x98\xa1\x0b%\xae\x0e.>\xc4'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_2C_41_A1_87_75_32': {'Address': '2C:41:A1:87:75:32', 'AddressType': 'public', 'Name': 'LE-headcrab', 'Alias': 'LE-headcrab', 'Class': 2360344, 'Icon': 'audio-card', 'Paired': True, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00000000-deca-fade-deca-deafdecacaff', '00001101-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '81c2e72a-0591-443e-a1ff-05f988593351', 'f8d1fbe4-7966-4334-8024-ff96c9330e15'], 'Modalias': 'bluetooth:v009Ep4020d0251', 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}, '/org/bluez/hci0/dev_F4_60_EF_35_22_EC': {'Address': 'F4:60:EF:35:22:EC', 'AddressType': 'random', 'Name': 'BART', 'Alias': 'BART', 'Appearance': 512, 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '24f00000-7d10-4805-bfc1-7663a01c3bff'], 'Adapter': '/org/bluez/hci0', 'ServiceData': {'0000c001-0000-1000-8000-00805f9b34fb': b'\x07\x01{*\xd1W\x9fWC\xd9\xf5\xaa\x89\x14O\xbc\x965'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_F0_18_98_4A_54_95': {'Address': 'F0:18:98:4A:54:95', 'AddressType': 'public', 'Name': 'Elwin’s MacBook Pro', 'Alias': 'Elwin’s MacBook Pro', 'Class': 3670284, 'Icon': 'computer', 'Paired': True, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00001101-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001112-0000-1000-8000-00805f9b34fb', '00001117-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '02030302-1d19-415f-86f2-22a2106a0a77'], 'Modalias': 'bluetooth:v004Cp4A00d1011', 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}}
DEBUG  : received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', True)>}, []]
DEBUG  : received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_F4_60_EF_35_22_EC): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -54)>, 'ServiceData': <dbus_next.signature.Variant ('a{sv}', {'0000c001-0000-1000-8000-00805f9b34fb': <dbus_next.signature.Variant ('ay', b'\x07\x01H\x91[|\xaceQo\x84\xf8\x1f\x16\xbf\x15NJ')>})>}, []]
DEBUG  : received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', False)>}, []]
DEBUG  : received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_F4_60_EF_35_22_EC): ['org.bluez.Device1', {}, ['RSSI']]
DEBUG  : received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_F4_60_EF_35_22_EC): ['org.bluez.Device1', {'Connected': <dbus_next.signature.Variant ('b', True)>}, []]
DEBUG  : Connection successful (/org/bluez/hci0/dev_F4_60_EF_35_22_EC)
DEBUG  : Waiting for ServicesResolved (/org/bluez/hci0/dev_F4_60_EF_35_22_EC)
[... cut ...]
DEBUG  : received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_F4_60_EF_35_22_EC/service0017/char001c): ['org.bluez.GattCharacteristic1', {'Notifying': <dbus_next.signature.Variant ('b', False)>}, []]
start_notify 2
DEBUG  : received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_F4_60_EF_35_22_EC/service0017/char001c): ['org.bluez.GattCharacteristic1', {'Notifying': <dbus_next.signature.Variant ('b', True)>}, []]
```

dlech, 2021-03-17:
I was not able to reproduce the issue using the `develop` branch, so I think this might be a problem with your device. Does it work if you add some delay between starting and stopping notifications?

vliedel, 2021-03-18:
My colleague can't reproduce it either, while connecting to the same type of device.

Adding `await asyncio.sleep(0.1)` between `stop_notify` and the 2nd `start_notify` fixes it for me indeed :)

So i guess it's more likely a difference in OS, bluez, and/or bluetooth adapter.

lightoze, 2023-03-24:
@vliedel, have you managed to control this service? I believe it's the same LED lamp as I have based on service UUID 🙂

vliedel, 2023-03-27:
> @vliedel, have you managed to control this service? I believe it's the same LED lamp as I have based on service UUID 🙂

Hey, this must be a Crownstone, as that's what I was working on at the time. I know very well how to control those ;)

lightoze, 2023-03-27:
> Hey, this must be a Crownstone, as that's what I was working on at the time. I know very well how to control those ;)

Haha, it means I was looking at the wrong device in the BLE sniffer tool, thanks for pointing that out! I have a Crownstone too )))