hbldh/bleak #490: Bluetooth dead-end after a timeout occurs

sneko, 2021-03-22:
* bleak version: v0.11.0
* Python version: v3.7.3
* Operating System: Rasp Zero
* BlueZ version (`bluetoothctl -v`) in case of Linux: bluetoothctl: 5.50

### Description

Hi,

I'm not able to find why I'm getting this but I can set some context:

I need to communicate with different devices and to avoid parallelism issues I'm using an `asyncio.Lock()` so there is no concurrent access to the bluetooth driver.

Each communication has this flow: `connect/send_command/read_response/disconnect` so the runtime can then communicate with another one.

Most of the time it works well until I get a timeout to connect to my probe (I set a timeout of 15s):
```
Traceback (most recent call last):
  File "/home/xxxxxxxxx/src/yyy.py", line 355, in _while_connected
    connect_success = await ble_client.connect(timeout=BLUETOOTH_TIMEOUT_CONNECT)
  File "/home/xxxxxxxxx/.venv/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 335, in connect
    await self.get_services()
  File "/home/xxxxxxxxx/.venv/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 589, in get_services
    await asyncio.wait_for(self._services_resolved_event.wait(), 5)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```

Which can seem fine because a network (wireless moreover) is not perfect. But on the next bluetooth communication I will no longer get a timeout error, but this one:
```
Traceback (most recent call last):
  File "/home/xxxxxxxx/src/yyy.py", line 355, in _while_connected
    connect_success = await ble_client.connect(timeout=BLUETOOTH_TIMEOUT_CONNECT)
  File "/home/xxxxxxxx/.venv/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 256, in connect
    self.services.add_service(BleakGATTServiceBlueZDBus(obj, path))
  File "/home/xxxxxxxx/.venv/lib/python3.7/site-packages/bleak/backends/service.py", line 126, in add_service
    "This service is already present in this BleakGATTServiceCollection!"
bleak.exc.BleakError: This service is already present in this BleakGATTServiceCollection!
```

**After that I'm always getting the above error.**

So I kill my program, restart it. I thought it would work fine but now my existing device is no longer found:
```
Traceback (most recent call last):
  File "/home/xxxxxx/src/yyyyyyyy.py", line 355, in _while_connected
    connect_success = await ble_client.connect(timeout=BLUETOOTH_TIMEOUT_CONNECT)
  File "/home/xxxxxx/.venv/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 135, in connect
    "Device with address {0} was not found.".format(self.address)
bleak.exc.BleakError: Device with address c2:5a:35:a1:da:28 was not found.
```

And that's the case forever...

I enabled the Bleak logs and saw it was in the cache registry even if it was not able to connect, just to give a try I removed my device with `bluetoothctl -- remove c2:5a:35:a1:da:28` and... surprize... after restarting my program it was able to connect to the device within a few seconds.

I'm a bit lost about what can be done. It's always the same scenario so I guess there is a logical (hardware?) reason..

The strange thing for me after looking at your code is the `This service is already present in this BleakGATTServiceCollection!` (https://github.com/hbldh/bleak/blob/4a3f8620a007be4c5a337e12c3019da916111363/bleak/backends/service.py#L125-L127). How is it possible since my device is then disconnect/connect/disconnect... this should not happen?

Thank you,

cc @hbldh @dlech

hbldh, 2021-03-23:
> So I kill my program, restart it.

In this case, BlueZ is probably still connected to your device. You have to terminate properly (and preferably use `asyncio.run` to start your program) to make certain that Bleak can disconnect all the way through to BlueZ. You have to use `bluetoothctl`, as you found out, to fix this in this case.

> bleak.exc.BleakError: This service is already present in this BleakGATTServiceCollection!

This is highly problematic. I had envisioned that the possibility to get this error would disappear in version 0.11.0...
Could you change the BleakError string (Line 125 - 127) to this, run again and report on the error log this time:

```python
raise BleakError( 
     f"Service {service} ({service.path}) is already present in this BleakGATTServiceCollection: {k:v.path for k,v in self.services.items()}!" 
 ) 
```

@dlech Should `get_services` timeout be fixed (it should be good enough, imo) or do we need to make it configurable? Or jsut use the already existing `self._timeout`?

sneko, 2021-03-23:
@hbldh sorry for the delay of answering since that's a "random" issue ^^.

Just to note, your log was missing squared brackets to work:
```
            raise BleakError(
                f"Service {service} ({service.path}) is already present in this BleakGATTServiceCollection: {[(k, v.path) for k,v in self.services.items()]}!"
            )
```

Here the log I had:
```
bleak.exc.BleakError: Service ed8095cf-8a94-71b3-b94d-548763d575af (Handle: 20): Unknown (/org/bluez/hci0/dev_C2_5A_35_A1_DA_28/service0014) is already present in this BleakGATTServiceCollection: [(20, '/org/bluez/hci0/dev_C2_5A_35_A1_DA_28/service0014'), (15, '/org/bluez/hci0/dev_C2_5A_35_A1_DA_28/service000f'), (11, '/org/bluez/hci0/dev_C2_5A_35_A1_DA_28/service000b'), (10, '/org/bluez/hci0/dev_C2_5A_35_A1_DA_28/service000a')]!
```

So indeed we can see the service is already in the collection... Is it possible when timed out, the service information was on the way, and Bleak received it whereas it throwed the `TimeoutError`?

Just trying to understand, I'm not that familiar. But having a the timeout should not make my device stuck as idle and me having to run `bluetooth remove`. For context, when having a connect() error, I get into a `finally` section that will check if the device is connected, and if so, will `bleclient.disconnect()`.

> You have to terminate properly (and preferably use asyncio.run to start your program) to make certain that Bleak can disconnect all the way through to BlueZ
About this, when "killing" my program, I'm not doing multiple interrupt signals (that kills the program without delay) I'm always waiting for asyncio to finish properly (so all `finally` sections should be processed from what I understand).

**Aside problem/question, but related to your expectations on running `asyncio.run`:**
A doubt remains... some bluetooth events can be triggered by websockets callbacks that wait for messages thanks to `asyncio.create_task(...)`. It works fine most of the time but sometimes I get:
```
txdbus.error.RemoteError: org.bluez.Error.InProgress: In Progress
```

I'm wondering if I should pass something to the `BLEClient` as `loop` parameter? For me since it was explicit I'm using `asyncio.run(main())` and inside the main function using `asyncio.create_task()` it was implicit Bleak clients would use same loop, no?

dlech, 2021-03-23:
> txdbus.error.RemoteError: org.bluez.Error.InProgress: In Progress

If you are seeing `txdbus`, then you are not using v0.11.0.

sneko, 2021-03-23:
@dlech hehe, took this error from a previous test last week (didn't store logs of yesterday about the `InProgress`) and indeed it was an older version. But I had also this error yesterday with `v0.11.0`.

dlech, 2021-03-23:
Can you share the full debug logging when the error happens? I have a hunch that we might be getting `invalidated_properties` for the services in `org.freedesktop.DBus.Properties.PropertiesChanged` that we are not handling.

sneko, 2021-03-29:
@dlech in the past tests I didn't get the `InProgress`. I will keep you updated on that.

@hbldh with the modified log you asked me to set, does it help you?

Thank you,

sneko, 2021-03-29:
@dlech here the Bluez logs associated with the `InProgress` that I just got:
```
2021-03-29 14:49:21,242 bleak.backends.bluezdbus.client DEBUG: Connecting to device @ df:0e:7e:8c:d7:26 with hci0
2021-03-29 14:49:22,370 bleak.backends.bluezdbus.scanner DEBUG: cached devices: {'/org/bluez/hci0/dev_DF_0E_7E_8C_D7_26': {'Address': 'DF:0E:7E:8C:D7:26', 'AddressType': 'random', 'Name': 'MY_NAME', 'Alias': 'MY_NAME', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000fe59-0000-1000-8000-00805f9b34fb', 'ed8095cf-8a94-71b3-b94d-548763d575af'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False, 'AdvertisingFlags': b'\x06'}, '/org/bluez/hci0/dev_C2_5A_35_A1_DA_28': {'Address': 'C2:5A:35:A1:DA:28', 'AddressType': 'random', 'Name': 'MY_NAME', 'Alias': 'MY_NAME', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000fe59-0000-1000-8000-00805f9b34fb', 'ed8095cf-8a94-71b3-b94d-548763d575af'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False, 'AdvertisingFlags': b'\x06'}, '/org/bluez/hci0/dev_FF_65_EC_B1_3F_80': {'Address': 'FF:65:EC:B1:3F:80', 'AddressType': 'random', 'Name': 'MY_NAME', 'Alias': 'MY_NAME', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000180f-0000-1000-8000-00805f9b34fb', '8ec91111-f315-4f60-9fb8-838830daea50'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False, 'AdvertisingFlags': b'\x06'}}
2021-03-29 14:49:37.447 | ERROR    | client.hardware.probe.bluetooth_probes_manager:_while_connected:475 - Connection has failed, please see attached details for the original error
Traceback (most recent call last):
  File "/xxxxxxxxxxxxx/yyyyyyyyyy/bluetooth_probes_manager.py", line 430, in _while_connected
    connect_success = await ble_client.connect(timeout=BLUETOOTH_TIMEOUT_CONNECT)
  File "/xxxxxxxxxxxxx/.venv/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 127, in connect
    self.address, timeout=timeout, adapter=self._adapter
  File "/xxxxxxxxxxxxx/.venv/lib/python3.7/site-packages/bleak/backends/scanner.py", line 175, in find_device_by_address
    for d in await scanner.get_discovered_devices()
  File "/xxxxxxxxxxxxx/.venv/lib/python3.7/site-packages/bleak/backends/scanner.py", line 67, in __aexit__
    await self.stop()
  File "/xxxxxxxxxxxxx/.venv/lib/python3.7/site-packages/bleak/backends/bluezdbus/scanner.py", line 167, in stop
    assert_reply(reply)
  File "/xxxxxxxxxxxxx/.venv/lib/python3.7/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name)
bleak.exc.BleakDBusError: org.bluez.Error.InProgress
```

_Note: contrary to other times it did not happen during the runtime, this time it was because my asyncio was stuck so I did Ctrl+C interruption 2 times (not disconnecting properly probes). I don't think it would give that much information though._

dlech, 2022-09-03:
There have been a number of fixes recently that should resolve most of the errors seen here. Let's start a new issue if similar problems are seen in Bleak >= 0.16. There are also several other open issues about `InProgress`.

closed 2022-09-03 by dlech